nfcgate/nfcgate #96: Mifare Ultralight

raffysommy, 2017-10-08:
Hi,
I'm testing nfcgate with a mifare ultralight tag, but it seems that does not work.
There is a know bug with this type of tag?
Thanks

malexmave, 2017-10-09:
We don't have access to a Mifare Ultralight tag ourselves, so we cannot really test this ourselves. We have not explicitly implemented support for these chips - if they are implemented separately from NFCA in Android, they will probably not work by default.

If you'd like to attempt implementing support for them yourselves, take a look at [this package](https://github.com/nfcgate/nfcgate/tree/dev/app/src/main/java/tud/seemuh/nfcgate/nfc/reader) - depending on the Ultralight backend in Android, implementing support may be trivial or impossible (we need all the magic bytes: UID, Hist, ATQA, SAK. If they are not provided by the Ultralight backend, or not part of the Ultralight standard, our current setup is unable to emulate them). As we don't have a compatible chip ourselves, we cannot implement and test this ourselves, sorry.

If you manage to get it to work, feel free to send a pull request :).

roussosalex, 2020-03-07:
Mifare Ultralight is not compatible with Android HCE

closed 2020-03-07 by roussosalex