nfcgate/nfcgate #98: The logcat about crashing bug

killshadow, 2018-05-29:
```bash
05-29 22:56:47.937 13360-13360/? I/art: Late-enabling -Xcheck:jni
05-29 22:56:47.957 13360-13360/? D/ActivityThread: ActivityThread,attachApplication
05-29 22:56:47.967 13360-13374/? E/art: Failed sending reply to debugger: Broken pipe
05-29 22:56:47.967 13360-13374/? I/art: Debugger is no longer active
05-29 22:56:48.017 13360-13360/? W/ActivityThread: Application tud.seemuh.nfcgate is waiting for the debugger on port 8100...
05-29 22:56:48.022 13360-13360/? I/System.out: Sending WAIT chunk
05-29 22:57:05.142 13360-13374/tud.seemuh.nfcgate I/art: Debugger is active
05-29 22:57:05.337 13360-13360/tud.seemuh.nfcgate I/System.out: Debugger has connected
    waiting for debugger to settle...
05-29 22:57:05.542 13360-13360/tud.seemuh.nfcgate I/System.out: waiting for debugger to settle...
05-29 22:57:05.742 13360-13360/tud.seemuh.nfcgate I/System.out: waiting for debugger to settle...
05-29 22:57:05.942 13360-13360/tud.seemuh.nfcgate I/System.out: waiting for debugger to settle...
05-29 22:57:06.142 13360-13360/tud.seemuh.nfcgate I/System.out: waiting for debugger to settle...
05-29 22:57:06.342 13360-13360/tud.seemuh.nfcgate I/System.out: waiting for debugger to settle...
05-29 22:57:06.547 13360-13360/tud.seemuh.nfcgate I/System.out: waiting for debugger to settle...
05-29 22:57:06.747 13360-13360/tud.seemuh.nfcgate I/System.out: debugger has settled (1327)
05-29 22:57:06.847 13360-13360/tud.seemuh.nfcgate I/HwCust: Constructor found for class android.app.HwCustApplicationPackageManagerImpl
05-29 22:57:06.847 13360-13360/tud.seemuh.nfcgate D/HwCust: Create obj success use class android.app.HwCustApplicationPackageManagerImpl
05-29 22:57:06.917 13360-13360/tud.seemuh.nfcgate E/ZipFileCache: init failed when open zip file:/data/skin/icons
05-29 22:57:06.977 13360-13360/tud.seemuh.nfcgate W/PG Utils: set mRatioMinBrightness to default value for other product
    mRatioMinBrightness = 35
05-29 22:57:06.982 13360-13447/tud.seemuh.nfcgate D/OpenGLRenderer: Use EGL_SWAP_BEHAVIOR_PRESERVED: false
05-29 22:57:07.387 13360-13360/tud.seemuh.nfcgate D/Atlas: Validating map...
05-29 22:57:07.397 13360-13360/tud.seemuh.nfcgate I/View: ssignParent(ViewParent parent) parent is: android.view.ViewRootImpl@38125ab2
05-29 22:57:07.552 13360-13360/tud.seemuh.nfcgate D/RelayFragment: onCreateView
05-29 22:57:07.567 13360-13360/tud.seemuh.nfcgate I/RelayFragment: onResume(): intent: android.intent.action.MAIN
05-29 22:57:07.667 13360-13447/tud.seemuh.nfcgate I/OpenGLRenderer: Initialized EGL, version 1.4
05-29 22:57:07.687 13360-13447/tud.seemuh.nfcgate D/OpenGLRenderer: Enabling debug mode 0
05-29 22:57:07.692 13360-13447/tud.seemuh.nfcgate D/mali_winsys: new_window_surface returns 0x3000
05-29 22:57:11.587 13360-13360/tud.seemuh.nfcgate D/android.widget.GridLayout: vertical constraints: y1-y0>=118, y2-y1>=0, y3-y2>=118, y4-y3>=0, y5-y4>=118, y6-y5>=0, y7-y6>=162, y8-y7>=0, y9-y8>=162, y10-y9>=0, y11-y10>=0, y12-y11>=0, y13-y12>=0, y14-y13>=0, y15-y14>=0, y16-y15>=0, y17-y16>=0, y18-y17>=0, y19-y18>=0, y20-y19>=0, y21-y20>=0, y22-y21>=0, y23-y22>=0, y24-y16>=199, y25-y24>=0, y26-y25>=0, y27-y26>=0, y28-y27>=0, y29-y28>=0, y30-y29>=0, y30-y0<=701 are inconsistent; permanently removing: y30-y0<=701. 
05-29 22:57:17.907 13360-13360/tud.seemuh.nfcgate D/SinkManager: addSinkCommon: Adding Sink type DISPLAY_TEXTVIEW
    addSinkCommon: Adding Sink type SESSION_LOG
05-29 22:57:17.907 13360-13360/tud.seemuh.nfcgate I/NfcManager: start: Starting SinkManager Thread
05-29 22:57:17.917 13360-13360/tud.seemuh.nfcgate D/ProtobufHandler: createSession: Trying to create session
05-29 22:57:17.922 13360-13499/tud.seemuh.nfcgate I/System: core_booster, getBoosterConfig = false
05-29 22:57:17.922 13360-13499/tud.seemuh.nfcgate I/System.out: [CDS]connect[/192.168.137.1:5566]
05-29 22:57:17.927 13360-13497/tud.seemuh.nfcgate I/SessionLoggingSink: run: Starting
05-29 22:57:17.927 13360-13499/tud.seemuh.nfcgate I/System.out: [CDS]port[47100]
05-29 22:57:17.932 13360-13500/tud.seemuh.nfcgate D/tud.seemuh.nfcgate.network.LowLevelTCPHandler$CommunicationThread: started new CommunicationThread
05-29 22:57:17.937 13360-13497/tud.seemuh.nfcgate I/SessionLoggingSink: run: New Session created: 3
05-29 22:57:17.957 13360-13360/tud.seemuh.nfcgate W/UnsafeUtil: platform method missing - proto runtime falling back to safer methods: java.lang.NoSuchMethodException: getByte [long]
    platform method missing - proto runtime falling back to safer methods: java.lang.NoSuchMethodException: getByte [class java.lang.Object, long]
05-29 22:57:17.967 13360-13500/tud.seemuh.nfcgate D/tud.seemuh.nfcgate.network.LowLevelTCPHandler$CommunicationThread: Got 4 bytes
05-29 22:57:17.967 13360-13500/tud.seemuh.nfcgate I/tud.seemuh.nfcgate.network.LowLevelTCPHandler$CommunicationThread: Reading bytes of length:14
05-29 22:57:17.967 13360-13500/tud.seemuh.nfcgate D/tud.seemuh.nfcgate.network.LowLevelTCPHandler$CommunicationThread: Read data: 320C080112063934373334351800
    Delegating to Callback.
05-29 22:57:17.972 13360-13500/tud.seemuh.nfcgate I/ProtobufCallback: onDataReceived: MessageCase.SESSION: Sending to handler
05-29 22:57:17.972 13360-13500/tud.seemuh.nfcgate D/ProtobufCallback: handleSession: SESSION_CREATE_SUCCESS: Delegating to Handler
05-29 22:57:17.972 13360-13500/tud.seemuh.nfcgate D/ProtobufHandler: confirmSessionCreation: Session created with token 947345
05-29 22:57:17.972 13360-13500/tud.seemuh.nfcgate D/tud.seemuh.nfcgate.network.LowLevelTCPHandler$CommunicationThread: Callback finished execution.
05-29 22:57:28.582 13360-13500/tud.seemuh.nfcgate D/tud.seemuh.nfcgate.network.LowLevelTCPHandler$CommunicationThread: Got 4 bytes
05-29 22:57:28.582 13360-13500/tud.seemuh.nfcgate I/tud.seemuh.nfcgate.network.LowLevelTCPHandler$CommunicationThread: Reading bytes of length:6
05-29 22:57:28.582 13360-13500/tud.seemuh.nfcgate D/tud.seemuh.nfcgate.network.LowLevelTCPHandler$CommunicationThread: Read data: 320408091800
    Delegating to Callback.
05-29 22:57:28.582 13360-13500/tud.seemuh.nfcgate I/ProtobufCallback: onDataReceived: MessageCase.SESSION: Sending to handler
05-29 22:57:28.582 13360-13500/tud.seemuh.nfcgate D/ProtobufCallback: handleSession: SESSION_PEER_JOINED: Delegating to Handler
05-29 22:57:28.582 13360-13500/tud.seemuh.nfcgate D/tud.seemuh.nfcgate.network.LowLevelTCPHandler$CommunicationThread: Callback finished execution.
05-29 22:57:39.907 13360-13500/tud.seemuh.nfcgate D/tud.seemuh.nfcgate.network.LowLevelTCPHandler$CommunicationThread: Got 4 bytes
05-29 22:57:39.907 13360-13500/tud.seemuh.nfcgate I/tud.seemuh.nfcgate.network.LowLevelTCPHandler$CommunicationThread: Reading bytes of length:34
05-29 22:57:39.907 13360-13500/tud.seemuh.nfcgate D/tud.seemuh.nfcgate.network.LowLevelTCPHandler$CommunicationThread: Read data: 2A200800121C1A1A0A04800960A7120B209040F716C6B4800960A71A020800220120
    Delegating to Callback.
05-29 22:57:39.917 13360-13500/tud.seemuh.nfcgate I/ProtobufCallback: onDataReceived: MessageCase.DATA: Sending to handler
05-29 22:57:39.927 13360-13500/tud.seemuh.nfcgate I/ProtobufCallback: onDataReceived: MessageCase.ANTICOL: Sending to handler
    handleAnticol: got anticol values
05-29 22:57:39.932 13360-13500/tud.seemuh.nfcgate E/AndroidRuntime: FATAL EXCEPTION: Thread-806
    Process: tud.seemuh.nfcgate, PID: 13360
    java.lang.ArrayIndexOutOfBoundsException: src.length=4 srcPos=2 dst.length=9 dstPos=0 length=9
        at java.lang.System.arraycopy(Native Method)
        at tud.seemuh.nfcgate.nfc.config.ConfigBuilder.parse(ConfigBuilder.java:40)
        at tud.seemuh.nfcgate.nfc.config.ConfigBuilder.<init>(ConfigBuilder.java:12)
        at tud.seemuh.nfcgate.network.ProtobufCallback.handleAnticol(ProtobufCallback.java:123)
        at tud.seemuh.nfcgate.network.ProtobufCallback.handleWrapperMessage(ProtobufCallback.java:75)
        at tud.seemuh.nfcgate.network.ProtobufCallback.handleData(ProtobufCallback.java:91)
        at tud.seemuh.nfcgate.network.ProtobufCallback.handleWrapperMessage(ProtobufCallback.java:59)
        at tud.seemuh.nfcgate.network.ProtobufCallback.onDataReceived(ProtobufCallback.java:46)
        at tud.seemuh.nfcgate.network.LowLevelTCPHandler$CommunicationThread.run(LowLevelTCPHandler.java:224)
        at java.lang.Thread.run(Thread.java:831)
05-29 22:57:40.017 13360-13500/tud.seemuh.nfcgate I/Process: Sending signal. PID: 13360 SIG: 9
```
> HCE-Phone is Huawei Mate 7 which system is EMUI3.1 android5.1.1, CPU is Hisilicon Kirin 925, and NFC chip is NXP pn544.
- The description about this bug I had upload in  commit. Further more, I have send PhD Max an email. He answered my question very seriously and kindly, thanked him here~

kleest, 2018-05-29:
Could you try the updated server version in nfcgate/server#1?

killshadow, 2018-05-30:
Sure, I will upload my server.py in github. But it's not very good on NXP chip phones. Could you please updated server version in nfcgate/server ?