nfcgate/nfcgate #99: About NFCgate developer question

killshadow, 2018-06-22:
Hey, guys. Now, I try coding nfcgate through your source code. I want to add a function that can edit logging detail data in LOGGING. When I update SQLite, the data(I input data in editText) cann't write in databases. I think it maybe something wrong in data type(I input String, however `COLUMN_NAME_NFCDATA` data type is Blob). Do you have any idea in order change Hex String to Blob in order update databases?
```java
    private class AsyncSessionEditor extends AsyncTask<NfcSession, Void, Void> {
        private final String TAG = "AsyncSessionEditor";
        private SQLiteDatabase mDB;

        @Override
        protected Void doInBackground(NfcSession... nfc) {
            Log.d(TAG, "doInBackground: Started");
            // Get a DB object
            SessionLoggingDbHelper dbHelper = new SessionLoggingDbHelper(getActivity());
            mDB = dbHelper.getWritableDatabase();

            ContentValues values = new ContentValues();

            values.put(SessionLoggingContract.SessionEvent.COLUMN_NAME_NFCDATA, nfc[0].getData().getBytes());

            String selection = SessionLoggingContract.SessionEvent._ID + " LIKE ?";

            String[] selectionArgs = { String.valueOf(nfc[0].getEID()) };

            Log.d(TAG, "doInBackground: EID = " + String.valueOf(nfc[0].getID()));
            mDB.update(
                    SessionLoggingContract.SessionEvent.TABLE_NAME,
                    values,
                    selection,
                    selectionArgs
            );
            Log.d(TAG, "doInBackground: Update done, returning ");
            return null;
        }

        @Override
        protected void onPostExecute(Void v) {
            mDB.close();
            confirmEditData();
        }
    }
```
Those code in `LoggingDetailFragment.java` . 

Thanks very much.

killshadow, 2018-06-26:
Does any people know how to send data what I want. :)

roussosalex, 2018-06-27:
Replaying logged data is currently not supported.

closed 2018-06-28 by malexmave