nfcgate/nfcgate #101: Error in building

ahvahsky2008, 2019-06-18:
```
C:\Users\1\Desktop\nfcgate-2\app\src\main\java\de\tu_darmstadt\seemoo\nfcgate\util\NfcComm.java:6: error: package de.tu_darmstadt.seemoo.nfcgate.network.c2c.C2C does not exist
import de.tu_darmstadt.seemoo.nfcgate.network.c2c.C2C.NFCData;
                                                     ^
C:\Users\1\Desktop\nfcgate-2\app\src\main\java\de\tu_darmstadt\seemoo\nfcgate\util\NfcComm.java:13: error: cannot find symbol
    private NFCData mData;
            ^
  symbol:   class NFCData
  location: class NfcComm
```

kleest, 2019-06-18:
Did you initialize the protocol submodule (e.g. `git submodule update --init`)?

ahvahsky2008, 2019-06-19:
> Did you initialize the protocol submodule (e.g. `git submodule update --init`)?

I simple download project

malexmave, 2019-06-19:
Then try initializing the submodules and compile again - it won't work without the submodules in place.

kleest, 2019-07-07:
Clarified instructions in 5a24b4ab49eeb01e8ae39d49e44e52ceddc800c3.

closed 2019-07-07 by kleest