nfcgate/nfcgate #103: Nexus 5X: Xposed framework not recognised by nfcgate

Geppetto123, 2019-07-24:
I am currently running three different versions of the Xposed framework on three Nexus 5X devices, and the devices are running on different Android versions (6.0.1, 7.0, 8.1.0).

In all devices nfcgate does not recognise the Xposed framework, even if the nfcgate module is activated, and the framework is up and running. (Warning: missing Xposed module)

Is there any other activation to perform, or have you encountered the same issue in any other device?

[image]
[image]
[image]

kleest, 2019-07-24:
Just tested it with the 2.0 release on a Nexus 5X 7.1.2 (N2G48C) and the Xposed module is detected properly. In our testings, we used different Android versions (6.0, 7.1, 8.0) to ensure [compatibility](https://github.com/nfcgate/nfcgate/blob/v2/doc/Compatibility.md).  
Have you restarted your phone after activating the NFCGate Xposed module?

Geppetto123, 2019-07-24:
I did the same procedure with the 8.0 as in the compatibility list, and keep getting the same error. I did the restart, but the error persists.

Is there an order when doing the procedure, or is it indifferent?

Geppetto123, 2019-07-24:
Xposed log:

`07-23 20:06:43.085 I/Xposed  (  557): Loading modules from /data/app/de.tu_darmstadt.seemoo.nfcgate-RgjCoohVcuItRzq4lsKTFw==/base.apk
07-23 20:06:43.098 I/Xposed  (  557):   Loading class de.tu_darmstadt.seemoo.nfcgate.xposed.Hooks
07-23 20:06:43.099 E/Xposed  (  557):     Failed to load class de.tu_darmstadt.seemoo.nfcgate.xposed.Hooks
07-23 20:06:43.099 E/Xposed  (  557): java.lang.ClassNotFoundException: Didn't find class "de.tu_darmstadt.seemoo.nfcgate.xposed.Hooks" on path: DexPathList[[zip file "/data/app/de.tu_darmstadt.seemoo.nfcgate-RgjCoohVcuItRzq4lsKTFw==/base.apk"],nativeLibraryDirectories=[/system/lib64, /vendor/lib64]]
07-23 20:06:43.099 E/Xposed  (  557): 	at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:125)
07-23 20:06:43.099 E/Xposed  (  557): 	at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
07-23 20:06:43.099 E/Xposed  (  557): 	at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
07-23 20:06:43.099 E/Xposed  (  557): 	at de.robv.android.xposed.XposedInit.loadModule(XposedInit.java:536)
07-23 20:06:43.099 E/Xposed  (  557): 	at de.robv.android.xposed.XposedInit.loadModules(XposedInit.java:466)
07-23 20:06:43.099 E/Xposed  (  557): 	at de.robv.android.xposed.XposedBridge.main(XposedBridge.java:97)`

kleest, 2019-07-24:
That's strange. There is no special setup needed: just install the app, activate the module and reboot, the order does not matter.  

Did you install the release version or build the app yourself?

Geppetto123, 2019-07-31:
Problem solved. Just needed to disable instant run when building with Android Studio.

Thanks for the support.

closed 2019-07-31 by Geppetto123

M1999-maestru, 2025-05-23:
Mă ajută și pe mine cu o instalare de nfcgate contra cost

M1999-maestru, 2025-05-23:
Pe un Samsung