nfcgate/nfcgate #104: Mifare classic Tag support

Mera-balou, 2019-12-11:
The project specified that : "it only supports tags supported by Android. For example, **Android no longer offers support for MiFare classic chips**, so these cards are not supported."

Do you know some android device (maybe old) that still support MiFare classic chips?
If not, do you think that it would be possible to connect usb external RFID antenna to the smartphone (like [DL533N XL OEM ](https://webshop.d-logic.net/products/nfc-rfid-reader-writer/dl533n-libnfc-series/dl533n-xl-oem.html)from digital logic online store - LibNFC-Compatible) and still use NFC gate?

Thanks

malexmave, 2019-12-12:
The problem is that part of our card emulation code is specific to broadcom NFC Chips, which do not support MiFare chips. So, if you were to find an old device that supports MiFare Classic (like the Nexus S), I think it would likely not work with emulation, as the chip would not be supported by our emulation code (correct me if I'm wrong, @roussosalex @kleest)

The same is true for external readers: We would have to write some new code to support the chip, and there is no guarantee that this is even possible, with the added problem that we would also have to support external readers (I don't believe Android supports them out of the box, but I may be wrong).

Mera-balou, 2019-12-13:
Many external readers are compatible with [lib nfc](https://github.com/nfc-tools/libnfc) (pn53x chip). I am not a developer, but it may be possible to add this library (which is already available on github) in your app.

For example, [RFIDtools](https://github.com/RfidResearchGroup/RFIDtools) (android apk) already support external antenna like ACS ACR-122u because they implemented [libnfc](https://github.com/RfidResearchGroup/RFIDtools/tree/master/pn53x) and android detect theses devices OTG.

kleest, 2019-12-17:
Android uses libnfc-nci as NFC stack which is different than libnfc despite their naming similarities. Adding support for libnfc requires a lot of work because we cannot use Android's NFC system APIs for tag communication and HCE when not using the builtin libnfc-nci stack: While possible, we would have to implement our own NFC stack in userspace like in RFIDtools.

The implementation is not specific to Broadcom NFC chipsets but to the libnfci-nci stack, so everything supported by Android's HCE is supported by NFCGate, while some devices may be able to read old Mifare Classic tags (Mifare Classic and Ultralight tags are only optionally supported). Besides that, NFCGate should only support tag technologies which can be emulated (see Android HCE documentation) as this tool is about cloning, relaying, and replaying NFC data. If you are looking for an application which reads tags and displays their details, I recommend [NFC TagInfo](https://play.google.com/store/apps/details?id=com.nxp.taginfolite).

closed 2019-12-17 by kleest