nfcgate/nfcgate #106: Use nfcgate with magisk

jekkos, 2019-12-22:
I think it would be nice to have the app installed while still passing safetynet attestation.

The documentation states that nfcgate won't work with eg Google pay as it needs the xposed framework which would trip safetynet.

However it seems xposed modules can be run without tripping safetynet using systemless magisk and edxposed. Also i found [a report](https://forum.xda-developers.com/apps/magisk/magisk-google-pay-gms-17-1-22-pie-t3929950/page78)  on a working Google pay in combination with magisk.

Do you think that there is a chance one could make nfcgate work by using edxposed instead? Not sure but in that case the binary instrumentation done here might need to be ported to riru (also a magisk module) instead? It are there other parts that require a system partition root?

Thanks!

kleest, 2019-12-22:
Using NFCGate with EdXposed and Magisk should be possible as long as the Android NFC service can be hooked. We have not tried this yet.

Here are some thoughts regarding our previous experience with the implementation:
* We had some difficulties with the class loader when we injected our own Java classes into the NFC service. The way Xposed handles the class loader on Nougat (and previous versions) and Oreo differs. EdXposed may be using yet another approach so that NFCGate might need implementation changes.
* Depending on the Android version the internal NFC service classes change so that NFCGate might need changes to support Pie and higher.

Regarding compatibility with Google Pay, I can tell you from my own experience that getting GPay and Magisk without modules to work heavily depends on the device and ROM. [EdXposed is a whole different story](https://github.com/ElderDrivers/EdXposed/issues/386). I recommend using the whitelist mode of EdXposed where you enable hooking only for the NFC service (`com.android.nfc`). Although I am actively following the development about EdXposed compatibility with GPay, I have tried it yet. I like to hear your feedback, if you happen to try it out.

jekkos, 2019-12-24:
Ok thanks for the pointers already. I might give it a go on my pixel 2 the coming weeks. I'll keep you posted on the progress.

matthew5025, 2020-01-02:
@jekkos Have you had any success? I tried and this is the resulting logcat.

`01-02 22:37:00.666   359   359 I ServiceManager: service 'nfc' died
01-02 22:37:00.673  1617  1639 I ActivityManager: Process com.android.nfc (pid 8730) has died: pers PER
01-02 22:37:00.674  1617  1639 W ActivityManager: Re-adding persistent process ProcessRecord{1897ab1 8730:com.android.nfc/1027}
01-02 22:37:00.675  1617  1659 W libprocessgroup: kill(-8730, 9) failed: No such process
01-02 22:37:00.677   490  8758 E NxpHal  :  Ignoring read , HAL close triggered
01-02 22:37:00.677   490  8758 W android.hardware.nfc@1.0-impl: Failed to call back into NFC process.
01-02 22:37:00.697  2770  5376 I GmsTraceSample: 8ms performing container.core-BackgroundBroadcastReceiverSupport$GmsReceiverIntentOperation_onHandleIntentWithRedelivery [0.5% sampled, 23 omitted, module=container.core, caller=null]
01-02 22:37:00.709   604   604 I Zygote  : Process 8730 exited due to signal (11)
01-02 22:37:00.720  1617  1659 W libprocessgroup: kill(-8730, 9) failed: No such process
01-02 22:37:00.734  1617  1658 W ActivityManager: Slow operation: 60ms so far, now at startProcess: returned from zygote!
01-02 22:37:00.734  1617  1658 W ActivityManager: Slow operation: 60ms so far, now at startProcess: done updating battery stats
01-02 22:37:00.734  1617  1658 W ActivityManager: Slow operation: 60ms so far, now at startProcess: building log message
01-02 22:37:00.734  1617  1658 I ActivityManager: Start proc 8767:com.android.nfc/1027 for restart com.android.nfc
01-02 22:37:00.734  1617  1658 W ActivityManager: Slow operation: 60ms so far, now at startProcess: starting to update pids map
01-02 22:37:00.735  1617  1658 W ActivityManager: Slow operation: 60ms so far, now at startProcess: done updating pids map
01-02 22:37:00.765  1617  1659 I chatty  : uid=1000(system) ActivityManager identical 1 line
01-02 22:37:00.810  1617  1659 W libprocessgroup: kill(-8730, 9) failed: No such process
01-02 22:37:00.810  1617  1659 I libprocessgroup: Successfully killed process cgroup uid 1027 pid 8730 in 136ms
01-02 22:37:00.825  8767  8767 E com.android.nf: Not starting debugger since process cannot load the jdwp agent.
01-02 22:37:00.859  8767  8767 V Riru    : hook removed
01-02 22:37:00.859  8767  8767 V Riru    : edxp: forkAndSpecializePost
01-02 22:37:00.927  8767  8767 D SandHook: method <public com.android.nfc.NfcService(android.app.Application)> hook <replacement> success!
01-02 22:37:00.930  8767  8767 D SandHook: method <java.lang.String com.android.nfc.cardemulation.HostEmulationManager.findSelectAid(byte[])> hook <replacement> success!
01-02 22:37:00.934  8767  8767 D SandHook: method <public int com.android.nfc.dhimpl.NativeNfcManager.getMaxTransceiveLength(int)> hook <replacement> success!
01-02 22:37:00.940  8767  8767 D SandHook: method <public android.nfc.TransceiveResult com.android.nfc.NfcService$TagService.transceive(int,byte[],boolean) throws android.os.RemoteException> hook <replacement> success!
01-02 22:37:00.944  8767  8767 D SandHook: method <public int com.android.nfc.NfcDispatcher.dispatchTag(android.nfc.Tag)> hook <replacement> success!
01-02 22:37:00.947  8767  8767 D SandHook: method <public void com.android.nfc.cardemulation.HostEmulationManager.onHostEmulationData(byte[])> hook <replacement> success!
01-02 22:37:00.950  8767  8767 D SandHook: method <public void com.android.nfc.cardemulation.HostEmulationManager.onHostEmulationActivated()> hook <replacement> success!
01-02 22:37:00.954  8767  8767 D SandHook: method <public boolean com.android.nfc.NfcService.sendData(byte[])> hook <replacement> success!
01-02 22:37:00.986  8767  8767 I NfcService: Starting NFC service
01-02 22:37:01.011  8767  8767 I         : [0102/223701.010894:INFO:NfcJniUtil.cpp(47)] NFC Service: loading nci JNI
01-02 22:37:01.011  8767  8767 I com.android.nf: ConfigFile - Parsing file '/etc/libnfc-nci.conf'
01-02 22:37:01.011  8767  8767 I com.android.nf: ConfigFile - [NFC_DEBUG_ENABLED] = 0
01-02 22:37:01.011  8767  8767 I com.android.nf: ConfigFile - [NFA_STORAGE] = "/data/nfc"
01-02 22:37:01.012  8767  8767 I com.android.nf: ConfigFile - [NFA_HCI_DEFAULT_DEST_GATE] = 0xF0
01-02 22:37:01.012  8767  8767 I com.android.nf: ConfigFile - [NFA_HCI_STATIC_PIPE_ID_C0] = 0x19
01-02 22:37:01.012  8767  8767 I com.android.nf: ConfigFile - [UICC_LISTEN_TECH_MASK] = 0x07
01-02 22:37:01.012  8767  8767 I com.android.nf: ConfigFile - [HOST_LISTEN_ENABLE] = 0x01
01-02 22:37:01.012  8767  8767 I com.android.nf: ConfigFile - [NXP_FWD_FUNCTIONALITY_ENABLE] = 0x01
01-02 22:37:01.012  8767  8767 I com.android.nf: ConfigFile - [UICC_IDLE_TIMEOUT] = 0
01-02 22:37:01.012  8767  8767 I com.android.nf: ConfigFile - [AID_FOR_EMPTY_SELECT] = {08:A0:00:00:01:51:00:00:00}
01-02 22:37:01.012  8767  8767 I com.android.nf: ConfigFile - [SCREEN_OFF_POWER_STATE] = 1
01-02 22:37:01.012  8767  8767 I com.android.nf: ConfigFile - [MAX_RF_DATA_CREDITS] = 1
01-02 22:37:01.012  8767  8767 I com.android.nf: ConfigFile - [POLLING_TECH_MASK] = 0xEF
01-02 22:37:01.012  8767  8767 I com.android.nf: ConfigFile - [P2P_LISTEN_TECH_MASK] = 0xC5
01-02 22:37:01.013  8767  8767 I com.android.nf: ConfigFile - [PRESERVE_STORAGE] = 0x01
01-02 22:37:01.013  8767  8767 I com.android.nf: ConfigFile - [NFA_MAX_EE_SUPPORTED] = 0x03
01-02 22:37:01.013  8767  8767 I com.android.nf: ConfigFile - [NCI_HAL_MODULE] = "nfc_nci"
[... cut ...]
01-02 22:37:01.782  8798  8798 F DEBUG   :     #81 pc 0000000000255ea8  /system/lib64/libart.so (offset 0x20f000) (_ZN3art11interpreterL7ExecuteEPNS_6ThreadERKNS_20CodeItemDataAccessorERNS_11ShadowFrameENS_6JValueEb.llvm.1271626068+496)
01-02 22:37:01.782  8798  8798 F DEBUG   :     #82 pc 000000000051cb98  /system/lib64/libart.so (offset 0x445000) (artQuickToInterpreterBridge+1032)
01-02 22:37:01.782  8798  8798 F DEBUG   :     #83 pc 0000000000565cfc  /system/lib64/libart.so (offset 0x445000) (art_quick_to_interpreter_bridge+92)
01-02 22:37:01.782  8798  8798 F DEBUG   :     #84 pc 0000000000000454  /dev/ashmem/dalvik-jit-code-cache (deleted)`

roussosalex, 2020-01-03:
@matthew5025 ` Error unprotecting pages` and `trampoline size larger than symbol alignment` looks like some symbols are missing. Could you check the symbols in your `/system/lib64/libnfc-nci.so`? It should at least have `NFC_SetConfig`.

matthew5025, 2020-01-03:
@roussosalex Thanks for the reply; I've checked and NFC_SetConfig is present.

jekkos, 2020-01-03:
So maybe the error is coming from another hooked symbol? From reading the code I understood that to make this work only the NFC_SetConfig should be adapted, the other hooks can be disabled. So that could be worth a try.

matthew5025, 2020-01-03:
@jekkos Were you referring to the line in nfcd.h?

I've changed it to `#define NFCD_DEBUG false` but am still getting the same problem

jekkos, 2020-01-05:
Yeah indeed that's what I meant. Maybe something else is wrong then, maybe the SetConfig method signature was changed, or the latest Android uses a newer compiler that breaks the hooking itself? Just guessing but well probably need to have a closer look by comparing the old with the new situation.

roussosalex, 2020-01-06:
Could you try it again with the improved logging in the latest commit?

matthew5025, 2020-01-19:
@roussosalex I've just tried with the latest version of edxposed (sandhook) and lineage-16.0-20200118 and it seems to work. It seems safetynet hasn't tripped; I'll try with google pay soon and report back.

matthew5025, 2020-01-19:
@roussosalex I've just tried with the latest version of edxposed (sandhook) and lineage-16.0-20200118 and it seems to work. Loading the module trips safetynet though.

matthew5025, 2020-01-28:
@roussosalex Android pay works with edxposed and nfcgate and I've managed to sucessfully purchase something in store. However, the app crashes when pressing stop capture.

> 01-28 21:39:09.006  2890  3095 D NATIVENFC: Command: SET_CAPTURE
01-28 21:39:09.008   880   936 I ActivityManager: START u0 {act=de.tu_darmstadt.seemoo.nfcgate.capture flg=0x10000000 pkg=de.tu_darmstadt.seemoo.nfcgate cmp=de.tu_darmstadt.seemoo.nfcgate/.gui.MainActivity (has extras)} from uid 1027
01-28 21:39:09.013   572  4445 D audio_hw_primary: start_output_stream: exit
01-28 21:39:09.041  4666  4666 D AndroidRuntime: Shutting down VM
--------- beginning of crash
01-28 21:39:09.051  4666  4666 E AndroidRuntime: FATAL EXCEPTION: main
01-28 21:39:09.051  4666  4666 E AndroidRuntime: Process: de.tu_darmstadt.seemoo.nfcgate, PID: 4666
01-28 21:39:09.051  4666  4666 E AndroidRuntime: java.lang.NullPointerException: Attempt to get length of null array
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at com.google.protobuf.ByteString.copyFrom(ByteString.java:310)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at de.tu_darmstadt.seemoo.nfcgate.util.NfcComm.<init>(NfcComm.java:30)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.CaptureFragment.fromBundle(CaptureFragment.java:89)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at de.tu_darmstadt.seemoo.nfcgate.gui.MainActivity.importCapture(MainActivity.java:230)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at de.tu_darmstadt.seemoo.nfcgate.gui.MainActivity.onNewIntent(MainActivity.java:141)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at android.app.Activity.performNewIntent(Activity.java:7021)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1309)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1321)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at android.app.ActivityThread.deliverNewIntents(ActivityThread.java:2932)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at android.app.ActivityThread.performNewIntents(ActivityThread.java:2948)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at android.app.ActivityThread.handleNewIntent(ActivityThread.java:2964)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at android.app.ActivityThread.-wrap14(Unknown Source:0)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1667)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:106)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:164)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at android.app.ActivityThread.main(ActivityThread.java:6494)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:440)
01-28 21:39:09.051  4666  4666 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
01-28 21:39:09.055   572   903 I SmartAmp: Tfa9887_SetMute return value is 0.
01-28 21:39:09.061   880  2106 W ActivityManager:   Force finishing activity de.tu_darmstadt.seemoo.nfcgate/.gui.MainActivity
01-28 21:39:09.080   880   995 I ActivityManager: Showing crash dialog for package de.tu_darmstadt.seemoo.nfcgate u0
01-28 21:39:09.185   880   995 W Looper  : Dispatch took 117ms on android.ui, h=Handler (com.android.server.am.ActivityManagerService$UiHandler) {2f4be61} cb=null msg=1
01-28 21:39:09.367   880   995 W Looper  : Dispatch took 163ms on android.ui, h=Handler (android.view.Choreographer$FrameHandler) {8f24a42} cb=android.view.Choreographer$FrameDisplayEventReceiver@2d5ea53 msg=0

kleest, 2020-01-29:
Thank you for you feedback. It looks like you have found a bug with capture mode in HCE mode which is not specific to Magisk. Sorry about that, we will fix it soon. I am amazed that you managed to make a successful payment with GPay while using EdXposed! May I ask for your EdXposed configuration (YAHFA/Sanhook, AppList mode / whitelist, etc.)?

Besides that, I looked into making NFCGate compatible with Android 10. With the current master, all modes, which use native function hooking, do not work on Android 10. This applies to all HCE related modes: clone mode, tag role in relay and replay mode. Capture mode works (as soon as the bug you found is fixed ;-)). See #107 for more information about Android 10.

matthew5025, 2020-02-02:
@kleest 
LineageOS 15.1-20191201 on the Nextbit Robin
Magisk 20.3 (Magisk Hide -> Yes)
Edxposed 4463 Sandhook
Blacklist Google Play Services and Google Services Framework
Force hook modules -> Yes
Pass SafetyNet -> Yes

roussosalex, 2020-06-15:
Added with e84644f0d7cae8b3b43991ddf4eb9371688e2ede

closed 2020-06-15 by roussosalex