nfcgate/nfcgate #109: Samsung S8 and one plus 5T for relay and no response

toiron, 2020-09-08:
hello
I used Samsung S8 and one plus 5T for relay and no response. Have you ever encountered any reason?Would you please tell me what type of mobile phone you used for the test?

roussosalex, 2020-09-08:
Could you provide us with the NFCGate status page for the devices you used? For our tested devices and Android versions, see the [compatibility document](https://github.com/nfcgate/nfcgate/blob/v2/doc/Compatibility.md).

quark9981, 2020-09-13:
[image]
[image]
Replace it with two Nesux 5x, replay and relay still fail, this is a screenshot of my device.
[image]

quark9981, 2020-09-17:
The problem has been solved. Thank you for your support

closed 2020-09-17 by quark9981

Thor-Heimdallr, 2022-05-24:
> 
Cool , How did you solve it ?