nfcgate/nfcgate #114: Can't get cloning to work

Diniboy1123, 2021-08-19:
Hi,

I decided to give this project a go. I got a simple NFC tag with a static uid and firstly I tried installing nfcgate on a xiaomi mi 9t pro running android 11 Pixel Experience and lsposed. The reading of the tag worked well, however the emulation doesn't work at all.

All good, the phone is not listed as compatible, the NFC chip is also shown as unknown within the nfcgate status page, so I thought its simply not compatible. Or android 11 changed something or lsposed doesn't work.

Then I decided to get a Nexus 5X just for this which I now have with the latest Lineage 17.1 available from XDA. Then I tried using lsposed first, gave nfcgate permissions to hook into android system and the com.android.nfc package and restarted the phone. Then it could also read tags, but couldn't emulate. So I tried edxposed as this was listed as compatible. Sadly it didn't work either. The NFC chip is shown as unknown here too. I obviously tried to reboot many times. Xposed is working fine with other modules.

What am I doing wrong? Maybe I need a different rom?

roussosalex, 2021-08-19:
Please provide some more information:

1. Which type of tag are you using?
2. For the Xiaomi Mi 9T Pro and Nexus 5X: please provide the Logcat output of the boot process and unsuccessful cloning attempt.
3. How are you checking if cloning "worked"? Which reader are you using?

The Nexus 5X has been tested with Pixel Experience 10 as listed in the [compatibility document](https://github.com/nfcgate/nfcgate/blob/v2/doc/Compatibility.md), not with Lineage. The unofficial 17.1 build from XDA does not specify the type of NFC chip in the [libnfc-nxp.conf file](https://github.com/Toomoch/android_device_lge_bullhead/blob/lineage-17.1/nfc/libnfc-nxp.conf) and is therefore listed as unknown.

Diniboy1123, 2021-08-20:
It's a mifare ultralight c (type a).

For the second requirement, boot process as in the actual phone boot or just when the app boots? If the earlier, I am unsure how could I do that. adb logcat quickly in an early boot state?

To check if the cloning worked I use the other phone always, so if I used the nexus 5x to clone then I use the mi 9t pro to read. And I use nfcgate's cloning function for the reading on the reader phone.

And regarding the unknown issue, that makes sense. Thank you for the explanation.

roussosalex, 2021-08-21:
Unfortunately Android [does not usually](https://developer.android.com/reference/android/nfc/tech/MifareUltralight) support Mifare Ultralight tags for data transfer, but cloning the static tag data should still work. 

If the device is off when you issue `adb logcat` it should hang until the device is online. 

Using another phone as reader did not work well for us in the past, because Android's built-in functionality for detecting nearby devices kept interfering. We ended up using a dedicated NFC reader board with a laptop for debugging purposes.

closed 2022-02-09 by Diniboy1123