nfcgate/nfcgate #116: Not working on Xiaomi Mi 6 (with Android 9)

robur-pace, 2022-03-14:
Hi,

Sorry for my poor English (not native language).

I have two Xiaomi Mi 6 phones, both with MIUI 11 stable (Android 9).
Magisk v24.3 with modules Riru(v25.4.4) & Riru - EdXposed(v0.5.2.2) has been installed.
EdXposed Manager version is v4.6.2.
The NFCGate version is 2.1.1. On the status page, everything looks fine.

[image]

[image]

I noticed that NFCGate is not working either in clone mode or relay mode.
I mean, NFCGate can collect (read) tag information correctly (All Type-A tags).
For example, this is a MiFare Classic 1K (S50) tag.

[image]

But when I put the phone on the NFC reader, during the anti-collision/select operation, the UID has been set to 0x01020304 instead of the specified tag UID.

[image]

In relay mode, the UID is still 0x01020304. The NFC reader sent commands (APDUs), but no response to the reader.

[image]

I flashed PixelExperience 9 (PQ3A.190705.003) into the phone, and reinstalled the frameworks and apps.
Unfortunately, same problem, still not working.

Is there a compatibility issue?

Thank you.

robur-pace, 2022-03-14:
I flashed Pixel Experience 8.1.0 (OPM2.171026.006.H1) into Mi 6 phone.
Xposed-v90-sdk27 and Xposed Installer 3.15 installed.

Finally, it works. Unstable, but worked. I have to on/off NFC switch several time in settings.

Relay mode/Clone mode working correctly with DESFire EV1.

Not working with Mifare Classic 1K.

roussosalex, 2022-03-16:
Hi, thanks for the detailed description. 

Android unfortunately does not support emulating [Mifare Classic](https://stackoverflow.com/a/27425931) tags. Have you tried using a different tag type on Android 9?

From our experience, using a phone as a reader of another phone has always been unstable due to interference by the OS. In that case also try disabling _Android Beam_ or _Nearby Share_ on the devices.

If cloning still does not work, could you **attach** the logcat output from phone boot to first clone attempt (as a separate file)? 
If it does work, I would like to add it to the [compatibility document](/nfcgate/nfcgate/blob/v2/doc/Compatibility.md).

robur-pace, 2022-03-16:
Thanks for reply.

I already disabled the Android Beam.
And I tried DESFire EV1 8K, same problem (UID is 0x01020304, wrong ATQA, wrong ATS).

This is the response of DESFire EV1 Tag. (Using PM3 as a NFC reader)
[image]

NFCGate shows the correct Tag information on the screen, but...

This the response of NFCTag clone mode.
(Mi 6, MIUI 11.0.5, EdXposed V93.0)
[image]

This phone (Mi6) cannot work in clone mode nor relay mode (Tag role). But this phone can act as a Reader in relay mode, with DESFire EV1 tag.

I got a Nexus 5X yesterday, OS is stock 8.1.0 (OPM7.181205.001).
NFCGate is not working on the Nexus 5X with EdXposed V93.0 framework, same problem.
But it worked with Xposed V90-beta3 framework.

It looks like NFCGate only works with Xposed framework.
The documents said that it is compatible with EdXposed.
May I know what Magisk version and EdXposed (and modules) version you have tested?

Here is the system logcat, and EdXposed logs.
(Xiaomi Mi 6, MIUI 11.0.5)
[Log.zip](https://github.com/nfcgate/nfcgate/files/8259361/Log.zip)

Thank you very much.

robur-pace, 2022-03-16:
Even the phone acts as a TAG in relay mode, it can read (collect) tag information using its own NFC antenna.

Is that normal?

I noticed that both Mi 6 and Nexus 5X are hard to communicate with some NFC readers.

roussosalex, 2022-03-16:
Here are some interesting excerpts from the log: 

```
[03-16 13:20:45.656 3236:3818 D/NATIVENFC]
Command: SET_CONFIG

[03-16 13:20:45.657 3236:3818 I/NATIVENFC]
uploadConfig

[03-16 13:20:45.657 3236:3818 F/libc]
Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0 in tid 3818 (ht), pid 3236 (com.android.nfc)

[...]

[03-16 13:20:45.730 5473:5473 F/DEBUG]
pid: 3236, tid: 3818, name: ht  >>> com.android.nfc <<<

[03-16 13:20:45.730 5473:5473 F/DEBUG]
signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0

[03-16 13:20:45.730 5473:5473 F/DEBUG]
Cause: null pointer dereference

[03-16 13:20:45.730 5473:5473 F/DEBUG]
    x0  0000007b9600d814  x1  0000007b9600d814  x2  0000000000000001  x3  0000000000000003

[03-16 13:20:45.730 5473:5473 F/DEBUG]
    x4  0000007b9600d815  x5  0000007b9600d815  x6  0000000000000080  x7  0000000000000080

[03-16 13:20:45.730 5473:5473 F/DEBUG]
    x8  0000000000000000  x9  0000000000000000  x10 00000000000000ae  x11 00000000ffffffff

[03-16 13:20:45.730 5473:5473 F/DEBUG]
    x12 0000000000000006  x13 0000007b96027a51  x14 0000000000000000  x15 0000000000000001

[03-16 13:20:45.730 5473:5473 F/DEBUG]
    x16 0000007b8226e218  x17 0000007c21ee4ef0  x18 0000000000000008  x19 0000007b8227b030

[03-16 13:20:45.731 5473:5473 F/DEBUG]
    x20 0000007b9600d800  x21 0000007b8227b070  x22 0000007b96027a50  x23 0000000000000015

[03-16 13:20:45.731 5473:5473 F/DEBUG]
    x24 0000007b8227b030  x25 0000000000000014  x26 0000007b8edabca0  x27 0000000000000002

[03-16 13:20:45.731 5473:5473 F/DEBUG]
    x28 0000007b821f96d0  x29 0000007b821f9620

[03-16 13:20:45.731 5473:5473 F/DEBUG]
    sp  0000007b821f95f0  lr  0000007b8222c83c  pc  0000007b8222c860

[...]

[03-16 13:20:45.912 5473:5473 F/DEBUG]

backtrace:

[03-16 13:20:45.912 5473:5473 F/DEBUG]
    #00 pc 0000000000031860  /data/app/de.tu_darmstadt.seemoo.nfcgate-3MwVv30vlL5SQ7n_royDJw==/lib/arm64/libnfcgate.so (offset 0x30000) (uploadConfig(Config&)+284)

[03-16 13:20:45.912 5473:5473 F/DEBUG]
    #01 pc 0000000000031a5c  /data/app/de.tu_darmstadt.seemoo.nfcgate-3MwVv30vlL5SQ7n_royDJw==/lib/arm64/libnfcgate.so (offset 0x30000) (Java_de_tu_1darmstadt_seemoo_nfcgate_xposed_Native_setConfiguration+360)

[03-16 13:20:45.912 5473:5473 F/DEBUG]
    #02 pc 00000000005667e0  /system/lib64/libart.so (offset 0x43f000) (art_quick_generic_jni_trampoline+144)

[03-16 13:20:45.912 5473:5473 F/DEBUG]
    #03 pc 000000000055d788  /system/lib64/libart.so (offset 0x43f000) (art_quick_invoke_stub+584)

[03-16 13:20:45.912 5473:5473 F/DEBUG]
    #04 pc 00000000000d074c  /system/lib64/libart.so (art::ArtMethod::Invoke(art::Thread*, unsigned int*, unsigned int, art::JValue*, char const*)+200)
[... 36 lines cut ...]

[03-16 13:20:45.912 5473:5473 F/DEBUG]
    #17 pc 00000000017b6fa4  /system/framework/arm64/boot-framework.oat (offset 0xa50000) (android.os.HandlerThread.run+692)

[03-16 13:20:45.912 5473:5473 F/DEBUG]
    #18 pc 000000000055d788  /system/lib64/libart.so (offset 0x43f000) (art_quick_invoke_stub+584)

[03-16 13:20:45.912 5473:5473 F/DEBUG]
    #19 pc 00000000000d074c  /system/lib64/libart.so (art::ArtMethod::Invoke(art::Thread*, unsigned int*, unsigned int, art::JValue*, char const*)+200)

[03-16 13:20:45.912 5473:5473 F/DEBUG]
    #20 pc 000000000045f850  /system/lib64/libart.so (offset 0x376000) (art::(anonymous namespace)::InvokeWithArgArray(art::ScopedObjectAccessAlreadyRunnable const&, art::ArtMethod*, art::(anonymous namespace)::ArgArray*, art::JValue*, char const*)+104)

[03-16 13:20:45.912 5473:5473 F/DEBUG]
    #21 pc 000000000046090c  /system/lib64/libart.so (offset 0x376000) (art::InvokeVirtualOrInterfaceWithJValues(art::ScopedObjectAccessAlreadyRunnable const&, _jobject*, _jmethodID*, jvalue*)+424)

[03-16 13:20:45.912 5473:5473 F/DEBUG]
    #22 pc 000000000048ed5c  /system/lib64/libart.so (offset 0x43f000) (art::Thread::CreateCallback(void*)+1092)

[03-16 13:20:45.912 5473:5473 F/DEBUG]
    #23 pc 00000000000901b0  /system/lib64/libc.so (__pthread_start(void*)+36)

[03-16 13:20:45.912 5473:5473 F/DEBUG]
    #24 pc 0000000000023930  /system/lib64/libc.so (__start_thread+68)
```

roussosalex, 2022-03-17:
So I narrowed it down to the following problem:

```
[03-16 13:20:46.784 5484:5484 I/NATIVENFC]
Library expected at /system/lib64/libnfc-nci.so

[03-16 13:20:46.784 5484:5484 E/NATIVENFC]
Library not accessible
```
It looks like the native hook can not find the library. Can you confirm that for the Mi 6 _PixelExperience 8.1.0_ the `libnfc-nci.so` is at `/system/lib64/libnfc-nci.so` but for _PixelExperience 9_ and _MIUI 11 stable_ it is somewhere else?

roussosalex, 2022-03-17:
> Even the phone acts as a TAG in relay mode, it can read (collect) tag information using its own NFC antenna.
>
>Is that normal?

Not usually, because polling ("collecting") _should be_ disabled while emulating a tag. If disabling polling does not work reliably, you could try experimenting with the code in [jni.cpp](/nfcgate/nfcgate/blob/v2/nfcd/src/main/jni/src/jni.cpp#L4)

> I noticed that both Mi 6 and Nexus 5X are hard to communicate with some NFC readers.

Could you attach another logcat output from the boot, including failed communication attempts at least one successful attempt?

robur-pace, 2022-03-17:
I think I got what is going on.

Most of the problems mentioned above, caused by Android OS version and/or EdXposde framework.

MIUI 11.0.5 stable for Mi 6 MISSING libnfc-nci.so at /system/lib64/.
PixelExperience 9 for Mi 6 has libnfc-nci.so at /system/lib64/.

I tested several combinations. NFCGate v2.1.1 WORKS with:
[image]

Sometimes, everything looks GOOD in NFCGate STATUS page, but the app works abnormally.
Run EdXposed Manager, click Modules, disable the NFCGate then reboot. After rebooting, re-enable NFCGate in EdXposed Manager, then reboot again. 
After this, NFCGate should works fine. (So I think there's a little issue within EdXposed framework.)

robur-pace, 2022-03-17:
> Not usually, because polling ("collecting") _should be_ disabled while emulating a tag. If disabling polling does not work reliably, you could try experimenting with the code in [...]

I checked the source code in 'CloneMode.java', I noticed that NFCGate should immediately disable clone mode to avoid cloning same tag again.
In Clone Mode, NFCGate keep trying to polling (collecting) even it already has tag information. This may caused by EdXposed framework. Disable / re-enable NFCGate in EdXposed Manager, then rebooting again-and-again, NFCGate should works correctly in Clone Mode. (At least it works for me.)

The abnormal "polling" in Tag emulation (Relay Mode - Tag role), caused by the source code in 'RelayMode.java'.
At onEnable(), it always enable polling, whether in "Reader" mode or not.
At onDisable(), it forget to disable polling in 

> > I noticed that both Mi 6 and Nexus 5X are hard to communicate with some NFC readers.
> 
> Could you attach another logcat output from the boot, including failed communication attempts at least one successful attempt?

I think this may caused by abnormal "polling" in TAG emulation. Both phone and NFC reader generates RF-field energy, may result interference.
NFCGate works normally, no more communication difficulty.

robur-pace, 2022-03-17:
I modified the code, and tested on my phones. It works fine.

    @Override
    public void onEnable() {
        if (mReader) {
            // look for a tag
			mManager.enablePolling();
		}
		else {
			// no-op
		}
        // mManager.enablePolling();

        // enable or disable reader mode
        mManager.setReaderMode(mReader);

        // connect to the network
        if (mOnline)
            mManager.getNetwork().connect();
    }

    @Override
    public void onDisable() {
        // disable Polling in Reader Mode
        if (mReader) {
            mManager.disablePolling();
        }

        // disable reader mode
        mManager.setReaderMode(false);

        // disconnect from the network
        if (mOnline)
            mManager.getNetwork().disconnect();
    }

roussosalex, 2022-03-17:
> MIUI 11.0.5 stable for Mi 6 MISSING libnfc-nci.so at /system/lib64/.

Can you find out where the library is located instead on MIUI 11.0.5? 

> Sometimes, everything looks GOOD in NFCGate STATUS page, but the app works abnormally.

Yes, we should improve the status page to include errors with native hooking in "hook status"

> In Clone Mode, NFCGate keep trying to polling (collecting) even it already has tag information. This may caused by EdXposed framework.

I think the [enableDisablePolling(bool)](https://github.com/nfcgate/nfcgate/blob/v2/nfcd/src/main/jni/src/jni.cpp#L4) code sometimes fails to disable polling properly.

> I modified the code, and tested on my phones. It works fine.

I will look into the changes.

roussosalex, 2022-03-18:
Can you test if the PR properly shows native hook working in status page on the Mi 6 with PixelExperience 9, and also shows the hook not working on MIUI 11.0.5?

I also added your changes to disable polling while relaying in tag mode.

robur-pace, 2022-03-18:
> Can you find out where the library is located instead on MIUI 11.0.5?
File list of MIUI 11.0.5.
[image]

File list of PixelExp 9.
[image]

robur-pace, 2022-03-18:
MIUI 11.0.5
[image]

PixelExp 9
Always not working. Even if I turned on / off the switch in EdXposed Manager, and reboot for several times.
[image]

robur-pace, 2022-03-18:
In MIUI 11.0.5, when a tag is at the proximity of the phone, a window pop-up, let user choose an application to use -- NFCGate, and built-in Xiaomi-SmartCard & Xiaomi-Wallet.

So I think nfcgate working on MIUI 11 is NOT a good idea.

robur-pace, 2022-03-19:
> Looks like the library was renamed for the NQ generation of NXP NFC chips. If you adjust [this path](https://github.com/nfcgate/nfcgate/blob/v2/nfcd/src/main/jni/include/nfcd/nfcd.h#L37) to libnqnfc-nci.so does it change anything in MIUI 11? If not, we would have to look at the differences with the new libnfc-nci.

I modified nfcd.h at v2 source code. It worked in clone mode.

> If NFCGate is closed, this behavior is expected. If it is open and waiting for a tag, this should not happen.

YES. NFCGate worked with the modified code, no pop-up window any more.

Thank you.

robur-pace, 2022-03-19:
Two phones run NFCGate, both in clone mode. The 1st one emulates a Tag, the 2nd one acts as reader.
The 2nd one can not collect the 1st one's tag information. The 2nd one always shows "Waiting for Tag...".
(Both Mi6 and Nexus 5X were tested.)

But a phone in clone mode (as a reader) can collect ApplePay (iPhone 13) tag information.

I don't know what is the difference?

roussosalex, 2022-03-22:
> YES. NFCGate worked with the modified code

Great, we will improve the search logic for the nfc-nci library

> The 2nd one hard to collect the 1st one's tag information

I am not sure why it is more difficult. If you try to clone the other phone without emulating anything, does it work more easily?

robur-pace, 2022-03-22:
> I am not sure why it is more difficult. If you try to clone the other phone without emulating anything on it, does it work more easily?

I made several tests, and I guess it is hardware compatibility issue (antenna or  something). Not software problem.

Thank you!

closed 2022-03-22 by robur-pace

roussosalex, 2022-04-05:
@robur-pace We have improved some things and will be publishing a new minor version. Could you test it with your devices, please? See #119

robur-pace, 2022-04-06:
> @robur-pace We have improved some things and will be publishing a new minor version soon. Could you test it with your devices, please? See #119

Thanks for your hard working on this.

I made several tests, and I have noticed:
1. Everything on status page is correct. The phones can collect the Tag Info in clone mode;
2. ACR122U reader cannot detect the phone which in clone mode, nor relay mode (as Tag). So this version not working at all;
3. MIUI 11, 'Native hook enabled' always says 'No'.

roussosalex, 2022-06-17:
@robur-pace We have improved #119 and used test hardware to confirm it working. Could you test it again?

robur-pace, 2022-07-02:
> @robur-pace We have improved #119 and used test hardware to confirm it working. Could you test it again?

@roussosalex 

I'm sorry, I just got this message.

I have tested this (https://github.com/nfcgate/nfcgate/tree/v2-dev) early today.

It worked fine on MI6 phone with PixelExperience OS.
But with MIUI, it cannot emulate TAG in RELAY mode. And it cannot work in CLONE mode.
It always says 'Xposed module is not working properly'.

You can find the log file and screenshots in the zip file. 

[logcat_07-02-2022_21-18-27.zip](https://github.com/nfcgate/nfcgate/files/9033100/logcat_07-02-2022_21-18-27.zip)

kleest, 2022-08-09:
@robur-pace We missed a special case with libnqnfc-nci.so on MIUI. Could you please test it again on MIUI?

robur-pace, 2022-08-14:
> @robur-pace We missed a special case with libnqnfc-nci.so on MIUI. Could you please test it again on MIUI?

@kleest I tested it on MIUI.
Everything on STATUS page is OK. Clone-Mode & Relay-Mode worked fine.

Thank you!

kleest, 2022-08-23:
Thank you @robur-pace for your input! We just a released a new version containing these fixes (and more) from #119.