nfcgate/nfcgate #118: Need CID (Card IDentifier) support

robur-pace, 2022-03-19:
I noticed that NFCGate in Relay Mode (as a Tag) doesn't work with some type of NFC readers. (Of course, it works with ACR122U.)
NFCGate doesn't send any APDU to remote phone (as a reader), and NO APDU show up on the screen.

I made several tests, and used Proxmark3 to capture data over the air.
May this issue caused by CID (Card Identifier) supporting of NFCGate.

The PCD (reader) sends RATS to Tag (or phone), with a specified CID. (Eg. 0xe0  21  b2  c7)
Then NFCGate sends ATS to PCD with TC1 = 0x00, which indicates NO CID supporting. (Eg. 0x10  78  80  70  _[00]_  20  90  00  00  00  00  00 69  1c  f1  b4  5d  83)
(By the way, the PHYSICAL DESFire EV1 or Chinese FM1208 CPU card generates ATS with TC1 = 0x02, means supporting CID.)

Some type of NFC readers always do NOT use CID in data blocks, such as ACR122U.

But, some else do NOT follow the protocol. They always send blocks containing CID to Tag, even if Tag already set TC1 = 0x00 in ATS.
(Eg. 0x0a  01  00  a4  00  00  02  10  02  56 68, CID = 0x1)

According to the 14443-4 protocol, Tag which doesn't support CID, must ignore any block containing CID.
Therefore NFCGate just ignore APDUs (blocks) containing CID.

Could you improve the CID supporting?

robur-pace, 2022-03-19:
113637872 |  113638864 | Rdr |52(7)                                  |     | WUPA
  114662912 |  114663904 | Rdr |52(7)                                  |     | WUPA
  114665156 |  114667524 | Tag |04  00                         |     |
  114691328 |  114693792 | Rdr |93  20                                  |     | ANTICOLL
  114694980 |  114700804 | Tag |69  1c  f1  b4  30                         |     |
  114739712 |  114750240 | Rdr |93  70  69  1c  f1  b4  30  bb  c7                                  |  ok | SELECT_UID
  114751412 |  114754996 | Tag |20  fc  70                         |     |
  114775792 |  114780560 | Rdr |e0  21  b2  c7                                  |  ok | RATS
  114784820 |  114805684 | Tag |10  78  80  71  00  20  90  00  00  00  00  00  69  1c  f1  b4  f0  86   |  ok |
  114840432 |  114846288 | Rdr |d1  11  00  8e  fc                                  |  ok |
  114874804 |  114878324 | Tag |d1  fa  96                         |     |
  114918512 |  114931344 | Rdr |0a  01  00  a4  00  00  02  10  02  56  68                               |  ok |

Reader waited for response about 280ms, got nothing. Then restart the process, sent WUPA.
(118,703,344-114,931,344)/13,560,000 = 0.278 second

  118703344 |  118704336 | Rdr |52(7)                                  |     | WUPA
  118705588 |  118707956 | Tag |04  00                         |     |
  118731760 |  118734224 | Rdr |93  20                                  |     | ANTICOLL

roussosalex, 2022-03-22:
The low-level handling of the 14443-4 protocol is not a part of NFCGate. It is mostly part of the Android NFC Stack and the NFCC firmware. As far as I can tell, all blocks (which may or may not contain the CID field) are handled by the firmware. There are several configuration options for the firmware, but none to control the CID handling.

closed 2022-03-24 by robur-pace