nfcgate/nfcgate #122: Rooted Moto G41 / android 11 Class not found

CardMaster2022, 2022-11-26:
Hi ☺️
I'm having this issue on my Moto G41 and nfcgate 

[ 2022-11-25T21:23:02.054     1027:  3370:  3370 E/LSPosed-Bridge  ] de.robv.android.xposed.XposedHelpers$ClassNotFoundError: java.lang.ClassNotFoundException: com.android.nfc.dhimpl.NativeNfcManager
    at de.robv.android.xposed.XposedHelpers.findClass(Unknown Source:12)
    at de.robv.android.xposed.XposedHelpers.findAndHookMethod(Unknown Source:0)
    at de.tu_darmstadt.seemoo.nfcgate.xposed.Hooks.handleLoadPackage(Hooks.java:64)
    at de.robv.android.xposed.IXposedHookLoadPackage$Wrapper.handleLoadPackage(Unknown Source:2)
    at de.robv.android.xposed.callbacks.XC_LoadPackage.call(Unknown Source:6)
    at de.robv.android.xposed.callbacks.XCallback.callAll(Unknown Source:26)
    at K.afterHookedMethod(Unknown Source:202)
    at de.robv.android.xposed.XposedBridge$AdditionalHookInfo.callback(Unknown Source:147)
    at LSPHooker_.getClassLoader(Unknown Source:8)
    at android.app.LoadedApk.getResources(LoadedApk.java:1222)
    at android.app.ContextImpl.createAppContext(ContextImpl.java:2817)
    at android.app.ContextImpl.createAppContext(ContextImpl.java:2809)
    at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6760)
    at android.app.ActivityThread.access$1500(ActivityThread.java:252)
    at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1999)
    at android.os.Handler.dispatchMessage(Handler.java:106)
    at android.os.Looper.loop(Looper.java:250)
    at android.app.ActivityThread.main(ActivityThread.java:7886)
    at java.lang.reflect.Method.invoke(Native Method)
    at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:970)
Caused by: java.lang.ClassNotFoundException: com.android.nfc.dhimpl.NativeNfcManager
    at java.lang.Class.classForName(Native Method)
    at java.lang.Class.forName(Class.java:454)
    at o.a(Unknown Source:20)
    at de.robv.android.xposed.XposedHelpers.findClass(Unknown Source:4)
    ... 20 more
Caused by: java.lang.ClassNotFoundException: com.android.nfc.dhimpl.NativeNfcManager
    ... 24 more

roussosalex, 2023-01-21:
Hi, can you provide some details like the Android version,  which build number and which version of NFCGate you are using? All of this information can be found on the "status" page in NFCGate, which you could also just provide a screenshot of.

XxTHKxX, 2023-10-02:
I'm having the same issue on a Redmi Note 12 Pro

Android Version: 13
Build Number: TP1A.220624.014
NFC/HCE/XposedModule enabled: Yes
NFC Chip: ST Device nfc_nci.st21nfc

Native Hook Enabled: No

When I'm in any mode it also says that the Xposed module is not working properly, although I cannot seem to find any setting that are interfering with this functionality

In addition, when I try to scan any tag, seems as if the device kept scanning it over and over and over again repeatedly, it yield no result however. I can provide extra logging as needed

kleest, 2023-10-03:
@XxTHKxX This could be a different issue. It says, that your device can successfully hook the Java classes with Xposed ("XPosed module enabled: Yes"), but it cannot hook native function ("Native Hook Enabled: No").

Did you use [NFCGate version 2.3.0](https://github.com/nfcgate/nfcgate/releases/tag/v2.3.0) which adds support for ST chipsets? If so, please create a new issue and attach a logcat since boot (see instructions when creating a new issue). Thanks!

roussosalex, 2024-01-09:
Closing due to inactivity

closed 2024-01-09 by roussosalex