nfcgate/nfcgate #123: warnning: xposed modules is not working properly

BirdNum, 2023-02-13:
Hi:
  While the nfcgae app is running, the top dock throws: warnning: xposed modules is not working properly.
  I didn't use the way of building from source to compile the application, but directly used the released APK provided by you.
  I will be very grateful if you can give me some suggestions, thans in advance!
[image]

BirdNum, 2023-02-13:
The warning info can be seen
[image]
 from the attached file.

kleest, 2023-02-14:
It looks like the Xposed module is not active. This might have the following reasons:
1. Module not enabled in module manager, including missing reboot or NFC service restart after activation
2. There was an error loading the module (invalid hooks, offset, incompatibilities, etc.)

If you can rule out 1., you can upload (or send by mail) an adb logcat. This logcat must be captured shortly after reboot so that it contains the whole boot sequence. Without the logcat, I cannot do much at this point.

k0mraid3, 2023-06-24:
generally same issue, except on status it only shows NAtive Hook has issues, but says Xposed issue in Clone

kleest, 2023-06-25:
Same response: Without logs or a screenshot from the status page we sadly cannot diagnose this issue or resolve it.

k0mraid3, 2023-06-25:
> Same response: Without logs or a screenshot from the status page we sadly cannot diagnose this issue or resolve it.

Working on this currently, ill try to get some logs from my side in the next few hours. I think it may have to do with SAR vrs SAS to be fair. Do you have a way i can message you and work with you directly on sorting this?

kleest, 2023-06-28:
@k0mraid3 I received your mail, thanks. Just for the public record: Personal contact details are available in the readme. :-)

closed 2023-08-23 by roussosalex