nfcgate/nfcgate #124: ChipToNFC: Relay & Sniffer

NFCrap, 2023-02-14:
- **If we think for a second about the possible technologies in a normal credit card, we know that we have three of them: Mag-stripe, Chip & NFC.**

In this article, we will focus on Chip and NFC data. The idea of this post comes from an article that I wrote a year ago for Troopers; While researching about relays, I found that it was possible to extract Chip data with NFC technology. It is possible because both technologies shared the same application layer(APDU).

**

**This means that the EMV Chip Card 💳 does not need to have built-in NFC  technology to be read it using NFC commands**

** for a third device.[NFCrap]  @kleest @malexmave @UweM @DanielAW @haxxproxx 
[image]

Any spare time to add USB CHIP SUPPORT TO NFCGATE 📞IN 2023 📢  Heroiam slava!

https://salmg.net/2019/02/15/chiptonfc-local-relay-sniffer/

NFCrap, 2023-02-14:
PS. Using 2 📱  📱 with USB OTG can make NFCGATE act as Chip to Chip EMULATOR USING NFC TECHNOLOGY) obviously adding the USB compatibility to NFCGATE  first.

[image]

kleest, 2023-02-14:
Hi, it's true that contact-based and contact-less payments share the same APDU Protocol (ISO 7816-4) which makes it quite interesting for research. However, the intention behind NFCGate was to exploit the wireless characteristics of NFC and its properties (e.g. NFCID) since NFC is supported out-of-the-box on most Android smartphones. There is no intention to add support for USB-based readers (contact-based or contact-less) since this would require re-implementing the whole NFC or other stacks within the application because Android does not support USB card readers. You were better off using these card-readers on a regular computer.
If you plan to relay contact-based card traffic to a contact-less terminal (or the way around), you could have a look at the NFCGate server code, which allows to insert or modify traffic in-flight. In addition, you can implement a client that runs on a computer and talks with the USB card reader. This client then connects to the server and acts like an endpoint by transmitting APDUs to/from the card reader.

NFCrap, 2023-02-16:
Thanks for clarifying this for me. 👏

Ps. Just a thought can NFCGATE be made to recognize NFC on windows subsystem for android? Possibly using virtual drivers?!

NFCrap, 2023-02-16:
Looks this guy did just that 🙈 silly me .

On Android, where a traditional PC/SC framework is not available, you can use our framework to make your real contact-less smart accessible through PKCS#11. For example, an email signing application can use the PKCS#11 interface of OpenSC, which is linked against our PC/SC implementation. Then an Android App (e.g. [Remote Smart Card Reader](https://frankmorgner.github.io/vsmartcard/remote-reader/README.html#remote-reader)) can connect as vpicc delegating all requests and responses via NFC to a contact-less smart card that signs the mail.

https://frankmorgner.github.io/vsmartcard/virtualsmartcard/README.html#vicc-install

Ps.LOL another though can NFCGATE be made to communicate with : https://frankmorgner.github.io/vsmartcard/remote-reader/README.html#remote-reader

Thank you

NFCrap, 2023-02-16:
In my view NFCGATE is still the best toolkit so why should not have ability to analyze partial chiptonfc traffic 🚦.  Must be another easy way 🤣🤣🤣 

Any advice other than before will be much appreciated. Thanks 🙏

NFCrap, 2023-02-17:
> Hi, it's true that contact-based and contact-less payments share the same APDU Protocol (ISO 7816-4) which makes it quite interesting for research. However, the intention behind NFCGate was to [...]

Smart Card toolkit:  https://m.apkpure.com/smart-card-toolkit/sasc.android.smartcard

This app can use external otg USB smard Card readers and the NFC chip in the phone 📱 , I test this back in 2015.

“Talk is cheap. Show me the code.” ― Linus Torvalds

NFCrap, 2023-02-17:
@kleest  will the PC client have to implement protobuff when connecting to server?

Thank you

kleest, 2023-02-17:
Please note that PKCS#11 is only a standard for doing cryptographic operations on cryptographic objects (like RSA keys or X.509 certificates). Without knowing any specifics, this is only a subset of what can be implemented on top of ISO 7816-4.

The app you linked most likely implements USB CCID (cf. [this SO thread](https://stackoverflow.com/q/40975868)).

Yes, the server <-> client communication uses protobuf. The .proto files are available in the repository.

NFCrap, 2023-02-17:
💕

closed 2023-02-19 by NFCrap