nfcgate/nfcgate #125: NFCGATE on WearOS  NFC(HCE) SUPPORT

WearOS-NFC, 2023-03-12:
WearOS smartwatches with NFC only support NFC payment (Card emulation mode), not NFC tag reading (Reader/Writer mode) . Is this true?
I've heard this is an WearOS issue that Google need to allow Reader/Writer mode from OS level before I can develop a smartwatch NFC tag app. Is this true? I see Android smartphones all support NFC tag reading/writing (Reader/Writer mode), either natively or via app and Android OS open to NFC app development through API and device.

This will be a real roadblock if it is true that WearOS smartwatch can't read/write NFC tags due to WearOS restriction.

PS. I managed to open nfcgate but with nfc closed. When I toggle on nfc , nfcgate crash. I heard manifest is incorrect for WearOS???? Strange thing smart card emulator https://frankmorgner.github.io/vsmartcard/ACardEmulator/README.html  works with nfc toggled on. ( did not test actual functionality) . Last thing is possible root on samsung galaxy watch 4? Any help will be much appreciated.  🙌 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

WearOS-NFC, 2023-03-12:
@kleest @malexmave @DanielAW  @UweM 

ANY SUPPORT ON THIS ISSUE??

KEEP UP THE GOOD WORK BLESSINGS

kleest, 2023-03-13:
NFCGate depends on the NFC capabilities and APIs that are provided by the operating system. So, using wearOS as a reader device might not be possible due to hardware limitations (cf. https://issuetracker.google.com/issues/213599626). If HCE APIs are available for "regular" apps (not only the system-provided payment apps), using NFCGate on a wearOS device as tag in relay or replay mode should be possible. However, for emulation of static tag data, root access and an Xposed-compatible hooking framework is required, which is not that common on wearOS devices.

WearOS-NFC, 2023-03-14:
Too bad they limit the device 😪,  thanks for clarification.

closed 2023-03-14 by WearOS-NFC

Flipo82, 2024-09-26:
Hi @kleest , THIS GUY USES GW4 AND GW6 TO EMULATE A TESLA KEYFOB.
https://github.com/pccr10001/TeslaWearKey
I think issue have been fix under the radar and if you have the right AndroidManifest.xml( check out above link) apps should now work.???
Is this correct ? I try now 2.4.2 on gw4 and it only opens when nfc is turned off but after it opens you can toggle nfc on again and app show green except xposed with is normal as it is not rooted yet ( https://xdaforums.com/t/rooting-galaxy-watch4.4452343/page-3 ),when i try to relay app crash and also clone mode dont detect nfc card this is expected as wearos limitation for reading. Is this to do with <uses-feature android:name="android.hardware.type.watch"/> or android:name="android.hardware.nfc.hce" . Thanks

kleest, 2024-10-08:
@Flipo82 HCE was never an issue, not all wearOS devices can read tags. NFCGate needs changes to support such devices.