nfcgate/nfcgate #126: libnfc_nci error

CrazyChaoz, 2023-06-28:
When trying to apply the read config options to clone a card on my Google Pixel 6a, Stock Android 13, STMicro ST21NFC Chip, I get a libnfc_nci error. The same happens on an old Samsung Galaxy S6, LineageOS 18, with the S3FWRN5 Chip.

The Status Page on both Phones shows full green checkmarks, aside from the Android Version (12 & 13) and the "Unknown" NFC Chip. 

In my logfile, several NFC_SetConfig Option log entries from https://github.com/nfcgate/nfcgate/blob/v2/nfcd/src/main/jni/src/nfcd.cpp#L136 are printed, followed by "NFA received unexpected NFC_SET_CONFIG_REVT" from https://android.googlesource.com/platform/external/libnfc-nci/+/refs/heads/master/src/nfa/dm/nfa_dm_act.c (Line 319)

Do you have any idea why this issue appears, apart from the obvious non-standard implementation, and if it is possible to fix it?
Is it possible that the ST21NFC cannot set (some) config parameters at all?
<br><br>
I would be willing to run some test code if needed/necessary.

kleest, 2023-06-28:
Thanks for the details. Yes, there might be a problem with the config stream or struct offsets, especially since we have not tested the app with chipsets from this manufacturer. However, we also sometimes got this error message and it did not cause issues.
To understand this correctly: Cloning seems to work (no app crashes) but your reader device does not read the cloned card details and the log shows the error message?

Could you send me a full logcat right after booting the device and doing a clone so that log messages from NFC service hooking are still there? You can also send this via email (see readme).

CrazyChaoz, 2023-06-29:
Update

After multiple System restarts it works on my Pixel 6a, right now it disconnected/reconnected to our scanner multiple times repeatedly, which is probably a bug somewhere ... (in the log there are multiple libnfc_nci errors _[ERROR:nfa_dm_discover.cc(1947)] Unexpected discovery event_ ), but on NFCA the ID is always the correct, static one.

After that initial issue and after restarting the Phone i can now use the cloned card in a stable manner.

closed 2023-06-29 by CrazyChaoz

kleest, 2023-06-30:
Good to hear! :slightly_smiling_face: