nfcgate/nfcgate #127: [Feature Requests]

CrazyChaoz, 2023-06-29:
- split reading and writing mode, maybe with a button in the menu 
- my device has multiple potential NFC interfaces, maybe add an option to switch between them when reading/writing
- add an option/button to clean up/deactivate the funtionality, as right now deactivation of the xposed module + restart is required to regain mobile banking functionality 
- add Android versions at least up to 13 as supported
- add Google Pixel 6a to list of (somewhat) compatible devices, at least in clone mode for NFC A

roussosalex, 2023-08-23:
1. Not sure what you mean by read or write mode. Could you elaborate what you mean?
2. We haven't seen devices with multiple NFC interfaces so far and Android has no built-in way (that we know of) to switch between multiple interfaces. Could you provide some details how your device has multiple interfaces, how they are connected and how you switch between them in Android?
3. Deactivating the functionality such that mobile banking works again (e.g. SafetyNet) requires turning off Xposed injection of the nfc system service and restarting the process. This is not in the scope of NFCGate.
4. What configuration do you mean? There is a status page for the most commonly viewed information.
5. / 6. Expanding the compatibility list requires providing the details (i.e. build number, Android version, ROM name etc.), ideally via screenshot of the status page.

CrazyChaoz, 2023-08-23:
1. Thats a bit of user-error. I wanted a way to either try to read or to write, as a sensor I tried got confused when my phone tried to read a config when it should have written.

2. I didn't mean multiple NFC chips, but instead NFC A,B and F, which should be present on most modern NFC-able phones. At least from my testing it looked like all of them where enabled while your app/module was running, as most terminals/sensors detected multiple "tags".
I also wanted a split when reading, because what tag was read when multiple (NFC A, NFC B) tags were present, such as when the app was running on a second phone, the detected tag (A or B) depended on timing/randomness.
I sadly have no clue what the underlying APIs do (yet), so I am of no help in that regard.

3. Do you kill any functionality? Otherwise you could just implement a lockfile and a dumb 
```
if(new File("LOCKFILEPATH").exists()){
do_nfcgate_functionality()
}else{
do_default_functionality()
}
```

4. The current data that should get written by the NFC A, B and F chip, but thats probably a lot to implement

5. 
- Google Pixel 6a
- Android 13
- Build: TQ3A.230605.010 (rooted stock ROM, June 5th 2023)
- HCE: yes
- NFC Chip: "Unknown" / I cannot find the concrete chip model

roussosalex, 2023-08-29:
1. NFCGate only supports ISO 7816, which is a bidirectional application-layer protocol and static tag emulation, so I am not sure what you mean by reading and writing. 
2. Sounds reasonable, please create a separate feature request for enabling/disabling different NFC technologies.
3. SafetyNet just does not work when any injection framework is used, and banking apps stop working when SafetyNet is tripped. There is nothing NFCGate can do about this, you have to disable the module and restart the service or the entire device.
4. Every time NFCGate emulates tag data, it is displayed in the UI. Which configuration do you mean?
5. We will add this data to the compatibility matrix, thanks! Your chip is probably the ST54J.

CrazyChaoz, 2023-08-29:
3. Speaking from current personal experience, this is simply not true. Spoofing the SafetyNet/Play Integrity Basic is easily done. I, for example, need certain XPosed modules for certain apps, but the other apps think they run on a pristine device, including my banking app. 
I have not yet looked if your code breaks anything, but simply having XPosed/LSPosed running doesn't kill my banking app (thanks Zygisk).

4. 1. The current config that is on NFC-A, B, F chip of the phone
   2. (Maybe impossible) if multiple "cards" are detected, all the currently "readable" configs, I have NO clue if you can switch between A, B, F and/or different cards on Android

kleest, 2023-08-30:
3. You are perfectly right that spoofing SafetyNet is possible. The difference between NFCGate and hooking certain apps is, that NFCGate hooks the com.android.nfc system service. But NFCGate should not interfere with other apps using NFC, if NFCGate is not in tag mode. The hooks are always active, however, they only alter the NFC service's behavior if NFCGate requires it ([see AID routing here](https://github.com/nfcgate/nfcgate/blob/882c1a296f8e9653b90845fa73551c6a9ce2ae0e/nfcd/src/main/java/de/tu_darmstadt/seemoo/nfcgate/xposed/Hooks.java#L48-L60)).
Do you know the reason why your banking apps stop working: Is it SafetyNet or something else? If SafetyNet is not the issue, I can imagine that emulating some static tag data may result in a payment terminal not detecting your phone and not starting a transaction, hence, the banking app stops working. Restarting NFC should be enough to fix this.

CrazyChaoz, 2023-08-31:
I am not entirely sure, but payment terminals usually complained about "too many cards present", so i think its either multiple NFC-A tags "active" at the same time (less likely) or the banking app and NFCGate keeping multiple interfaces (NFC-A, B) active and the stupid payment terminals being unable to differentiate (more likely).

I always had to disable the module + restart the service (i usually restarted the phone for good measure) to regain the ability to pay.

I have not yet looked what cards are "active" during that time (payment with XPosed module running vs. without)

kleest, 2023-09-12:
We created a new pre-release ([v2.3.0-rc](https://github.com/nfcgate/nfcgate/releases/tag/v2.3.0-rc)) that ...
1. limits listening to the emulated tag technology.
2. should be a lot more stable.
3. should be able to detect your NFC chipset on the status page.

 
Furthermore, your compatibility information is now part of the compatibility matrix.

We would appreciate if you could give this version a test and provide some feedback!

roussosalex, 2024-01-09:
Closing due to inactivity

closed 2024-01-09 by roussosalex