nfcgate/nfcgate #131: Warning: Xposed module is not working properly

ryu0q, 2023-09-04:
But the module is enabled.
Device: Samsung S8+ with custom Android 10 OneUI 2.5 mod
LSPosed info:
```
API version
100

Xposed API call protection
Enabled

Dex Optimizer Wrapper
Unsupported (SELinux is permissive)

Framework version
1.9.1 (6990)

Manager package name
com.android.shell

System version
10 (API 29)

Device
Samsung SM-N960N 

System ABI
arm64-v8a
```
[image]
[image]
[image]
[image]
[image]
[image]

ADB logcat after boot:
[logcat.txt](https://github.com/nfcgate/nfcgate/files/12512146/logcat.txt)

roussosalex, 2023-09-13:
Thanks for reporting this issue. We were slightly confused by your device identifying as a Samsung Galaxy Note 9 when you said it was a Samsung Galaxy S8+, which seems to be due to the OneUI mod ROM. We have attached an experimental "Samsung" version based on bea61ddb688b4a1f2961dacddaa8f88c7d3f096c. It allows to delay the hooking process, which seems to be the issue with your device.

Please give it a try and report back if it works and if it is now able to detect your NFC chip on the status page.
[nfcgate-samsung-debug.zip](https://github.com/nfcgate/nfcgate/files/12591896/nfcgate-samsung-debug.zip)

ryu0q, 2023-09-13:
It detects the LSposed module now, but the Native Hook still returns 'No'
[image]

Gabboxl, 2023-09-13:
Hi, I am running the same OneUI mod ROM on my Galaxy S8 and in fact, my device is identified as a Galaxy Note 9 from other apps also. 
I'm experiencing the same problem with the Native hook.

kleest, 2023-09-13:
Thanks for testing! Could you attach a logcat from boot to start of NFCGate app, please?

ryu0q, 2023-09-14:
I am sorry for the delay, here I have attached the logs:
[image]

kleest, 2023-09-14:
@ryu0q Unfortunately, the look is incomplete and misses the relevant hooking logs :slightly_frowning_face:. Please capture it shortly after the device start so it is not truncated because of the limited log buffer. As an alternative to restarting the device, you can also kill the NFC service via a root shell: `killall com.android.nfc`

ryu0q, 2023-09-15:
[logcat.txt](https://github.com/nfcgate/nfcgate/files/12622891/logcat.txt)

Sorry for the delay caused again, I have now attached the logs using the `adb logcat -d` command. I hope this has the complete information. Feel free to ask me for new logs again if these aren't the right ones, or if I captured them wrong.

roussosalex, 2023-09-15:
Unfortunately, the log is still incomplete. The log should contain the tag `NATIVENFC` with the line `Library found at`, as your first log does but your second and third are missing this part. It may be that the log is going too fast and these lines were dropped by logcat, so you could try:

1. Running logcat on the device directly via `adb shell "logcat > /sdcard/log.txt"`. This should record the log directly on the device and does not need to transfer everything via adb.
2. You could also try killing the nfc process at a less busy time in a root shell: `killall com.android.nfc`, wait a few seconds, then use the `adb logcat -d` command to dump the logs. Since this is nowhere near boot, the log should be less busy and not skip anything.

dmunozv04, 2023-09-17:
Hello, I'm also experiencing the same issue with a S8. I've attached the logs gathered by `logcat` after killing the nfc service. I'm using the experimental samsung version.
[log.txt](https://github.com/nfcgate/nfcgate/files/12643126/log.txt)

ryu0q, 2023-09-17:
This one has that line:
[log.txt](https://github.com/nfcgate/nfcgate/files/12643215/log.txt)

roussosalex, 2023-09-18:
Thanks for posting the log, it really helped. We have made some small adjustments and have attached a new experimental version based on bba040eab3f95b9dab4da52a211a002f660691e3. 

Please give it a try and report back if it works, if it does not work please also attach a boot log with the  `NATIVENFC` line as last time.
[nfcgate-samsung-debug-v2.zip](https://github.com/nfcgate/nfcgate/files/12644033/nfcgate-samsung-debug-v2.zip)

ryu0q, 2023-09-18:
Not working
[image]

roussosalex, 2023-09-18:
The log seems to show native hook success now. Try force-stopping NFCGate and opening it back up to make sure the status page is up to date.

ryu0q, 2023-09-18:
I totally forgot to do that. Sorry for wasting your time :p
It's working now. Thank you all so much!!

closed 2023-09-18 by ryu0q

roussosalex, 2023-09-18:
Thanks for testing! Could you check if the hook also works if you open NFCGate for the first time after reboot, just to be sure?

ryu0q, 2023-09-24:
it does!

mr7Ouroboros, 2024-01-25:
Hello, I'm experiencing the same problem with the Native hook.
Samsung SM-J730FM/DS
I tried to force stop NFCGate and open it again, but nothing changed.
Please help me solve the problem. 
[image]
[image]
[image]
[image]

mr7Ouroboros, 2024-01-25:
Here are my logs
[log.txt](https://github.com/nfcgate/nfcgate/files/14057191/log.txt)

roussosalex, 2024-01-25:
Hello, you are most likely experiencing #141, we are working on a solution. Please open up a new issue next time.

mr7Ouroboros, 2024-01-25:
Okay, thank you.