nfcgate/nfcgate #138: native hook doesn't work in xiaomi redmi note 11 pro+

zimakkkk, 2023-10-12:
[image]

Device: xiaomi redmi note 11 pro+
OS: stock miui rom. v14.0.5.0 eu

[log2.txt](https://github.com/nfcgate/nfcgate/files/12882918/log2.txt)

roussosalex, 2023-12-16:
Looks like the image for this device does not contain any "lib*nfc*nci.so". It seems that the lib is directly loaded from an APK. Can you check that 

1. your filesystem does not contain any "lib*nfc*nci.so"?

roussosalex, 2023-12-16:
Thanks for reporting this issue! Looks like the image for this device does not contain any "lib\*nfc\*nci.so" and that the lib is directly loaded from an APK. Can you check that:

1. your filesystem does not contain any "lib\*nfc\*nci.so"?
2. you have the `NQNfcNci.apk` (probably in `/product/pangu/system/app/NQNfcNci/NQNfcNci.apk`)
3. the `NQNfcNci.apk` contains the file `lib/arm64-v8a/libsn100nfc-nci.so`

If this is the case, NFCGate might have to be adapted to expect to find the library in memory and no longer in the filesystem.

zimakkkk, 2023-12-17:
i checked in /system/lib/ and there was no "lib*nfc*nci.so".
and the app and library inside the app exist.

RandLily, 2023-12-20:
same as my note12-turbo and note11t-pro

roussosalex, 2024-01-09:
Work on this issue is in progress. Updates will follow.

roussosalex, 2024-01-27:
We have made some changes on the development branch (28a89c9ab8e8298c7df716a047c9cf481a64aed6) that should fix this issue. Could you test it and report if it works? If not, please provide a log as above. Thanks!

wait-what, 2024-02-05:
I have a similar xiaomi device with the same issue, same nfc chip and no such library in /system/lib. Can you please sent a prebuilt package so that I could test it?

i404788, 2024-02-06:
@roussosalex Also had this issue, building from https://github.com/nfcgate/nfcgate/commit/28a89c9ab8e8298c7df716a047c9cf481a64aed6 and a reboot seems to have fixed the 'native hook not found' issue (haven't tested any further).

@wait-what I've attached my build, but I recommend building it yourself (or wait for an official release) if possible, it's relatively easy to do
 [app-debug.zip](https://github.com/nfcgate/nfcgate/files/14186670/app-debug.zip)

roussosalex, 2024-02-08:
@i404788 Thank you for verifying!

closed 2024-02-23 by roussosalex