nfcgate/nfcgate #141: Native hook not available on Galaxy M31 (Magisk, Rooted, LSposed)

BenEgeDeniz, 2023-11-26:
**App Information\***
Screenshot of "status" page
[image]

**Device Information\*
Device: Samsung Galaxy M31
OS: Android 12 (stock, oneui)

**Log\***
[logcat1.txt](https://github.com/nfcgate/nfcgate/files/13467744/logcat1.txt)

**Additional context**
Tried relay attack on tag mode with a contactless payment card, POS device didnt recognize it as a payment card. All my setup is correct. The only reason should be the native hook.

balltobi, 2023-11-30:
Exactly the same issue for Samsung Galaxy S20FE, works fine with same setup on other devices

BenEgeDeniz, 2023-12-14:
I have checked the status page again and the native hook seemed to be enabled. I restarted the phone and it turned back to the yellow questionmark.

roussosalex, 2023-12-16:
Thanks for reporting! This issue seems a bit mysterious, because the log really has no "fatal" errors. Something Samsung-specific might be going on with the NFC startup that is not "standard". Could you try: 

* reboot the device
* open NFCGate
* if the question mark is yellow:
  * turn off NFC, wait a bit, turn NFC back on
  * force stop/kill NFCGate, wait a bit, open NFCGate again
  * check question mark again.  Report back on what happens

In any case, please try to pull the log from boot after this process and attach it.

roussosalex, 2023-12-16:
@balltobi Please attach a boot log so we can try to better understand the issue

BenEgeDeniz, 2023-12-21:
Now the native hook seems to be enabled. But I can't emulate a 4 byte UID.
I am trying to use NFCGate in Clone Tag mode and with the emulation on, my phone' UID changes to 4 bytes of 0x00 (00-00-00-00).
BUT there is some hope. It actually can modify my UID like that AND STILL MORE, it successfully emulates a contactless payment card' 7 byte UID perfectly every single time.

So, why can it emulate a 7 byte UID while it can't emulate a 4 byte UID?

BenEgeDeniz, 2023-12-22:
Yepi now I see why. Google is kinda dumb. They actually made it impossible to emulate any Mifare Classic 1K type cards. I can emulate a JCOP30 card with a 4 byte uid.

roussosalex, 2023-12-25:
> Now the native hook seems to be enabled.

Can you confirm that the native hook now works as expected, with no extra steps required?

> They actually made it impossible to emulate any Mifare Classic 1K type cards.  

Android no longer offers support for MiFare classic chips, so these cards are not supported

BenEgeDeniz, 2023-12-28:
> > Now the native hook seems to be enabled.
> 
> Can you confirm that the native hook now works as expected, with no extra steps required?
> 
> > They actually made it impossible to emulate any Mifare Classic 1K type cards.
> 
> Android no longer offers support for MiFare classic, so these cards are not supported

Nope, when I reboot the phone, native hook went away again.

roussosalex, 2024-01-04:
Did you try the steps I suggested above after rebooting the phone?

BenEgeDeniz, 2024-01-06:
> Did you try the steps I suggested above after rebooting the phone?

Yes, I did. Even if the native hook is not working, cloning a tag works well.

roussosalex, 2024-01-09:
If cloning works, the native hook must be working as well. Could you please share another log when the cloning is successful? It's possible that the UI is incorrectly displaying the status of the native hook. Your cooperation is greatly appreciated.

BenEgeDeniz, 2024-01-10:
I only checked the UID, and it worked. I don't think it emulates only the UID and tag type, am I correct?

If yes, please give me further instructions about saving a log file of NFCGate.

kleest, 2024-01-12:
> I think cloning only emulates only the UID and tag type, am I correct?

That's correct. 

> If yes, please give me further instructions about saving a log file of NFCGate.

Just a regular ADB logcat like you shared previously. Thanks!

roussosalex, 2024-02-23:
See #146

closed 2024-02-23 by roussosalex

italodeveloper, 2024-10-24:
> See #146

This build version not working today