nfcgate/nfcgate #145: [ Improvement ] Add support for Android14

jayb-g, 2024-01-14:
This is an awesome project and works beautifully on Android 13(tested on Asus ROG5) probably works on most other devices too. But when I try on an A14 device(Pixel7 and 8) it just crashes saying something like 'This app was developed for an older version of Android, please contact the developer.' and then crashes.

roussosalex, 2024-01-16:
Thank you! For Android 14 support, please follow #139

jayb-g, 2024-01-16:
Is everything in the checklist in the screenshot of #139 absolutely necessary? I mean would it work on non-root device e.g. on pixel 7/8? 
On 7, it detects the NFC chip correctly

roussosalex, 2024-01-16:
NFCGate does _work_ without root, but the most important features (e.g. emulating a tag with UID and other static tag data) require root and an xposed-compatible framework. You can read more about all requirements [here](https://github.com/nfcgate/nfcgate?tab=readme-ov-file#requirements-for-specific-modes).

weebl2000, 2024-01-31:
I guess this one can be closed.

jayb-g, 2024-02-01:
Yes

closed 2024-02-03 by roussosalex