nfcgate/nfcgate #146: Native hook could not be found

Lassie111, 2024-01-21:
**App Information\***
[image]

**Device Information\* (please complete the following information)**
Device: Samsung Galaxy S21 FE
OS: Android 14

**Log\***
Tell me what exactly you need from the log, since it contains sensitive data like my email and it's too big to clean it up.

roussosalex, 2024-01-22:
Thanks for reporting this issue! The log can sometimes just be filtered by "nfc" to avoid sharing sensitive data. In this case it looks like there are two issues in the log:
1. Android 14 incompatibility (that is currently being worked on, see #139)
2. Samsung specific quirks around NFC startup (see #141)

Lassie111, 2024-01-22:
> Thanks for reporting this issue! The log can sometimes just be filtered by "nfc" to avoid sharing sensitive data. In this case it looks like there are two issues in the log:
> 
>     1. Android 14 incompatibility (that is currently being worked on, see [Make NFCGate work on Android 14 #139](https://github.com/nfcgate/nfcgate/pull/139))
> 
>     2. Samsung specific quirks around NFC startup (see [Native hook not available on Galaxy M31 (Magisk, Rooted, LSposed) #141](https://github.com/nfcgate/nfcgate/issues/141))

Thanks for the reply! I've already seen 2. but didn't find any solution there. I tried the steps from your [comment](https://github.com/nfcgate/nfcgate/issues/141#issuecomment-1858687405) but "Native hook" always stays to "No".
Should I wait for an update for this to be fixed?

roussosalex, 2024-01-23:
You could try the version from #139 and see if that resolves the issue already. If not, you could post another logcat and we could see about issue 2.

Lassie111, 2024-01-23:
> You could try the version from #139 and see if that resolves the issue already. If not, you could post another logcat and we could see about issue 2. Or you can wait for the update fixing Android 14 and some other issues.

With #139 I have green on Android 14 but still Native hook is not enabled.
[log.txt](https://github.com/nfcgate/nfcgate/files/14026446/log.txt)

roussosalex, 2024-01-24:
The log contains no obvious errors, but the hooking is never completed as it should be. Could you try cloning some tag UID with NFCGate (using #139) even if the native hook is yellow on the status page and submit another log for that? Thanks for contributing your time!

Lassie111, 2024-01-24:
> The log contains no obvious errors, but the hooking is never completed as it should be. Could you try cloning some tag UID with NFCGate (using #139) even if the native hook is yellow on the status page and submit another log for that? Thanks for contributing your time!

Unfortunately, I don't have a reader available at the moment. Is it enough to just scan a tag? If now, I will have access to a reader next week and I will send you a log then.

roussosalex, 2024-01-24:
For this purpose just scanning the tag in clone mode should be enough

Lassie111, 2024-01-24:
> For this purpose just scanning the tag in clone mode should be enough

`01-24 20:09:20.015  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 20:09:20.015  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 00
01-24 20:09:23.366  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 20:09:23.366  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 01
01-24 20:10:04.592  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 20:10:04.592  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 00
01-24 20:10:04.622  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 20:10:04.622  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 00
01-24 22:01:49.239  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 22:01:49.239  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 00
01-24 22:03:18.101  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 22:03:18.101  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 00
01-24 22:03:27.833  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 22:03:27.833  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 00
01-24 22:04:37.331  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 22:04:37.331  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 00
01-24 22:04:54.682  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 22:04:54.682  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 00
01-24 22:05:25.441  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 22:05:25.441  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 00
01-24 22:05:28.301  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 22:05:28.301  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 00
01-24 22:05:58.933  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 22:05:58.933  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 00
01-24 22:40:28.651  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 22:40:28.651  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 00
01-24 22:41:03.339  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 22:41:03.339  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 00
01-24 22:41:38.601  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 22:41:38.601  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 00
01-24 22:41:58.678  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 22:41:58.678  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 00
01-24 23:21:32.462  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 23:21:32.462  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 00
01-24 23:21:34.753  3657  2945 I NATIVENFC: NFC_SetConfig()
01-24 23:21:34.753  3657  2945 I NATIVENFC: NFC_SetConfig Option Unknown(2, system, pass) 01
01-24 23:22:26.112  3657  4080 D NATIVENFC: Command: GET_HOOK_STATUS`

I can only find these in the log, but I don't think they are caused by the scanning attempt (since I didn't scan few hours ago but there are the same lines).
Cannot even scan a tag with #139, with the previous version (not supporting android 14) I was able to scan, but the reader didn't read the correct data.
Currectly, the phone only vibrates when trying to scan and nothing appears in the nfcgate app.

Lassie111, 2024-01-24:
Here is a new log after reboot and trying to scan a tag:
[log.txt](https://github.com/nfcgate/nfcgate/files/14043873/log.txt)

roussosalex, 2024-01-24:
So the `NFA_Enable` hook _should_ have been called right after the `nfcManager_doInitialize`  line in the log, but it was not. We will investigate this further and report back.

Lassie111, 2024-01-31:
> So the `NFA_Enable` hook _should_ have been called right after the `nfcManager_doInitialize` line in the log, but it was not. We will investigate this further and report back.

I built the version from the latest commit to test it and reading is working again. Though native hook was still not enabled.
I tried restarting NFC several times and force stopping nfcgate (+delete cache and data) and finally it became green (I think it will stay like this only until reboot). However, even with native hook enabled, cloning does not work. I attach a new log which may contain useful information for you (may contain lines related to nfc-tools as well, sorry about it).
[log.txt](https://github.com/nfcgate/nfcgate/files/14116844/log.txt)

weebl2000, 2024-01-31:
> I built the version from the latest commit to test it and reading is working again. Though native hook was still not enabled. I tried restarting NFC several times and force stopping nfcgate (+delete [...]

Just double checking, did you run: `git submodule update --init` before building?

Lassie111, 2024-01-31:
> Just double checking, did you run: `git submodule update --init` before building?

Yes.

weebl2000, 2024-01-31:
> Also I built it in debug mode, because when I build it in release, for some reason my phone says that the apk file is invalid and cannot install it.

Hmm, I've also had issues with the module not loading properly at times, usually it loaded after installing it and would be unloaded again after a reboot. My thought at the time was that perhaps it's using an older version upon reboot.

If you want you can try the build I'm currently using: https://github.com/weebl2000/nfcgate/releases/tag/20240131

roussosalex, 2024-01-31:
@Lassie111 unfortunately, the latest commits in v2-dev only address Android 14 and a different issue (libnfc being loaded directly from an APK). We are working on another fix for this (Samsung-specific) issue. Until then, it looks like despite the native hook being green in the status page, it did not properly install.

weebl2000, 2024-01-31:
@Lassie111 

Unfortunately I do not have a Samsung device on which I am able to test this. You could try building `[this](https://github.com/weebl2000/nfcgate/tree/log-library-loading)` branch and search for HOOKNFC in the logs - it might give more insight into why it is failing.

Lassie111, 2024-02-01:
> @Lassie111
> 
> Unfortunately I do not have a Samsung device on which I am able to test this. You could try building [this](https://github.com/weebl2000/nfcgate/tree/log-library-loading) branch and search for HOOKNFC in the logs - it might give more insight into why it is failing.
> 
> All I've added is some logging around the library loading point ![image](https://private-user-images.githubusercontent.com/2385844/301340355-063345e1-a54e-4d0a-86cd-e073a54cd0ab.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY3NDU1NjksIm5iZiI6MTcwNjc0NTI2OSwicGF0aCI6Ii8yMzg1ODQ0LzMwMTM0MDM1NS0wNjMzNDVlMS1hNTRlLTRkMGEtODZjZC1lMDczYTU0Y2QwYWIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEzMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMzFUMjM1NDI5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDI2OGU0NWQxNjU3ZjM3MzFmODRmYzEzZjcxZjE1OGVmNWM2ZDlhZDU4OWQwNTk2NTcwOGE4Zjc3MDcwM2RmMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.4ZM4opCwUEE5G3SSFat-ufldlG31_k3f-dGNT8Nt_uU)

Am I doing something wrong?
```
git clone https://github.com/weebl2000/nfcgate
cd nfcgate/
git switch log-library-loading
git submodule update --init
```
Output:
```
Submodule 'nfcd/xHook' (https://github.com/iqiyi/xHook.git) registered for path 'nfcd/xHook'
Submodule 'protocol' (https://github.com/weebl2000/protocol.git) registered for path 'protobuf/src/main/proto/protocol'
Cloning into '/mnt/c/Users/lassi/StudioProjects/nfcgate_test/nfcgate/nfcd/xHook'...
Cloning into '/mnt/c/Users/lassi/StudioProjects/nfcgate_test/nfcgate/protobuf/src/main/proto/protocol'...
remote: Repository not found.
fatal: repository 'https://github.com/weebl2000/protocol.git/' not found
fatal: clone of 'https://github.com/weebl2000/protocol.git' into submodule path '/mnt/c/Users/lassi/StudioProjects/nfcgate_test/nfcgate/protobuf/src/main/proto/protocol' failed
Failed to clone 'protobuf/src/main/proto/protocol'. Retry scheduled
Cloning into '/mnt/c/Users/lassi/StudioProjects/nfcgate_test/nfcgate/protobuf/src/main/proto/protocol'...
remote: Repository not found.
fatal: repository 'https://github.com/weebl2000/protocol.git/' not found
fatal: clone of 'https://github.com/weebl2000/protocol.git' into submodule path '/mnt/c/Users/lassi/StudioProjects/nfcgate_test/nfcgate/protobuf/src/main/proto/protocol' failed
Failed to clone 'protobuf/src/main/proto/protocol' a second time, aborting
```

weebl2000, 2024-02-01:
> fatal: repository 'https://github.com/weebl2000/protocol.git/' not found
> fatal: clone of 'https://github.com/weebl2000/protocol.git' into submodule path [...]

Easiest way would be to add my fork as a remote in your existing local project.

i.e. something like this:
```
cd /mnt/c/Users/lassi/StudioProjects/nfcgate/
git remote add weebl2000 https://github.com/weebl2000/nfcgate.git
git fetch --all
git checkout weebl2000/log-library-loading
```

Then you can checkout separate branches from different github forks (remotes).

Lassie111, 2024-02-01:
@weebl2000 Thank you! I have no experience with git.
```
02-01 02:32:37.003  3736  3736 I HOOKNFC : Trying to load library from path: /data/app/~~kDy6OgzhuNgkaKvtX9iEeg==/de.tu_darmstadt.seemoo.nfcgate-HlwW4p0AQOM1GH-bgfX2zQ==/lib/arm64/libnfcgate.so
02-01 02:32:37.791  3736  3736 I HOOKNFC : Loading library did not throw any exceptions.
02-01 02:32:37.792  3736  3736 D HOOKNFC : Library load failed
```

weebl2000, 2024-02-01:
> @weebl2000 Thank you! I have no experience with git.
>
> ```
> 02-01 02:32:37.003  3736  3736 I HOOKNFC : Trying to load library from path: [...]

I take it the file exists and is around 1 MB?

roussosalex, 2024-02-01:
@Lassie111 @weebl2000 the issue here has already [been identified](https://github.com/nfcgate/nfcgate/issues/146#issuecomment-1909108649) with the help of @Lassie111 last week and is being resolved. The "Library load failed" does not refer to `libnfcgate.so`, but to the libnfc library on the device (in this case `/system/lib64/libnfc-nxpsn.so`).

weebl2000, 2024-02-01:
> @Lassie111 @weebl2000 the issue here has already [been identified](https://github.com/nfcgate/nfcgate/issues/146#issuecomment-1909108649) with the help of @Lassie111 last week and is being resolved. [...]

Ah okay, got it.

@Lassie111 probably need to update the NFC chip code, your S21 for sure has an SN100, and it's falling back to pn547 since it is probably the name it shows up as in the /dev listing.

Can you post the contents of this file on your phone: `/vendor/etc/nfc/libnfc-nxp.conf` ?

Lassie111, 2024-02-01:
> Can you post the contents of this file on your phone: `/vendor/etc/nfc/libnfc-nxp.conf`? I'm mainly interested and cannot promise I will fix your problem lol - hopefully others will. :)

No such file. The only file in that directory is libnfc-nxp_RF.conf (renamed it to libnfc-nxp_RF.txt to upload it here):
[libnfc-nxp_RF.txt](https://github.com/nfcgate/nfcgate/files/14122972/libnfc-nxp_RF.txt)

> Oh and also: did it work on previous Android versions?
I haven't tried it before.

weebl2000, 2024-02-07:
> > Can you post the contents of this file on your phone: `/vendor/etc/nfc/libnfc-nxp.conf`? I'm mainly interested and cannot promise I will fix your problem lol - hopefully others will. :)
> 
> @weebl2000 No such file. The only file in that directory is libnfc-nxp_RF.conf (renamed it to libnfc-nxp_RF.txt to upload it here): [libnfc-nxp_RF.txt](https://github.com/nfcgate/nfcgate/files/14122972/libnfc-nxp_RF.txt)
> 
> > Oh and also: did it work on previous Android versions?
> 
> I haven't tried it before.

Thanks. I managed to get my hands on a Samsung Galaxy S21 today, if I have time this weekend I might try to reproduce this issue.

roussosalex, 2024-02-07:
The file path is `/vendor/etc/libnfc-nxp.conf` on the device in question, it indeed has `0x0B` as the chip code (which would be `SN1x0` according to the newer NXP chip codes). I have prepared a commit to add the newer chip codes, it will land here with the potential fix for the samsung NFC stack issue with native hook.

weebl2000, 2024-02-11:
> The file path is `/vendor/etc/libnfc-nxp.conf` on the device in question, it indeed has `0x0B` as the chip code (which would be `SN1x0` according to the newer NXP chip codes). I have prepared a [...]

I've created a draft PR for the fix, https://github.com/nfcgate/nfcgate/pull/152, I noticed at least the key NXP_NFC_CHIP was been replaced with NXP_NFC_CHIP_TYPE in the Samsung S21.

I'm not too familiair with native code debugging, but it I'm guessing the offset in the native library has changed - not sure where to start with how to fix this though. If you have any pointers I can have a go at it.

roussosalex, 2024-02-14:
@Lassie111 @weebl2000 Please try the new development version on the `v2-dev` branch and see if it fixes the native hook issue.

weebl2000, 2024-02-14:
> @Lassie111 @weebl2000 Please try the new development version on the `v2-dev` branch and see if it fixes the native hook issue.

Getting errors while building. Did not look into the exception yet but maybe you know if I'm missing a dependency?

```
$ ./gradlew build

> Task :nfcd:buildCMakeDebug[arm64-v8a] FAILED
C/C++: ninja: Entering directory `/home/weebl/git/nfcgate/nfcd/.cxx/Debug/73w4319f/arm64-v8a'
C/C++: : && /home/weebl/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ --target=aarch64-none-linux-android21 --sysroot=/home/weebl/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/linux-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security   -fno-limit-debug-info  -static-libstdc++ -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Wl,--gc-sections -Wl,--no-undefined -Qunused-arguments  -fuse-ld=lld -ffixed-x18 -shared -Wl,-soname,libnfcgate.so -o ../../../../build/intermediates/cxx/Debug/73w4319f/obj/arm64-v8a/libnfcgate.so CMakeFiles/nfcgate.dir/src/main/jni/src/helper/Config.cpp.o CMakeFiles/nfcgate.dir/src/main/jni/src/helper/MapInfo.cpp.o CMakeFiles/nfcgate.dir/src/main/jni/src/helper/System.cpp.o CMakeFiles/nfcgate.dir/src/main/jni/src/hook/IHook.cpp.o CMakeFiles/nfcgate.dir/src/main/jni/src/hook/Symbol.cpp.o CMakeFiles/nfcgate.dir/src/main/jni/src/hook/impl/ADBIHook.cpp.o CMakeFiles/nfcgate.dir/src/main/jni/src/hook/impl/XHook.cpp.o CMakeFiles/nfcgate.dir/src/main/jni/src/jni.cpp.o CMakeFiles/nfcgate.dir/src/main/jni/src/nfcd.cpp.o  -landroid  -llog  libxh.a  -latomic -lm && :
C/C++: ld.lld: error: undefined symbol: SymbolTable::parse(std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> > const&)
C/C++: >>> referenced by SymbolTable.h:12 (/home/weebl/git/nfcgate/nfcd/src/main/jni/include/nfcd/helper/SymbolTable.h:12)
C/C++: >>>               CMakeFiles/nfcgate.dir/src/main/jni/src/nfcd.cpp.o:(SymbolTable::create(std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> > const&))
C/C++: clang++: error: linker command failed with exit code 1 (use -v to see invocation)
C/C++: ninja: build stopped: subcommand failed.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':nfcd:buildCMakeDebug[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `/home/weebl/git/nfcgate/nfcd/.cxx/Debug/73w4319f/arm64-v8a'
  [1/1] Linking CXX shared library ../../../../build/intermediates/cxx/Debug/73w4319f/obj/arm64-v8a/libnfcgate.so
  FAILED: ../../../../build/intermediates/cxx/Debug/73w4319f/obj/arm64-v8a/libnfcgate.so 
  : && /home/weebl/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ --target=aarch64-none-linux-android21 --sysroot=/home/weebl/Android/Sdk/ndk/25.1.8937393/toolchains/llvm/prebuilt/linux-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security   -fno-limit-debug-info  -static-libstdc++ -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Wl,--gc-sections -Wl,--no-undefined -Qunused-arguments  -fuse-ld=lld -ffixed-x18 -shared -Wl,-soname,libnfcgate.so -o ../../../../build/intermediates/cxx/Debug/73w4319f/obj/arm64-v8a/libnfcgate.so CMakeFiles/nfcgate.dir/src/main/jni/src/helper/Config.cpp.o CMakeFiles/nfcgate.dir/src/main/jni/src/helper/MapInfo.cpp.o CMakeFiles/nfcgate.dir/src/main/jni/src/helper/System.cpp.o CMakeFiles/nfcgate.dir/src/main/jni/src/hook/IHook.cpp.o CMakeFiles/nfcgate.dir/src/main/jni/src/hook/Symbol.cpp.o CMakeFiles/nfcgate.dir/src/main/jni/src/hook/impl/ADBIHook.cpp.o CMakeFiles/nfcgate.dir/src/main/jni/src/hook/impl/XHook.cpp.o CMakeFiles/nfcgate.dir/src/main/jni/src/jni.cpp.o CMakeFiles/nfcgate.dir/src/main/jni/src/nfcd.cpp.o  -landroid  -llog  libxh.a  -latomic -lm && :
  ld.lld: error: undefined symbol: SymbolTable::parse(std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> > const&)
  >>> referenced by SymbolTable.h:12 (/home/weebl/git/nfcgate/nfcd/src/main/jni/include/nfcd/helper/SymbolTable.h:12)
  >>>               CMakeFiles/nfcgate.dir/src/main/jni/src/nfcd.cpp.o:(SymbolTable::create(std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> > const&))
  clang++: error: linker command failed with exit code 1 (use -v to see invocation)
  ninja: build stopped: subcommand failed.
  
  C++ build system [build] failed while executing:
      /home/weebl/Android/Sdk/cmake/3.22.1/bin/ninja \
        -C \
        /home/weebl/git/nfcgate/nfcd/.cxx/Debug/73w4319f/arm64-v8a \
        nfcgate
    from /home/weebl/git/nfcgate/nfcd

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

```

kleest, 2024-02-14:
Please find a compiled version attached. 
[app-debug.zip](https://github.com/nfcgate/nfcgate/files/14289035/app-debug.zip)

@weebl2000 Maybe you need to reload the CMake project because `bool SymbolTable::parse` is defined in a new file `SymbolTable.cpp`.

weebl2000, 2024-02-14:
> Please find a compiled version attached. [app-debug.zip](https://github.com/nfcgate/nfcgate/files/14289035/app-debug.zip)
> 
> @weebl2000 Maybe you need to reload the CMake project because `bool SymbolTable::parse` is defined in a new file `SymbolTable.cpp`.

Had to actually remove .cxx folder in nfcd/ before it would successfully build. It might be useful to add ccache to the CMakeList.txt - but I've never done it myself. https://medium.com/swlh/cross-platform-mobile-and-web-development-with-c-explained-8fb5fb916d35

More on topic: after installing the v2-dev version the native hook says to load, but clone mode does not read any tags for me yet. Will play around a bit more with it tomorrow or this weekend.
[image]

weebl2000, 2024-02-14:
On the S21 that I have my hands on the key `NXC_NFC_CHIP` was changed to `NXP_NFC_CHIP_TYPE`.

If I incorporate my own change from this branch: https://github.com/nfcgate/nfcgate/pull/152/files#diff-685182d753645bed3b6bc11f265e47b84c39b3957abd25fe65f6490292346b5R71

Then it does show the SN1x0 chip on the status page, but still no working clone mode.
[image]

roussosalex, 2024-02-15:
@weebl2000 Could you post a log from either reboot or killing the `com.android.nfc` process until trying to clone a card?

roussosalex, 2024-02-15:
@Lassie111 Thanks! There seem to be no obvious issues in the log.

Lassie111, 2024-02-19:
> @Lassie111 Thanks! There seem to be no obvious issues in the log.

I tested it and cloning still doesn't work.
[log.txt](https://github.com/nfcgate/nfcgate/files/14336174/log.txt)

roussosalex, 2024-02-20:
@Lassie111 can you describe in more detail how it does not work? The log looks like it should, the native hook seems to install properly

Lassie111, 2024-02-20:
> @Lassie111 can you describe in more detail how it does not work? The log looks like it should, the native hook seems to install properly

I scan the tag, it shows these lines
[image]
While showing this I put the phone near the reader, reader says "put only one card" or something like this.

roussosalex, 2024-02-21:
@Lassie111 could you:  
* comment out L114/115 in `nfcd/src/main/jni/src/jni.cpp`:  
```c++
// if (mask != 0)
//    nfaSetListenTech(mask);
```
* Rebuild the project (it may be required to delete the build or `.externalNativeBuild` directory)
* Try again!

Lassie111, 2024-02-21:
> @Lassie111 could you:
> 
>     * comment out L114/115 in `nfcd/src/main/jni/src/jni.cpp`:
> 
> 
> ```c++
> // if (mask != 0)
> //    nfaSetListenTech(mask);
> ```
> 
>     * Rebuild the project (it may be required to delete the build or `.externalNativeBuild` directory)
> 
>     * Try again!

Again the same.

roussosalex, 2024-02-23:
@Lassie111 from the log it seems like the native hook is working and being installed as it should. With that, this issue is technically resolved and I would like to ask you to open a new issue with the last log and a short problem description. We will have to look more into why cloning is not working for this device.

closed 2024-02-23 by roussosalex