nfcgate/nfcgate #147: Native hook could not be found [Samsung SM-J730FM/DS]

mr7Ouroboros, 2024-01-25:
**App Information\***
Screenshot of "status" page
[image][image]
[image]

**Device Information\***
Device: Samsung SM-J730FM/DS  
OS: Android 9

**Log\***
Test 1: [log-1.txt](https://github.com/nfcgate/nfcgate/files/14058326/log-1.txt)
Test 2: [log-2.txt](https://github.com/nfcgate/nfcgate/files/14058368/log-2.txt)

**Additional context**
Hello, can you please advise me how to solve the problem with the native hook on my device?

mr7Ouroboros, 2024-01-26:
The retransmission of traffic is slow, can you tell me if there are no timeouts or delays for retransmission in the source code?

kleest, 2024-01-26:
@mr7Ouroboros The network code uses a regular TCP connection with no additional retransmission or delay logic.

mr7Ouroboros, 2024-01-27:
If you replace Google Protocol Buffers with websocket, will it work?

mr7Ouroboros, 2024-02-02:
Hi, can you tell me the difference between v2.3.0 and v2.3.0-rc ?

roussosalex, 2024-02-02:
@mr7Ouroboros the `v2.3.0` has d2937334349e5d8c9e0c9b11f7db901f0bdf9fca and some very minor changes compared to the release candidate.

mr7Ouroboros, 2024-02-03:
After rebooting the problem with native hook started again, please advise, is it only in samsungs such a problem? What smartphones do not have this problem? Tired of this problem already, I think just change the smartphone.
Tell me which smartphones do not have problems with native hooks.

roussosalex, 2024-02-03:
There is a Samsung-specific issue at the moment with several devices, we are actively working on a fix (see #146). If you want to try another device in the meantime, see the [compatibility matrix](/nfcgate/nfcgate/blob/v2/doc/Compatibility.md#matrix) for suggestions

roussosalex, 2024-02-14:
@mr7Ouroboros Please try the new development version on the `v2-dev` branch and see if it fixes the native hook issue on your Samsung device.

kleest, 2024-02-14:
You can find a compiled version attached to [my comment in the other issue](https://github.com/nfcgate/nfcgate/issues/146#issuecomment-1944757296).

closed 2024-02-23 by roussosalex