nfcgate/nfcgate #153: Cloning does not work (Galaxy S21 FE)

Lassie111, 2024-02-23:
**Describe the bug**
After scanning a tag and bringing the phone near the reader, the reader does not accept it and gives messages like "put only one card near the reader" or "return the card to the reader and try again". While holding the phone near the reader, the reader beeps several times (like 2-3 per second) and displays an error.

**App Information\***
[image]

**Device Information\* (please complete the following information)**
Device: Samsung Galaxy S21 FE
OS: Android 14

**Log\***
[log6.txt](https://github.com/nfcgate/nfcgate/files/14380991/log6.txt)

Lassie111, 2024-02-29:
Update:
I tried using another phone as a reader to see what is received. It seems like the cloning "works" but it sends wrong data. Firstly, no matter if I'm trying to clone a tag with 4-byte or 7-byte UID, the other phone always receives 4-byte UID. Secondly, the received UID looks completely unrelated to the UID being cloned.

Examples:
- When trying to clone a tag with UID 0x045451721E7380, the other phone receives 0x08395E87 or 0x08AD9A6E or 0x086B9A85;
- When trying to clone a tag with UID 0xE6B3A5D3, the other phone receives UID 0x08E84E52 or 0x089CE4B7 or 0x08AEDE9F.

Screenshot 1 - what is nfcgate showing:
[image]

Screenshot 2 - what is NFC tools showing on the second phone when the ORIGINAL tag is scanned:
[image]

Screenshot 3 - what is NFC tools showing on the second phone when the first phone with nfcgate is scanned:
[image]

roussosalex, 2024-03-07:
This issue is not a fault of NFCGate. We have seen similar issues with OnePlus devices:  

First, the UIDs you are seeing are not set by NFCGate, they are "random" UIDs generated by the NFC chip (e.g. `0x08E84E52`). You can recognize random UIDs with the "0x08" prefix. This is the default behavior of the NFC chip and should happen even if NFCGate is not running or not even installed.

Second, NFCGate correctly sends the NFCID to the NFC chip as seen in the log. Everything works, except the NFC chip does not accept the NFCID and keeps presenting random UIDs. This exact issue is why cloning does not work on any OnePlus device we tested so far (@kleest). Unfortunately, we have not found any solution for this issue yet except switching to a different device.

Lassie111, 2024-03-07:
> First, the UIDs you are seeing are not set by NFCGate, they are "random" UIDs generated by the NFC chip (e.g. `0x08E84E52`). You can recognize random UIDs with the "0x08" prefix. This is the default behavior of the NFC chip and should happen even if NFCGate is not running or not even installed.

Yeah, I found that about the random UIDs later after I posted the comment.

Though, I didn't know that no solution is known.
Thanks for your hard work trying to help!

Lassie111, 2024-04-01:
> Second, NFCGate correctly sends the NFCID to the NFC chip as seen in the log. Everything works, except the NFC chip does not accept the NFCID and keeps presenting random UIDs. This exact issue is [...]

Do you know if relay/replay should work in that case? Currently, I don't have a second rooted device to test it.

Also do you know if this is a hardware limitation and it's impossible to be fixed or just no solution is found yet? If it's possible to be fixed, I may try to fix it myself after some months when I get some free time.

kleest, 2024-04-01:
> @roussosalex Do you know if relay/replay should work in that case? Currently, I don't have a second rooted device to test it.

Relaying and replaying APDUs does not require successfully setting NFCID or otherwise applying a config stream. This means that relay and replay *could* work if the devices do not check NFCID (or other static tag data). It depends on your applications, for example, with EMV systems the NFCID is not relevant.

> Also do you know if this is a hardware limitation and it's impossible to be fixed or just no solution is found yet? If it's possible to be fixed, I may try to fix it myself after some months when I get some free time.

We actually don't know. Public datasheets of the NFC chipset do not list this problem and we debugged it in detail on OnePlus devices and could not make it work. The NFC chipset just does not respect the NFCID we set. It could also be a chipset configuration option, but finding documentation about possible configuration settings is hard or not even possible with public datasheets.
Feel free to investigate it yourself, it would be very nice to find a solution! If you need a starting point just ask. :-)

roussosalex, 2024-04-02:
> Do you know if relay/replay should work in that case

Replay and relay work even without root, as long as you do not care about setting any "static tag data" such as the NFCID.

> Also do you know if this is a hardware limitation

AFAIK the specification requires the NFC chip to support setting the NFCID. 

> If it's possible to be fixed, I may try to fix it myself after some months when I get some free time.

It's most likely some state-management or configuration issue.  
From reading your `logfile6.txt`, I came up with some theories on what could be the problem:
- the NFCID is being set correctly, but the NFC chip is not in the "right" state to accept a change in the discovery parameters
- perhaps the chip only accepts a change to these discovery parameters the first time the state is "idle"  
- maybe the secure element or SIM interfere with the config after is has been set (through e.g. `NFCC_CONFIG_CONTROL`)
- there is some configuration setting that, perhaps as a side-effect, always enables random NFCIDs (maybe something like `CON_DISCOVERY_PARAM`)
- some proprietary firmware config could disallow changing the NFCID
- it may also be something completely different. 

Some of these theories could be tested. If you would like to disprove any (or all) of them or introduce new theories of your own, it would be very much appreciated!

Lassie111, 2024-04-02:
Thank you once again! I will try to investigate it when I have some time.

Lassie111, 2024-04-02:
I started by looking at the `hook_NFC_SetConfig` function and printing the returned value of `globals.hNFC_SetConfig->call<def_NFC_SetConfig>(actual.total(), bin_stream.get())` to the log. It returns 0 (which I found is `NCI_STATUS_OK`).

I looked at the code of `NFC_SetConfig` in the nfc library to see what is going on inside. I found it [here](https://android.googlesource.com/platform/external/libnfc-nci/+/android-4.4_r1/src/nfc/nfc/nfc_main.c#814).

In the function `nci_snd_core_set_config` (found [here](https://android.googlesource.com/platform/external/libnfc-nci/+/refs/heads/jb-mr1-dev/src/nfc/nci/nci_hmsgs.c#139)), which is called by `NFC_SetConfig`, I saw that `NCI_STATUS_FAILED` is returned only if either the memory allocation fails, or the arguments (more precisely, the buffer pointed to by `p_param_tlvs`) passed are not in the expected format. In all other cases the function returns `NCI_STATUS_OK`.

To me, it looks like the actual config setting is done by [this](https://android.googlesource.com/platform/external/libnfc-nci/+/refs/heads/jb-mr1-dev/src/nfc/nci/nci_hmsgs.c#178) `nfc_ncif_send_cmd (p);` (the rest is just preparation of the buffer pointed to by `p`). But the returned value of `nci_snd_core_set_config` does not depend on it. So, according to the library, `nfc_ncif_send_cmd (p);` should always succeed? I didn't have time to look further into nfc_ncif_send_cmd, because it looks a lot more complicated.

So my main guesses would be

- `nfc_ncif_send_cmd (p);` fails silently, because the returned status does not depend or it;
- everything succeeds, but something else immediately resets it (if this is the case, that other thing should be using something other than `NFC_SetConfig`, because otherwise `hook_NFC_SetConfig` would be called since it's hooked?).

roussosalex, 2024-04-02:
The `NFC_SetConfig` works as expected and sends the config stream down to the NFC chip, the problem likely lies in the entire system working together, not in any single point or function. This issue might require a deeper understanding of the NFC NCI stack and the inner workings of Android Hardware. 

We have prepared test code on the [v2-dev-debug-nfcid-153 branch](https://github.com/nfcgate/nfcgate/tree/v2-dev-debug-nfcid-153). If you look in `nfcd.h`, there are several `DEBUG_N` switches to test with. If you could test the following combinations (check the NFCID after starting NFCGate without cloning anything and the NFCID after cloning a tag, also collect a log):

* `any one option` and `all options`: as a baseline
* `(DEBUG_1, DEBUG_3)` and `(DEBUG_1, DEBUG_2, DEBUG_3)` may set the NFCID to `04 54 51 72 1E 73 80` after restarting the `com.android.nfc` process
* `(DEBUG_4, DEBUG_5)` may set the NFCID after cloning, this would be ideal

Thank you for contributing to the solution for this issue!

Lassie111, 2024-04-03:
I've tried all of the combinations you described and in all cases a random ID was presented (both after starting NFCGate and after cloning a tag).

[log_10000.txt](https://github.com/nfcgate/nfcgate/files/14856087/log_10000.txt)
[log_01000.txt](https://github.com/nfcgate/nfcgate/files/14856088/log_01000.txt)
[log_00100.txt](https://github.com/nfcgate/nfcgate/files/14856089/log_00100.txt)
[log_00010.txt](https://github.com/nfcgate/nfcgate/files/14856090/log_00010.txt)
[log_00001.txt](https://github.com/nfcgate/nfcgate/files/14856091/log_00001.txt)
[log_11111.txt](https://github.com/nfcgate/nfcgate/files/14856092/log_11111.txt)
[log_10100.txt](https://github.com/nfcgate/nfcgate/files/14856093/log_10100.txt)
[log_11100.txt](https://github.com/nfcgate/nfcgate/files/14856095/log_11100.txt)
[log_00011.txt](https://github.com/nfcgate/nfcgate/files/14856097/log_00011.txt)

roussosalex, 2024-04-04:
Thank you for putting in all that effort! It is very unfortunate that nothing has worked so far.

We will read through the logs and hopefully come up with new ideas soon.

roussosalex, 2024-05-22:
Good news, while looking into #159 we stumbled upon a possible solution for this issue (and other devices, including the OnePlus device limitation). Please try the new debug apk [app-debug.zip](https://github.com/nfcgate/nfcgate/files/15396627/app-debug.zip) and confirm if it works for you now!

Lassie111, 2024-05-22:
Unfortunately, it still does not work and the app also the app crashes frequently.

[log.txt](https://github.com/nfcgate/nfcgate/files/15406361/log.txt)

roussosalex, 2024-05-23:
After some bugfixing we have a new debug apk [app-debug.zip](https://github.com/nfcgate/nfcgate/files/15411016/app-debug.zip). Please try it and report if it works.

Lassie111, 2024-05-23:
> After some bugfixing we have a new debug apk [app-debug.zip](https://github.com/nfcgate/nfcgate/files/15411016/app-debug.zip). Please try it and report if it works.

Now it works! How did you fix it?

Here is a log anyway if you need it for additional improvements 
[log2.txt](https://github.com/nfcgate/nfcgate/files/15413147/log2.txt)

roussosalex, 2024-05-27:
> How did you fix it

To simplify, we discovered that the Embedded Elements (EEs) are interfering with the NFCGate operation. Then, we added code to disable the EEs when emulating and re-enable them when we are done. 

This fix has led to fixing #159  as well as the issues with OnePlus devices

roussosalex, 2024-05-30:
This fix is now included in [v2.4.0](https://github.com/nfcgate/nfcgate/releases/tag/v2.4.0)

DimiDimit, 2024-12-23:
Interestingly enough, v2.4.4 still has this problem on my S22 Ultra (everything is green but the UID is random), but the [aforementioned debug version](https://github.com/nfcgate/nfcgate/issues/153#issuecomment-2126152940) works on it…

kleest, 2025-01-04:
In 54337e5a8c631e45b1bad7da49cc45d9ebdb93e9 we added disabling EEs. Later on we noticed that we should only disable proprietary EEs since disabling all caused problems on some devices (77052edab5e5d79d0018ea80f59c19b1bb6ae2fd). However, it looks like this causes problems on your Samsung devices.

Could you please try out the APK from the PR? See https://github.com/nfcgate/nfcgate/actions/runs/12611484927

Lassie111, 2025-01-04:
> In [54337e5](https://github.com/nfcgate/nfcgate/commit/54337e5a8c631e45b1bad7da49cc45d9ebdb93e9) we added disabling EEs. Later on we noticed that we should only disable proprietary EEs since disabling all caused problems on some devices ([77052ed](https://github.com/nfcgate/nfcgate/commit/77052edab5e5d79d0018ea80f59c19b1bb6ae2fd)). However, it looks like this causes problems on your Samsung devices.
> 
> Could you please try out the APK from the PR? See https://github.com/nfcgate/nfcgate/actions/runs/12611484927

The apk from the PR works as expected.

closed 2025-01-05 by kleest

midozalouk, 2025-06-13:
> The apk from the PR works as expected.

@Lassie111 
Sorry mate if you have the apk , can you upload it because it has been expired 😔!

Lassie111, 2025-06-13:
> [@Lassie111](https://github.com/Lassie111) Sorry mate if you have the apk , can you upload it because it has been expired 😔!

I do not have it. As suggested already, you can build it yourself.

teknoasistan, 2025-08-03:
[image]
[image]
[image]

The application reads the NFC tag but I get an error message.
I have a native hook enableled yellow exclamation.

kleest, 2025-08-03:
@teknoasistan please stop bumping issues with comments unrelated to the thread. If you have issues with the native hook on your device, please create a new issue and follow the template closely. Thank you.

PR #194 (2025-01-05): Fix EE disabling on Samsung devices
Fixes #153