nfcgate/nfcgate #156: Native hook issues with SM-A526B

cydef1, 2024-03-30:
**App Information***

[image]

**Device Information*** 

Samsung Galaxy A52 5G (SM-A526B)

OS:   Android OS 13

**Log***
[image]

roussosalex, 2024-03-31:
Thanks for contributing!

Your issue looks like this other Samsung-specific issue: #146. Please try the debug-version linked here: https://github.com/nfcgate/nfcgate/issues/146#issuecomment-1944757296. We will soon incorporate this into a future release.

cydef1, 2024-03-31:
[image]

cydef1, 2024-03-31:
Now it works perfectly

closed 2024-03-31 by cydef1

Lassie111, 2024-03-31:
> Now it works perfectly

Does the cloning work correctly though? For my device even after Native hook is enabled, the NFC chip does not accept the NFCID and presents a random one instead (#153).

I wonder if it is a problem with my device only or with all Samsung devices.

roussosalex, 2024-04-01:
@Lassie111 It looks like the behavior you are describing has nothing to do with Samsung devices and everything to do with the type of NFC Chip. Your NXP chip has the same issues as many OnePlus models with the same chip (i.e. it only uses random NFCIDs and does not accept a manually set one). @cydef1 has a Samsung NFC chip, where that issue has never been reported so far.