nfcgate/nfcgate #159: NFC

linsui, 2024-04-19:
**Describe the bug**

A clear and concise description of the bug. What is happening that is not supposed to? Or vice versa.

**App Information***

Please provide a screenshot of the "status" page

**Device Information*** Which device are you using and what OS are you running on it?

Device: [e.g. Google Pixel 7]

OS: [e.g. Android 14]

**Log***

Attach a boot log (system logcat since boot).
See [here](https://www.xda-developers.com/how-to-take-logs-android/#system-logs) for detailed
instructions.

**Steps to reproduce**

Please describe the steps taken to reproduce the problem.
Be as descriptive as can you be, e.g.:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**

A clear and concise description of what you expect to happen.

**Observed/unwanted behavior**

What is the behavior that is occurring?

**Screenshots**

Optionally, add screenshots to help show the problem.

**Additional context**

Add any other context about the problem here.

**Proposed solution**

If you have thought about a solution for the problem, please elaborate here!

roussosalex, 2024-04-19:
Thanks for contributing!

Please also post a logcat snippet around the time of the crash. Would you be able to modify and recompile the code to test a possible fix?

linsui, 2024-04-20:
> Please also post a logcat snippet around the time of the crash.

I have posted the logcat. :) It's collapsed by default.

> Would you be able to modify and recompile the code to test a possible fix?

Yep, this apk is built from master.

roussosalex, 2024-04-20:
> I have posted the logcat 

You have, but I meant more of the preceding logcat around the time of the crash. Especially lines containing "nfc", "NATIVENFC" or "HOOKNFC" are relevant.

linsui, 2024-04-21:
[21_04-12-40-07_258.log](https://github.com/nfcgate/nfcgate/files/15050682/21_04-12-40-07_258.log)

kleest, 2024-04-21:
Your ROM is missing the `NFA_SetP2pListenTech` function in `libnfc-nci.so`. This was removed in https://android-review.googlesource.com/c/platform/system/nfc/+/2152040. This is also an issue on Pixel 7 Android 14. We will have a look, thanks for reporting!

closed 2024-05-22 by roussosalex

roussosalex, 2024-05-22:
We have found a new way to limit the listen tech (see 0ccbe62). Please try the new debug version [app-debug.zip](https://github.com/nfcgate/nfcgate/files/15396611/app-debug.zip) to confirm everything works for you now.

linsui, 2024-05-22:
It doesn't work for me. 
[22_05-08-43-21_313.log](https://github.com/nfcgate/nfcgate/files/15396766/22_05-08-43-21_313.log)

roussosalex, 2024-05-22:
Looks like there is still a bug in the code, could you post a bit more of the log preceding the crash? Especially lines containing "nfc", "NATIVENFC" or "HOOKNFC" are relevant, it seems like you are currently only filtering by "nfc".

linsui, 2024-05-22:
[nfc.log](https://github.com/nfcgate/nfcgate/files/15397790/nfc.log)

Does this have more info?

roussosalex, 2024-05-22:
Thanks! I was able to fix the bug with the help of your log. Please try the newest debug version [app-debug.zip](https://github.com/nfcgate/nfcgate/files/15407977/app-debug.zip) to confirm everything works as expected now.

linsui, 2024-05-23:
This crash in clone mode is fixed, thanks! But after cloning a tag, when I enter the relay mode it crashes.

<details>

```
FATAL EXCEPTION: main
Process: de.tu_darmstadt.seemoo.nfcgate, PID: 8911
java.lang.NullPointerException: Attempt to invoke virtual method 'de.tu_darmstadt.seemoo.nfcgate.nfc.NfcManager de.tu_darmstadt.seemoo.nfcgate.gui.MainActivity.getNfc()' on a null object reference
	at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.BaseFragment.getNfc(BaseFragment.java:10)
	at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.CloneFragment.lambda$onActivityCreated$3$de-tu_darmstadt-seemoo-nfcgate-gui-fragment-CloneFragment(CloneFragment.java:104)
	at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.CloneFragment$$ExternalSyntheticLambda1.onChange(Unknown Source:2)
	at de.tu_darmstadt.seemoo.nfcgate.nfc.NfcManager.notifyStatusChanged(NfcManager.java:121)
	at de.tu_darmstadt.seemoo.nfcgate.nfc.hce.DaemonManager.onResponse(DaemonManager.java:33)
	at de.tu_darmstadt.seemoo.nfcgate.gui.MainActivity.onNewIntent(MainActivity.java:133)
	at android.app.Activity.performNewIntent(Activity.java:8785)
	at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1619)
	at android.app.Instrumentation.callActivityOnNewIntent(Instrumentation.java:1632)
	at android.app.ActivityThread.deliverNewIntents(ActivityThread.java:4166)
	at android.app.ActivityThread.handleNewIntent(ActivityThread.java:4173)
	at android.app.servertransaction.NewIntentItem.execute(NewIntentItem.java:58)
	at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:60)
	at android.app.servertransaction.TransactionExecutor.executeNonLifecycleItem(TransactionExecutor.java:195)
	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:157)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:90)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2544)
	at android.os.Handler.dispatchMessage(Handler.java:107)
	at android.os.Looper.loopOnce(Looper.java:232)
	at android.os.Looper.loop(Looper.java:317)
	at android.app.ActivityThread.main(ActivityThread.java:8501)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:878)
```

</details>

And NFC service also crashes later. 
[nfc.log](https://github.com/nfcgate/nfcgate/files/15410490/nfc.log)

roussosalex, 2024-05-23:
Thanks for testing! Looks like you are really good at finding all the bugs. I have fixed the nfc service crash and hopefully the clone mode crash as well. Please try the newest debug version [app-debug.zip](https://github.com/nfcgate/nfcgate/files/15410806/app-debug.zip) yet again to confirm everything works as expected.

linsui, 2024-05-23:
Everything works well for me, thanks for you quick fix!

roussosalex, 2024-05-30:
This issue has been fixed in release [v2.4.0](https://github.com/nfcgate/nfcgate/releases/tag/v2.4.0)