nfcgate/nfcgate #164: Scanner reports

IzzySoft, 2024-06-01:
May I kindly ask to take a look [at VT](https://www.virustotal.com/gui/file/810845d24ee9c69e9101370a24cf2e14ebe54242587ce6339535e984f23973c6/detection/f-810845d24ee9c69e9101370a24cf2e14ebe54242587ce6339535e984f23973c6-1717151985) for the latest release? With two major engines agreeing and being supported by 2 others, I'm not 100% sure to call it a "false positive" (maybe something slipped in?). I'll remove this update from IzzyOnDroid for now, just to be on the safe side (no offense meant of course!) – waiting for your response. Thanks for taking care!

kleest, 2024-06-01:
That's odd. We will have a look. Thanks for making us aware!

IzzySoft, 2024-06-01:
Odd indeed! Hope it's something easy-to-fix :crossed_fingers:

kleest, 2024-06-01:
The previous NFCGate version 2.3.0 [is also flagged now](https://www.virustotal.com/gui/file/ea85c0bdff4eb519d80627f6cae9f2ce6926bf9ce60ad103324fe43cf8801a6e). Building a debug version locally from source also triggers the detection. This is a false positive. We will reach out to the vendors.

IzzySoft, 2024-06-01:
Oof, by 5 engines even. Guess it's the "cart in front of the horse" again, scanners detecting something malware would use as well instead of the other way around (wouldn't be the first time). Hope you can identify the culprit and have it fixed by its vendor. Thanks again for digging in!

kleest, 2024-07-08:
After some communication with vendors, the newest version 2.4.2 [is no longer flagged as malware or unwanted software](https://www.virustotal.com/gui/file/452e321ee5f48030331edd5e55ca936ddd2b20488ced726ad96d467fc04ca5e5). However, older versions and debug versions still are. I will keep this issue open for further updates.

GlassGruber, 2024-08-29:
I'm afraid the false positive is in part also because of this
https://www.welivesecurity.com/en/eset-research/ngate-android-malware-relays-nfc-traffic-to-steal-cash/

IzzySoft, 2024-08-29:
Ah, so again they blame the producer of the getaway car for the bank robbery performed by the car's driver. Not the first time I see such… :man_facepalming:

roussosalex, 2024-09-01:
We have added a statement regarding the recent news.

IzzySoft, 2024-12-12:
Ouch, it's [getting worse](https://www.virustotal.com/gui/file/89ebb5491189245058529365c0c8db53e1ed1bb2e289a0260e7a9e82c90f5b4a/detection/f-89ebb5491189245058529365c0c8db53e1ed1bb2e289a0260e7a9e82c90f5b4a-1733944274): this time it's 9 engines, so v2.4.4 showed up with a red shield screaming "Alert!"

Could you please include a hint with your app description (`full_description.txt`) linking to your statement? I've just injected the following paragraph here locally, but it would be overwritten with your next update:

```html
<p><b>Important notice:</b> Currently, NFCGate is wrongly flagged as malware by multiple scanning engines. Please see <a href='https://github.com/nfcgate/nfcgate/issues/176'>the statement of the NFC Gate team</a> for background on this. In short, this app is <b>not</b> malware. Malicious actors copied significant portions of source code from NFCGate when creating their malware, so this app is wrongly detected as such.</p>
```

I wrote it in the third person to make clear it's not part of your description, but be welcome to copy (and optionally adjust) it to your `full_description.txt`. Thanks!

@roussosalex @kleest

kleest, 2024-12-12:
Hi @IzzySoft, sounds like a good idea! We will include a statement as part of the next update.

IzzySoft, 2024-12-13:
Preview from here:

[image]

Hm, no icon? Ah, couldn't be fetched because it's a symlink in the fastlane structure (something the Github API doesn't handle easily or I'm to stupid to figure how). OK, fixed now manually :wink:

roussosalex, 2025-08-05:
In the never-ending quest of tilting at windmills that is sending each and every release to the AV vendors, we are unfortunately getting the short end of the stick due to time constraints. They are also very reluctant to fix the root of the issue, instead we are only  whitelisted for a specific release, the next one (or any debug build) is still flagged.

IzzySoft, 2025-11-24:
That's good news, congrats! Hope the answer from the remaining 10 (I've just triggered a rescan for the latest release we have here, versionCode 15; went down from 26 to 10 "hits") will be appropriate as well 🤞