nfcgate/nfcgate #165: Not Working on Samsung S22 (Android 14) but Status All Green

IPFSae, 2024-06-09:
**Describe the bug**

1. In **Clone Mode**, it showed red error message: "Error: NFC is disabled or unsupported". And no data show in the "Saved:" area.

[image]

[image]

2. In **Relay Mode**, (bank Visa) card head information can be relayed to another phone successfully, but there won't be any data transmit while tap the relay phone tag on a reader (a POS machine).

[image]

3. Other mode not tested...

**App Information***

[image]

**Device Information*** 

Device: Samsung Galaxy S22

OS: Android 14 (One UI 6.1)

**Log***

Attach a boot log (system logcat since boot).

[logcat.txt](https://github.com/user-attachments/files/15754258/logcat.txt)

**Steps to reproduce**

N/A

**Expected behavior**

What to try relay a bank Visa card from place A (Rooted Samsung S22 as reader) to place B (stock Samsung S23 as tag) and tap on a POS machine. I don't really need the clone mode, but as it didn't work with red error so I believe something wrong inside.

**Proposed solution**

I didn't test on more phone yet... maybe Samsung S22 is not supported ?

Lassie111, 2024-06-11:
You can try the debug version from this [comment](https://github.com/nfcgate/nfcgate/issues/153#issuecomment-2126152940).
I had similar issue with version 2.4.0, while the older debug version works fine for me.

I was going to open an issue about this, but didn't have much time recently.

kleest, 2024-06-11:
Thanks for the report! The service crash is fixed with newest release [2.4.2](https://github.com/nfcgate/nfcgate/releases/tag/v2.4.2). The problem was a function callback that went out-of-scope. This did not cause issues in the debug version but only in the release build. Sorry!

This should fix the problem that wrong data is displayed and transported to the other phone. If I understood your setup correctly, the phone you are using to emulate the tag and tap against the POS machine is a non-rooted phone, am I right? This might not work, because such a phone cannot emulate NFCID, HIST and other parameters (static tag data). You should swap the phones. Reading all of this data with the reader phone does not require root, however, emulation does.

IPFSae, 2024-06-12:
> Thanks for the report! The service crash is fixed with newest release [2.4.2](https://github.com/nfcgate/nfcgate/releases/tag/v2.4.2). The problem was a function callback that went out-of-scope. [...]

Thanks. NFC error Issue fixed on my phone. I can copy and replay UID cards now. But relay to a POS still be a challenge even after swipe. I believe in POS system there are some functions to avoid such relay...

closed 2024-06-12 by IPFSae