nfcgate/nfcgate #167: Please update instructions / EdXposed and Magisk "Downloads" unavailable

xsrf, 2024-06-15:
**Describe the bug**

I can't get required software to work because of outdated instructions

**App Information***

- Xposed Module Enabled: ?
- Native Hook Enabled: ?

Banner "Warning: Xposed module is not working properly"

**Device Information*** Which device are you using and what OS are you running on it?

Device: Google Nexus 5X

OS: Android 8.1.0 rooted with Magisk 25.2 / (also tested 27.0)

**Steps to reproduce**

EdXposed stable is only available in the Magisk "Downloads" Tab which it was removed some time ago, so it can't be installed.
EdXposed alpha is available as zip, but install via Magisk fails because it insists it wants riru from Magisk "Downloads" Tab which doesn't exist and rirus regular install (which is available as zip) wasn't recognized by EdXposed alpha install script.

LSPosed can be downloaded as zip and installed in Zygisc mode but is not recognized by NFCGate.
LSPosed can also be installed in riru mode but is also not recognized by NFCGate.

I never get a Superuser request for NFCGate btw.

**Expected behavior**

Instructions should work

**Screenshots**
Device 1

[image]
[image]
[image]
[image]

Device 2

[image]
[image]
[image]
[image]

Tomatcree01, 2024-06-15:
You didn't properly quit and then activate. I am able to get it to detect the xposed environment but not native hooks...

xsrf, 2024-06-16:
@Tomatcree01 the phones rebooted several times by now, that shouldn't be an issue

Attaches is the logcat after `killall com.android.nfc`: [logcat_nfc_kill.txt](https://github.com/user-attachments/files/15856512/logcat_nfc_kill.txt)

I cannot find any `NATIVENFC` or `Library found at` in the complete logcat since reboot (as mentioned in #131)

kleest, 2024-06-16:
Did you enable the module in the *posed module manager? At least LSposed requires you to tick the app/service for which you want to enable the module. NFCGate lists com.android.nfc as a suggestion there. After the module is enabled the NFC service must be restarted either by killing it or restarting the device.
We had great success using NFCGate on the Nexus 5X with the same ROM version. It was one of the primary development devices. Using NFCGate there should be straightforward: Install Magisk, install LSposed, install NFCGate, enable the module in LSposed manager, kill NFC service.

kleest, 2024-06-16:
> You didn't properly quit and then activate. I am able to get it to detect the xposed environment but not native hooks...

@Tomatcree01 If xposed module is shown as enabled but the native hook is not active, please open a separate issue with logs.

xsrf, 2024-06-16:
These Tools really require lots of prior knowledge. I didn't enable a Module in LSposed Manager. I didn't even know it exists. On the ruri version there is also no notification to open the manager.
I enabled the module and now all checks are green. Guess it works, thx.
Would be nice if this could be mentioned in the readme. I've done lots of android related development but never used *posed and this is not at all obvious.

kleest, 2024-06-16:
Since devices and instructions to install and use  *posed frameworks on those are very diverse, we refrain from adding specific instructions to the readme which might not universally apply. The idea is to keep the readme rather simple and treat dependencies and requirements as a blackbox. So, we send the user to the frameworks where they find specific instructions. For example, how to use LSposed is described in [its wiki](https://github.com/LSPosed/LSPosed/wiki/How-to-use-it).

We consider improving the instructions and UX of the onboarding process, thanks for your input! If additional problems arise, feel free to re-open or create a new issue.

closed 2024-06-16 by kleest

xsrf, 2024-06-16:
At least noting that EdXposed is not available anymore via the documented sources would have saved me lots of time though

Tomatcree01, 2024-06-18:
> > You didn't properly quit and then activate. I am able to get it to detect the xposed environment but not native hooks...
> 
> @Tomatcree01 If xposed module is shown as enabled but the native hook is not active, please open a separate issue with logs.

I managed to get it to work lol