nfcgate/nfcgate #168: Clone Mode not reacting to Reader on Nexus 5X

xsrf, 2024-06-16:
Hey,
so I have two Nexus 5X with stock Android 8.1.0 (OPM7.181205.001) rooted with Magisk 27, Zygisk enabled in Magisk, LSPosed installed and enabled in Magisk and NFCGate Module enabled in LSPosed. Both devices were barely used after the last reset (just used them to capture CellBroadcasts).
I get all green checkmaks in the App.

[image]
[image]
[image]
[image]

However, I have issues with Clone Mode. I can scan and save/load a Tag. Once a Tag is scanned the device stops probing for Tags (validated via Oscilloscope and wire loop). However, it's not responding to other readers. I tried the other Nexus 5X (reading with NFCGate Clone Mode), my Pixel 6a (Wakdev NFC Tools and NXP Taginfo) and an Elatec USB reader (Elatec Director reading ISO14443 UID).
`logcat | grep -e "NFC"` gives me this when reading a tag in clone mode: 
[clone_logcat_nfc.txt](https://github.com/user-attachments/files/15857546/clone_logcat_nfc.txt)

I found that when Clone Mode is open without a Tag read/loaded, the Nexus identifies as FeliCa 64bit to the Elatec Reader and can't be read by my Pixel 6a. Othe Phones I have usually identify as ISO14443A/Mifare 32bit. (probably not relevant)

Capture Mode works, I successfully captured the traffic of NXP Taginfo.

kleest, 2024-06-16:
> Once a Tag is scanned the device stops probing for Tags (validated via Oscilloscope and wire loop)

We disable polling once we emulate a tag, so this should be working as intended.

> I found that when Clone Mode is open without a Tag read/loaded, the Nexus identifies as FeliCa 64bit to the Elatec Reader and can't be read by my Pixel 6a. Othe Phones I have usually identify as ISO14443A/Mifare 32bit. (probably not relevant)

This is actually standard behavior of Android HCE devices. If you scan the phone multiple times with a compatible reader you will see (more or less randomly) NFC-A, NFC-B and NFC-F (FeliCa) since the reader polls for all technologies. When no tag configuration is applied, NFCGate's patch is not enabled which will therefore restore the HCE default behavior.
  
  
  
Regarding your problem: Do you try to emulate a Mifare Classic tag? I noticed a SEL_RES value of 0x08 which indicates that ISO-DEP is not supported (for ISO-DEP bit 0x20 must be set). Emulating Mifare Classic (or Ultralight) tags is sadly not supported.

xsrf, 2024-06-16:
> We disable polling once we emulate a tag, so this should be working as intended.

Yeah, got that. I just wanted to confirm this happens ;)

> Do you try to emulate a Mifare Classic tag?

Yes, according to NXP TagInfo I tried NTAG216 (ultralight), NTAG203(F) (ultralight) and two Mifare Classic.

Thx. It actually worked with my public transport ticket and passport.

I also have a card that triggers the android NFC sound but is not recognized by NFCGate at all (also Mifare Classic acording to NXP TagInfo) and I have a Legic CTC MM410 (ISO14443A / ISO15693) that causes just a white screen and can't be cloned: 

[image]

xsrf, 2024-06-16:
So I guess there is nothing wrong with my setup. I just tried the server and successfully relayed my Deutschlandticket through both Nexus 5X, read by [eTicket Leser](https://play.google.com/store/apps/details?id=de.erichambuch.ticketreader) on my Pixel 6a.
Guess I have to experiment more. It really wasn't obvious that the tag(s) could be the issue.

xsrf, 2024-06-16:
Trying to relay the Legic Advant CTC MM410 when it's read in ISO14443A mode also fails. I get empty rows in the relay view and the pcap only shows 5 Bytes `00 fb 00 01 02` for each of the rows/attempts. Is this something you're interested in?
Otherwise this issue can also be closed. Thx for the help!

kleest, 2024-06-16:
Thanks for testing! The classic tag that is not recognized by NFCGate as well as Legic with empty rows are very interesting. Could you share logs and tag information that is  displayed when reading it with another reader? It's also possible to send it privately by mail (see README) if you don't want to share this data publicly.

xsrf, 2024-06-17:
I'll send you a mail ;) What Tools do you recommend to generate the data you need?
I can send you the output of NXP TagInfo and NFCGate Logs while Taginfo is reading the Tag. Anything else I can do with Android? Should I also have logcat running while doing this?

closed 2024-06-17 by xsrf

kleest, 2024-06-17:
adb logcat with all log entries mentioning "nfc" case-insensitive while you read the tag in clone mode with NFCGate + output from TagInfo (can also be the XML export). We don't need the on-device capture data from NFCGate while TagInfo scans the tag. Thanks!

ildar, 2025-01-21:
> Do you try to emulate a Mifare Classic tag? I noticed a SEL_RES value of 0x08 which indicates that ISO-DEP is not supported (for ISO-DEP bit 0x20 must be set). Emulating Mifare Classic (or Ultralight) tags is sadly not supported.

I don't quite catch it: it's unsupported in the app or because the Android NFC can't ?

P.S. BTW, it's better be indicated in the app.

kleest, 2025-01-25:
The Android NFC stack does not have support for emulating Mifare Classic/Ultralight tags. Showing an indicator that the cloned tag's technology is not supported is already on our backlog.

ildar, 2025-01-25:
that's great, thank you