nfcgate/nfcgate #172: Question on permissions

IzzySoft, 2024-07-07:
The scanner at IoD just reported:

```
! repo/de.tu_darmstadt.seemoo.nfcgate_12.apk declares sensitive permission(s): android.permission.READ_EXTERNAL_STORAGE*
! repo/de.tu_darmstadt.seemoo.nfcgate_12.apk contains signature block blobs: 0x504b4453 (DEPENDENCY_INFO_BLOCK; GOOGLE)
```

Concerning storage permissions, I've just added them to the app's green list – please confirm I got that right:

```yaml
    android.permission.READ_EXTERNAL_STORAGE: imnplicitly granted due to WRITE_EXTERNAL_STORAGE
    android.permission.WRITE_EXTERNAL_STORAGE: needed to export of captured NFC traffic
```

As for `DEPENDENCY_INFO_BLOCK`, that can be easily avoided with a tiny adjustment to your `build.gradle`:

```gradle
android {
    dependenciesInfo {
        // Disables dependency metadata when building APKs.
        includeInApk = false
        // Disables dependency metadata when building Android App Bundles.
        includeInBundle = false
    }
}
```

For some background: that BLOB is supposed to be just a binary representation of your app's dependency tree. But as it's encrypted with a public key belonging to Google, only Google can read it – and nobody else can even verify what it really contains. More details can be found e.g. here: [Ramping up security: additional APK checks are in place with the IzzyOnDroid repo](https://android.izzysoft.de/articles/named/iod-scan-apkchecks#blobs).

Thanks in advance!

kleest, 2024-07-08:
Hi,

> Concerning storage permissions, I've just added them to the app's green list – please confirm I got that right:

Yes. NFCGate has not been adapted to storage changes in newer Android versions yet, that's why these permissions are still there. They are used for exporting pcapng files of captured communication in relay and on-device capture mode, and where needed at the time of development of these features.

Regarding `DEPENDENCY_INFO_BLOCK`, that's new information to me. Thanks for the tip of removing it, I guess this block does not make much sense if we do not upload the APK to GPlay.

IzzySoft, 2024-07-08:
> has not been adapted to storage changes in newer Android versions yet

Once that's the case, the updater here should warn us their green-listing is obsolete and we can remove them again :wink:

> I guess this block does not make much sense if we do not upload the APK to GPlay.

With Play being the only place able to read them: nope, no really use for that. But you know how the folks are there, they always think they're the only ones…

kleest, 2024-12-10:
Regarding `DEPENDENCY_INFO_BLOCK`, this is fixed in the latest release, see #185.

IzzySoft, 2024-12-11:
Updated, confirmed, RB¹ shield up – will go out with the next sync, thanks!

¹ <sub>see: [Reproducible Builds, special client support and more at IzzyOnDroid](https://android.izzysoft.de/articles/named/iod-rbs-mirrors-clients)</sub>

closed 2024-12-11 by IzzySoft