nfcgate/nfcgate #173: Reader doesnot send SELECT command

NgoHuy, 2024-08-09:
**Describe the bug**

Some applications do not send SELECT applet/AID command, then HCE will fail to route to our application. In real implementation, some cards receive select df command directly, when it happens HCE has no information about remote command, then reader terminated connection. We need a hook which routes 1st command to HCE which doesnot not select AID.

**App Information***

status page is green

**Device Information*** Which device are you using and what OS are you running on it?

Device: S21+ Exynos

OS: Android 14

kleest, 2024-08-11:
If I understood you correctly, this is already implemented in NFCGate and should work.

[Here](https://github.com/nfcgate/nfcgate/blob/8f58d7aa25dc88446e076494bd8f4f5ae0def391/nfcd/src/main/jni/src/nfcd.cpp#L137-L138), we convince the system that SELECT AID already happened. And [there](https://github.com/nfcgate/nfcgate/blob/8f58d7aa25dc88446e076494bd8f4f5ae0def391/nfcd/src/main/java/de/tu_darmstadt/seemoo/nfcgate/xposed/Hooks.java#L53-L57), we set NFCGate's custom AID for which the app is registered.
This should route every APDU to NFCGate if the hook is enabled (clone mode, other modes as "tag"), even if the reader did not issue a SELECT AID command.

NgoHuy, 2024-08-11:
In capture mode, when using card emulator, log cannot record 1st command without select AID, the log is empty if reader rejected connection after selecting df instead of AID

NgoHuy, 2024-08-11:
I use nfcgate on reader phone, and make sure that they sent exactly this command. But on hce phone, I didnot see that command until reader resent select AID command. On HCE phone, I only saw response 6A82 or empty if reader didnot resend select AID again.

kleest, 2024-09-05:
I tested it locally on Pixel 7 Android 14. Sending commands without an initial SELECT AID works fine and these APDUs are also properly relayed to the other phone.
Since it affects a couple of users (looks like an issue on Samsung devices), we added better debugging in recent commits that might help narrowing down the issue. Could you please attach logcat logs using this version?

You can find the most recent automatically created APK [in the workflow run details on the actions tab](https://github.com/nfcgate/nfcgate/actions/workflows/build-android.yml).

NgoHuy, 2024-09-05:
I think your application send APDU without verifying data, in this case, if reader didnot terminate after sending select DataFile, it continue working as expected. But in this case reader didnot send AID command and sent some select DataFile, but android does not know which applet will handle this data. It terminated connection here.
I will rebuild on my side and feedback later.

NgoHuy, 2024-09-11:
It works as expected. My purpose is different.

closed 2024-09-11 by NgoHuy

NgoHuy, 2024-09-11:
To recap why I closed this issue:
When reader didnot send select applet command, findSelectAid was never called, then the code patched ce_select_t4t is useless. 
My purpose is send all adpu to my application, therefore I think it works as expected, not nfcgate's bug

Allen-Delone, 2024-09-18:
Dear experts, can someone tell me if this app is working now? I am collecting it, but purchases do not pass, someone can help me to collect it, I will financially pay for the help. Contact me please