nfcgate/nfcgate #176: NFCGate and "NGate" Malware

roussosalex, 2024-09-01:
As some of you may have heard, researchers at ESET [published their findings](https://www.welivesecurity.com/en/eset-research/ngate-android-malware-relays-nfc-traffic-to-steal-cash/) about a novel family of Android malware they named "NGate", which tricks victims into relaying NFC traffic to facilitate unauthorized ATM withdrawals. We were contacted by the research team prior to the publication and appreciate their professionalism in bringing this to our attention.

Regrettably, the attackers have copied significant portions of source code from NFCGate when creating the malware. After tricking the victims into entering their PIN, they covertly used the relay functionality of NFCGate to send the traffic of the victim's payment card to the attacker's device. This allowed the attacker to conduct an unauthorized ATM transaction and withdraw money from the victim's bank account. All of the recent malware scanner reports (see #164) are related to this discovery.

As an open-source project, NFCGate is designed to support legitimate NFC research, and we are deeply troubled by its misuse in this case.  We are currently investigating the situation thoroughly and are looking for ways to implement additional safeguards to help prevent this from happening in the future. Unfortunately, we cannot control how the open-source code is being utilized by bad actors.  

Furthermore, we strongly encourage everyone to remain vigilant when interacting with unfamiliar banking apps or responding to unsolicited communications. If you encounter any suspicious activity related to NFCGate, please contact us directly via email (see [README.md](../blob/v2/README.md#contact)).

We are committed to transparency and will keep the community informed as we continue to address this issue. Thank you for your continued support!

ETennDude, 2025-07-04:
How funny because I just downloaded it today JULY 4th 2025 and it is saying it is maleware

[image]
[image]

jficz, 2026-10-08:
looks like the security morons infected other banks, right @Ceskasporitelna?