nfcgate/nfcgate #178: A51 A34

Art4Islami, 2024-09-02:
**App Information***

Please provide us with a screenshot of the "status" page in NFCGate.

**Device Information*** I use a A34 and A51

Device: [e.g. Google Pixel 7]
[image]
O
[image]

OS: [Andorid 10 A51 and android 14 a34]

**Additional context**

Add any other relevant information for troubleshooting the problem here.

Art4Islami, 2024-09-02:
One time it worked on my a34

roussosalex, 2024-09-02:
Thanks for reporting this issue! Please attach a boot log (system logcat since boot). See [here](https://www.xda-developers.com/how-to-take-logs-android/#system-logs) for detailed instructions as required by the issue template. Also, could you provide some details on how it worked "one time" on the A34? When did it work, how did it work and what changed to prevent it from working now? Did it ever work on the A51?

Art4Islami, 2024-09-03:
> Thanks for reporting this issue! Please attach a boot log (system logcat since boot). See [here](https://www.xda-developers.com/how-to-take-logs-android/#system-logs) for detailed instructions as [...]

I was one the relay and connected my a51 with a34 used the a34 as tag then reader a51 for the boot log i dont have a computer and does it need root cus my phones arent rooted

roussosalex, 2024-09-03:
Please see the [README](../?tab=readme-ov-file#requirements-for-specific-modes) about the requirements for the different modes. 

Relay mode works best when at least the tag phone is rooted. It is **technically** possible to use it without root, but then it would not be able to:
* clone the static tag data (e.g. NFCID), which would no longer work with all readers
* "catch all" AIDs for the reader. You would need to know the AIDs required by your reader and put them in the [hce file](../blob/v2/app/src/main/res/xml/hce.xml), then recompile and re-install the application.

Art4Islami, 2024-09-04:
> Please see the [README](../?tab=readme-ov-file#requirements-for-specific-modes) about the requirements for the different modes.
>
> Relay mode works best when at least the tag phone is rooted. It is [...]

Thank you very much i will rooted at least one phone

closed 2024-09-04 by Art4Islami