nfcgate/nfcgate #179: LG ThinQ 5G Verizon native hook not enabled

NgoHuy, 2024-09-11:
**App Information***

[image]

**Device Information*** Which device are you using and what OS are you running on it?

Device:  LG ThinQ 5G

OS: android 12

**Log***

[image]

**Additional context**

I didnot see any NATIVENFC log on my device, I build latest v2 branch, reboot did not help

roussosalex, 2024-09-27:
Thanks for contributing! Please try the newest commit and report back, it should now work with your LG device as well.

closed 2024-09-27 by roussosalex

NgoHuy, 2024-10-22:
it failed for native nfc.
[image]

kleest, 2024-10-22:
@NgoHuy Can you please show logs containing NATIVENFC?

imyxh, 2025-01-15:
Same issue on alphalm (LG G8 ThinQ). Output of `logcat | grep NFC` attached.

I think libnfc is one of these (both attached):

```
/system/system_ext/lib64/libLGnfc_nxp_jni.lge.so
/system/system_ext/lib64/libnfc-nxp-nci.lge.so
```

but it seems neither is picked up as a candidate to findLibNFC.

[libnfc.zip](https://github.com/user-attachments/files/18427868/libnfc.zip)

[nfc.log](https://github.com/user-attachments/files/18427852/nfc.log)