nfcgate/nfcgate #180: Native hook could not be found on POCO X4 GT

RandLily, 2024-09-27:
**App Information***

[image]

**Log***

just
09-27 17:31:19.898 16953 17017 I NATIVENFC: Command: RESET_CONFIG
09-27 17:31:19.927 16953 17017 I NATIVENFC: Command: GET_HOOK_STATUS
09-27 17:32:50.224 16953 17017 I NATIVENFC: Command: GET_HOOK_STATUS
09-27 17:36:41.875 16953 17017 I NATIVENFC: Command: GET_HOOK_STATUS
09-27 17:36:48.786 16953 17017 I NATIVENFC: Command: RESET_CONFIG
09-27 17:36:48.815 16953 17017 I NATIVENFC: Command: GET_HOOK_STATUS
09-27 17:38:40.804 16953 17017 I NATIVENFC: Command: GET_HOOK_STATUS
09-27 17:38:55.275 16953 17017 I NATIVENFC: Command: RESET_CONFIG
09-27 17:38:55.306 16953 17017 I NATIVENFC: Command: GET_HOOK_STATUS
09-27 17:38:59.419 16953 17017 I NATIVENFC: Command: RESET_CONFIG
09-27 17:39:00.321 16953 17017 I NATIVENFC: Command: RESET_CONFIG

RandLily, 2024-10-11:
It seems that all xiaomi phone can't use native hook

kleest, 2024-10-12:
According to https://github.com/nfcgate/nfcgate/issues/181, at least the Poco X3 Pro is working fine. Please attach logs that contain NATIVENFC and HOOKNFC.

RandLily, 2024-10-14:
> According to #181, at least the Poco X3 Pro is working fine. Please attach logs that contain NATIVENFC and HOOKNFC.

10-14 14:00:56.529  6461  6461 D NATIVENFC: findLibNFC: candidate contains 'nfc', checking symbols: /data/app/~~y-4PlRtFuqo74gu2yu96kQ==/de.tu_darmstadt.seemoo.nfcgate-VL5gs-f4ZuyWvLOVZhOy6w==/lib/arm64/libnfcgate.so
10-14 14:00:56.732  6461  6461 E NATIVENFC: Error at nfcd/src/main/jni/src/nfcd.cpp:150
10-14 14:00:56.732  6461  6461 E NATIVENFC: Library not found or not accessible
10-14 14:03:01.021  6461  8629 I NATIVENFC: Command: RESET_CONFIG
10-14 14:03:01.041  6461  8629 I NATIVENFC: Command: GET_HOOK_STATUS

10-14 14:13:32.603  6497  6497 I HOOKNFC : Specific NfcService selected: NfcService
10-14 14:13:34.124  6497  6497 I HOOKNFC : NfcService constructor called: public com.android.nfc.NfcService(android.app.Application)
10-14 14:13:34.128  6497  6497 I HOOKNFC : Launching InjectionBroadcastWrapper
10-14 14:13:39.001  6497  6497 I HOOKNFC : Native hook failed (for now)

cat /proc/6461/maps | grep nfc
7e156b2000-7e16200000 rw-p 00000000 fe:33 295593                         /data/app/~~y-4PlRtFuqo74gu2yu96kQ==/de.tu_darmstadt.seemoo.nfcgate-VL5gs-f4ZuyWvLOVZhOy6w==/oat/arm64/base.vdex
7e1705c000-7e171d4000 r-xp 00000000 fe:33 301596                         /data/app/~~y-4PlRtFuqo74gu2yu96kQ==/de.tu_darmstadt.seemoo.nfcgate-VL5gs-f4ZuyWvLOVZhOy6w==/lib/arm64/libnfcgate.so
7e171d4000-7e171e0000 r--p 00177000 fe:33 301596                         /data/app/~~y-4PlRtFuqo74gu2yu96kQ==/de.tu_darmstadt.seemoo.nfcgate-VL5gs-f4ZuyWvLOVZhOy6w==/lib/arm64/libnfcgate.so
7e171e0000-7e171e1000 rw-p 00182000 fe:33 301596                         /data/app/~~y-4PlRtFuqo74gu2yu96kQ==/de.tu_darmstadt.seemoo.nfcgate-VL5gs-f4ZuyWvLOVZhOy6w==/lib/arm64/libnfcgate.so
7e2f733000-7e2f857000 r--s 0069d000 fe:33 311405                         /data/app/~~y-4PlRtFuqo74gu2yu96kQ==/de.tu_darmstadt.seemoo.nfcgate-VL5gs-f4ZuyWvLOVZhOy6w==/base.apk
7f53a24000-7f53a44000 r--s 00000000 00:10 473                            /dev/__properties__/u:object_r:vendor_nfc_mi_prop:s0
7f53d01000-7f53d21000 r--s 00000000 00:10 288                            /dev/__properties__/u:object_r:nfc_prop:s0
7f5d36a000-7f5d386000 r--s 007db000 fe:33 311405                         /data/app/~~y-4PlRtFuqo74gu2yu96kQ==/de.tu_darmstadt.seemoo.nfcgate-VL5gs-f4ZuyWvLOVZhOy6w==/base.apk
7f610f4000-7f61103000 r--p 00000000 fe:0a 6563964                        /system_ext/framework/com.nxp.nfc.nq.jar
7f641c2000-7f641c6000 r--p 00000000 fe:0a 6591595                        /system_ext/framework/com.xiaomi.nfc.jar
7f6f35b000-7f6f35c000 rw-p 00000000 fe:0a 7187188                        /system_ext/framework/oat/arm64/com.xiaomi.nfc.vdex
7f6f35e000-7f6f35f000 rw-p 00000000 fe:0a 7185536                        /system_ext/framework/oat/arm64/com.nxp.nfc.nq.vdex

roussosalex, 2024-10-14:
Thanks for the info! 
Please attach the following information to help us investigate this issue:

1. Can you confirm that your system contains a file similar to  `/system/system_ext/lib64/libsn100nfc-nci-v1.so`?
2. Output of `/proc/../maps` for the `com.android.nfc` process directly after using NFC in any way (like reading a tag). You can filter it for `nfc` **case insensitive** (`grep -i`). This output should contain the file from the first step.
3. Boot log (reboot and try to clone a tag with NFCGate, then collect logcat). You can filter it for `nfc|HOOKNFC|NATIVENFC`.

RandLily, 2024-10-15:
cat /proc/6497/maps | grep -i nfc
38010000-3804c000 rw-p 00000000 00:00 0                                  [anon:dalvik-/data/dalvik-cache/arm64/system@app@NQNfcNci@NQNfcNci.apk@classes.art]
7cb3533000-7cb4081000 rw-p 00000000 fe:33 295593                         /data/app/~~y-4PlRtFuqo74gu2yu96kQ==/de.tu_darmstadt.seemoo.nfcgate-VL5gs-f4ZuyWvLOVZhOy6w==/oat/arm64/base.vdex
7cb6255000-7cb63cd000 r-xp 00000000 fe:33 301596                         /data/app/~~y-4PlRtFuqo74gu2yu96kQ==/de.tu_darmstadt.seemoo.nfcgate-VL5gs-f4ZuyWvLOVZhOy6w==/lib/arm64/libnfcgate.so
7cb63cd000-7cb63d9000 r--p 00177000 fe:33 301596                         /data/app/~~y-4PlRtFuqo74gu2yu96kQ==/de.tu_darmstadt.seemoo.nfcgate-VL5gs-f4ZuyWvLOVZhOy6w==/lib/arm64/libnfcgate.so
7cb63d9000-7cb63da000 rw-p 00182000 fe:33 301596                         /data/app/~~y-4PlRtFuqo74gu2yu96kQ==/de.tu_darmstadt.seemoo.nfcgate-VL5gs-f4ZuyWvLOVZhOy6w==/lib/arm64/libnfcgate.so
7cbf0d8000-7cbf135000 r--p 01756000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7cbf135000-7cbf1a2000 r-xp 017b3000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7cbf1a2000-7cbf1a9000 r--p 01820000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7cbf1a9000-7cbf1aa000 rw-p 01826000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7cc0e92000-7cc0ee3000 r--p 015f3000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7cc0ee3000-7cc0fb2000 r-xp 01644000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7cc0fb2000-7cc0fb6000 r--p 01713000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7cc0fb6000-7cc0fb8000 rw-p 01716000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7cc65f0000-7cc6b90000 r--p 004cf000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7ccfae0000-7ccfb28000 r--p 014ed000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7ccfb28000-7ccfbd8000 r-xp 01535000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7ccfbd8000-7ccfbdb000 r--p 015e5000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7ccfbdb000-7ccfbe0000 rw-p 015e7000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7ccfc02000-7ccfca3000 r--p 00d24000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7ccfca3000-7ccfd40000 r-xp 00dc5000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7ccfd40000-7ccfd46000 r--p 00e62000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7ccfd46000-7ccfd47000 rw-p 00e67000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d25529000-7d25530000 r--p 00b95000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d25530000-7d2553a000 r-xp 00b9c000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d2553a000-7d2553b000 r--p 00ba6000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d2553b000-7d2553c000 rw-p 00ba6000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d2556b000-7d25588000 r--p 00f2c000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d25588000-7d255bc000 r-xp 00f49000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d255bc000-7d255be000 r--p 00f7d000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d255be000-7d255bf000 rw-p 00f7e000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d255d6000-7d255e3000 r--p 00bfe000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d255e3000-7d255f1000 r-xp 00c0b000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d255f1000-7d255f4000 r--p 00c19000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d255f4000-7d255f5000 rw-p 00c1b000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d25604000-7d25610000 r--p 01884000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d25610000-7d2561f000 r-xp 01890000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d2561f000-7d25621000 r--p 0189f000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d25621000-7d25622000 rw-p 018a0000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d25640000-7d25656000 r--p 0171d000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d25656000-7d25678000 r-xp 01733000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d25678000-7d25679000 r--p 01755000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d25695000-7d256a4000 r--p 00ba8000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d256a4000-7d256ba000 r-xp 00bb7000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d256ba000-7d256bd000 r--p 00bcd000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d256bd000-7d256be000 rw-p 00bcf000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d256de000-7d256e8000 r--p 01852000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
7d256e8000-7d256f3000 r-xp 0185c000 fe:0b 97865984                       /system/app/NQNfcNci/NQNfcNci.apk
[... cut ...]
7df729f000-7df72a0000 rw-p 00000000 fe:0a 7187188                        /system_ext/framework/oat/arm64/com.xiaomi.nfc.vdex
7df72a2000-7df72a3000 rw-p 00000000 fe:0a 7185536                        /system_ext/framework/oat/arm64/com.nxp.nfc.nq.vdex

[NQNfcNci.zip](https://github.com/user-attachments/files/17371258/NQNfcNci.zip)
[log.txt](https://github.com/user-attachments/files/17371827/log.txt)

The `libsn100nfc-nci-v1.so` is in `/system/app/NQNfcNci/NQNfcNci.apk` ，but this so is not in `com.android.nfc` and nfcgate maps，seems that just load the apk but not unzip

pakistan699, 2024-10-20:
Una pregunta a los expertos yo descargue el programa y mi teléfono es un
POCO X5 Pro 5G será que está versión si funciona les agradezco la ayuda y
el apoyo

pakistan699, 2024-10-20:
A question to the experts. I downloaded the program and my phone is a
POCO X5 Pro 5G. Will it work? If it works, I appreciate the help and
support.

deep2web, 2024-10-29:
Do you use the Stock ROM?

My Poco X3 Pro, which worked, use a custom ROM.

This explain maybe the different behavior.

roussosalex, 2024-11-05:
@RandLily Please try the experimental version from the [v2-dev-debug-maps branch](nfcgate/nfcgate/tree/v2-dev-debug-maps).

Please report if it works or not and attach the (now much more verbose) log as well.
[NfcGate_dev_debug_maps.zip](https://github.com/user-attachments/files/17626029/NfcGate_dev_debug_maps.zip)

RandLily, 2024-11-05:
@roussosalex 
It worked !!!Thank you very much, but a new problem came up, when I use two POCO X4 GT, it doesn't work properly
But it works fine when I use a POCO X4 GT and a Pixel 6，here is the log of POCO X4 GT
[log.txt](https://github.com/user-attachments/files/17628150/log.txt)

kleest, 2024-12-10:
The changes from the experimental version are now available in the newest release [v2.4.3](https://github.com/nfcgate/nfcgate/releases/tag/v2.4.3).
If you still face any problems, please create a new issue.

closed 2024-12-10 by kleest