nfcgate/nfcgate #181: Poco X3 Pro "Xposed module not working properly"

deep2web, 2024-10-09:
**App Information***

[image]

**Device Information*** Which device are you using and what OS are you running on it?

Device: Poco X3 Pro 8GB 256GB

OS: [image]

**Log***

[image]

[image]

**Additional context**

LSPosed: [image]

HCEF unlocker is not installed, because NFCGate shows HCE as enabled.

kleest, 2024-10-10:
Could you please try out the latest v2 branch version? Please see the [actions page](https://github.com/nfcgate/nfcgate/actions) for downloads.

deep2web, 2024-10-10:
Thanks! With the latest V2 branch version everything now works perfect!

closed 2024-10-10 by deep2web