nfcgate/nfcgate #182: oneplus 12r/ace3 nfc SN220U

bm16ton, 2024-11-30:
Hello, firstly great project, thankyou truely. i have a oneplus 12r. The chip id is c1, i added it to the list with SN220U found from  /proc/oplus_nfc/chipset .Project version is found at /proc/oplusVersion/prjName and i added it to the oppo list. That was ez enough but the app still shows it doesnt know the nfc chipset, if i make guess = SN220U it obviously works. I have no files named libnfc-nxp.conf_$PROJECTNUM but i do have libnfc-nxp.conf_$PROJECTNUM. Here is where my programming experience (lack there of) falters.Not sure how to add the new filename tho have a couple test versions compiled but dunno if it found it, some log somewhere might say? Im not sure why its not getting the name from the list using chipid c1. I was able to get chip id from   /odm/bin/nqnfcinfo. Any chance nfcgate doesnt see the chipid? it didnt immediatly seem clear to me where the chip id was being culled from. If i figure it out ill send a PR.

bm16ton, 2024-12-03:
> Thanks for reporting! Please test the newest version of the v2 branch and
> let us know if that did anything. We added chip name detection based on the
> name of the firmware on Oppo/OnePlus devices.
>
> Regarding /proc/oplus_nfc/chipset and /odm/bin/nqnfcinfo: these detection
> options exist, but have not (yet) been added to NFCGate, because we would
> like to keep the chip detection rootless. We might add an additional
> root-only detector in the future for nq-nci specifically.
>
> Regarding c1: Since the NXP config file either does not exist, or does
> not contain any chip type, no guess is made. This chip ID may also not be
> the same as the ones we have seen in other config files.
>
> Does NFCGate work for you, aside from the chip detection ? If so, you
> could always send a PR for the compatibility matrix.

Hello and thank you repllying. First off yes as far as i can tell its
working! Only done aimple things like read. Any place for me to put keys i
have for various cards like disney infinity characters? It probly sounds
silly but itd be one more thing i could test easily. I just found my
oneplus 6t gonna try v2 amd also the pass through option with them. Only
thing is i think i only have mifare devices and i think i read some of them
dont work with pass through, not sure i have compatible onea or not. I will
try and find time tomorrow to test and report back.

bm16ton, 2024-12-04:
i was able to test the new v2 and it worked! Not only did it detect the chipset but it no longer requires me to turn off then back on  the nfc before working! Hopefully test passthrough with oplus 6t tomorrow. Thank you!

kleest, 2024-12-10:
The chip name detection changes are now available in the newest release [v2.4.3](https://github.com/nfcgate/nfcgate/releases/tag/v2.4.3).

bm16ton, 2024-12-14:
I didnt forget yeah, so i assume non of my cards were compatible with the relay option, it indeed relayed info to second phone, and i assume the 2nd phone (tag) was suposed to emulate the tag. The emulation part did not seem to work. I ordered a proxmart 3 from china with sum cards, hopefully shipping wont take to long. The same card/same phone when booting into linux emulated a couple different card ok so hardwares capable. Not sure if Oneplus hinders capabilities in firmware etc. I like the idea of man in the middle after hearing the issues with antennas and there placement using proxmark 3 for sniffing. Ill give it the pepsi coke challange hopefully before christmas.

closed 2025-03-03 by roussosalex