nfcgate/nfcgate #184: Native hook could not be found on Android 15

linsui, 2024-12-09:
**Describe the bug**

After upgrading my LineageOS to Android 15, NGCGate can't find the native hook anymore.

**App Information***

Please provide a screenshot of the "status" page
[image]

**Device Information*** Which device are you using and what OS are you running on it?

Device: Xiaomi Mi 12 pro

OS: Android 15 LineageOS

**Log***

Attach a boot log (system logcat since boot).
See [here](https://www.xda-developers.com/how-to-take-logs-android/#system-logs) for detailed
instructions.
[modules_2024-12-09T23:16:01.49925.log](https://github.com/user-attachments/files/18063847/modules_2024-12-09T23.16.01.49925.log)

**Steps to reproduce**

Please describe the steps taken to reproduce the problem.
Be as descriptive as can you be, e.g.:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**

A clear and concise description of what you expect to happen.

**Observed/unwanted behavior**

What is the behavior that is occurring?

**Screenshots**

Optionally, add screenshots to help show the problem.

**Additional context**

Add any other context about the problem here.

**Proposed solution**

If you have thought about a solution for the problem, please elaborate here!

roussosalex, 2024-12-09:
Thanks for contributing! Have you tried the newest commit on [v2](/nfcgate/nfcgate/tree/v2)?

linsui, 2024-12-09:
I just tested the apk built by the CI. It works! Thanks!

closed 2024-12-09 by linsui