nfcgate/nfcgate #193: Repeated crashes while not using app

gbakeman, 2025-01-02:
**Describe the bug**

I noticed my phone's battery being drained recently, which led me to discover that the ADB crash log is being filled with crashes that appear to be initiated from NFCGate. I haven't been using the app at all, and NFC capability is turned off in my phone for now.

**App Information**

```
Version: 2.4.4 (e9fe1f2064)
Device Name: ONEPLUS A6013 [OnePlus6T] (OnePlus6T)
Android Version: 9
Build Number: ONEPLUS A6013_41_190801
NFC Enabled: No
HCE Capability: Yes
Xposed Module Enabled: Yes
Native Hook Enabled: No
NFC Chip: NXP PN80T
```

**Logs**

[crashlog3.txt](https://github.com/user-attachments/files/18293903/crashlog3.txt)
[crashlog4.txt](https://github.com/user-attachments/files/18293904/crashlog4.txt)
[crashlog5.txt](https://github.com/user-attachments/files/18293905/crashlog5.txt)
[crashlog1.txt](https://github.com/user-attachments/files/18293901/crashlog1.txt)
[crashlog2.txt](https://github.com/user-attachments/files/18293902/crashlog2.txt)

**Steps to reproduce**

Install latest version of app as far as I know. Sorry I can't be more help, but I'll provide more information if requested.

kleest, 2025-01-05:
Hi @gbakeman , thank you for reporting this issue! This should be fixed now. Please check the latest debug version from https://github.com/nfcgate/nfcgate/actions/runs/12614882588

closed 2025-01-05 by kleest

gbakeman, 2025-01-05:
Fantastic work getting a fix out and back to me so quickly, thank you! I've had the debug build installed in my phone since yesterday. I checked adb today and no sign of excessive crash logging, and by phone battery did well over the day. I think you can consider this fixed!