nfcgate/nfcgate #196: LG G8 ThinQ native hook issues

kleest, 2025-01-25:
> Same issue on alphalm (LG G8 ThinQ) on v2.4.4. Output of `logcat | grep NFC` attached.> 
> > 
> I think libnfc is one of these (both attached):> 
> > 
> ```> 
> /system/system_ext/lib64/libLGnfc_nxp_jni.lge.so> 
> /system/system_ext/lib64/libnfc-nxp-nci.lge.so> 
> ```> 
> > 
> but it seems neither is picked up as a candidate to findLibNFC.> 
> > 
> [libnfc.zip](https://github.com/user-attachments/files/18427868/libnfc.zip)> 
> > 
> [nfc.log](https://github.com/user-attachments/files/18427852/nfc.log)> 
> > 
> Fwiw I could find these by simply scanning /proc/*/maps, so that might be a more robust option to scan for libs instead of whatever is currently happening. 

 _Originally posted by @imyxh in [#179](https://github.com/nfcgate/nfcgate/issues/179#issuecomment-2593461968)_

kleest, 2025-01-25:
@imyxh It looks like the log is truncated during the candidate checking phase. Could you please comment out
```
LOGD("findLibNFC: candidate: %s", candidate.c_str());
```
https://github.com/nfcgate/nfcgate/blob/27687777355d02c4c61c21ccac298af2e7b157fb/nfcd/src/main/jni/src/nfcd.cpp#L199

and attach a new log?

> Fwiw I could find these by simply scanning /proc/*/maps, so that might be a more robust option to scan for libs instead of whatever is currently happening.

We moved away from /proc/*/maps in e5205b035f04f64b6248c0a686d7a9fadd207f4e because of issues with some devices (see #180). In general, using `dl_iterate_phdr` should be more robust.

imyxh, 2025-01-25:
Sure. Here's what I did:

1. Build latest commit (27687777355d02c4c61c21ccac298af2e7b157fb), with that line commented. Reinstall, enable LSposed, reboot. Open app, click on "Status" from the sidebar. App crashes.
2. Build latest commit (27687777355d02c4c61c21ccac298af2e7b157fb), without that line commented. Reinstall, enable LSposed, reboot. Open app, click on "Status" from the sidebar. App crashes.
3. Build v2.4.4 (06d934a5be7dc894063be1365b3715697f11c391) with the line commented. Reinstall, enable LSposed, reboot. No crash, but native hook still fails, as expected.

`logcat | grep -i nfc` for all three attempts attached. In the log for attempt 3:

```
01-25 00:55:45.193  5865  5865 D NATIVENFC: findLibNFC: candidate contains 'nfc', checking symbols: /data/dalvik-cache/arm64/system@system_ext@app@LgNfc@LgNfc.apk@classes.dex
01-25 00:55:45.194  5865  5865 E NATIVENFC: Error at nfcd/src/main/jni/src/nfcd.cpp:150
01-25 00:55:45.194  5865  5865 E NATIVENFC: Library not found or not accessible
01-25 00:55:45.194  5865  5865 I HOOKNFC : Native hook failed (for now)
```

so it does seem like neither libnfc-nxp-nci.lge.so nor libLGnfc_nxp_jni.lge.so are picked up.

[nfc_git_comment.log](https://github.com/user-attachments/files/18545025/nfc_git_comment.log)
[nfc_git_nocomment.log](https://github.com/user-attachments/files/18545026/nfc_git_nocomment.log)
[nfc_v244_comment.log](https://github.com/user-attachments/files/18545024/nfc_v244_comment.log)

imyxh, 2025-01-25:
Actually even if I revert to right before e5205b035f04f64b6248c0a686d7a9fadd207f4e and comment out that log line, it can't find it:

```
01-25 01:44:25.748  5968  5968 D NATIVENFC: findLibNFC: candidate contains 'nfc', checking symbols: /data/app/~~n6Yeg4ec5ah8Swz6sZ0Iag==/de.tu_darmstadt.seemoo.nfcgate-9s5G9Q9piLe_y7qenRYZJQ==/lib/arm64/libnfcgate.so
01-25 01:44:25.775  5968  5968 E NATIVENFC: Error at nfcd/src/main/jni/src/nfcd.cpp:150
01-25 01:44:25.775  5968  5968 E NATIVENFC: Library not found or not accessible
```

So how come I can with `grep -rns -e libnfc -e libLGnfc /proc/*/maps` in a root termux shell?

Right, why doesn't nfcgate trigger a magisk popup for root when it does this?

imyxh, 2025-01-25:
Oh, I didn't realize nfcgate was only checking /proc/self/maps, and not all of /proc/*/maps. Maybe an option to scan the latter would be prudent, or a way to override the path to libnfc manually.

(Still not sure whether we need to hook libnfc or libLGnfc or both. IIRC both had some symbols that seemed relevant. They're in libnfc.zip in my original post.)

kleest, 2025-01-25:
Thanks for the logs and your quick response!

NFCGate uses an Xposed module to inject the libnfcgate.so library into the NFC service. Being in the NFC service process, the NFCGate code scans for the symbols. That's why there is no need for requesting root access with su and scanning only this process (previously `/proc/self/maps`, now `dl_iterate_phdr`) is sufficient.

Looking at the logs, it seems like the JNI library containing the required symbols is loaded *after* libnfcgate.so, hence, the symbols are not yet available. That's certainly new behavior, we will work on a fix.

kleest, 2025-01-25:
> Also notice that the status page lists "Version 2.4.5-dev (https://github.com/nfcgate/nfcgate/commit/e9fe1f2064b04a427dd8ab8201c17d6d4b49e951)" which is very odd—did I need to clean some cache somewhere?

That's odd and indeed looks like some caching problem. However, you don't need to clean caches manually and clicking the "run" button should work. Checking the following option solves some problems:

[image]

imyxh, 2025-01-25:
Oh, I wasn't using Android Studio, just `./gradlew assembleDebug`. I did `git reset --hard v2.4.4` to get to the old version, but forgot that untracked files stay in a hard reset, so I should probably `gradlew clean` or something in the future.

roussosalex, 2025-03-25:
Hello, we added the option for NFCGate to recover from a failed hook. That should fix this issue. Could you please try the newest 
commit? (get a debug build [here](https://github.com/nfcgate/nfcgate/actions/runs/14048956992))

imyxh, 2025-03-27:
Hmm, still shows native hook failed, and `logcat | grep NFC` is pretty quiet now....

[20250327_logcat_grep_NFC.txt](https://github.com/user-attachments/files/19495171/20250327_logcat_grep_NFC.txt)

kleest, 2025-03-28:
Since there is  no mention of HOOKNFC or NATIVENFC, the log may be truncated or the Xposed module not correctly loaded. Could you please try restarting the NFC service will using logcat? `killall com.android.nfc` in a root shell terminates the NFC service and the system should automatically restart it.

roussosalex, 2025-04-28:
@imyxh Do you have an update?

closed 2025-05-05 by roussosalex