nfcgate/nfcgate #204: Hello! Currently, I'm encountering an issue in the Relay HCE

kyhns7, 2025-03-13:
Hello! Currently, I'm encountering an issue in the Relay HCE. The processCommandApdu method of the ApduService returns a value indicating that the reader needs to wait, but the reader does not support waiting. Can I change the asynchronous data requirement to a synchronous waiting for the response data, using the wait method for blocking in the processCommandApdu method, and then wake up the processCommandApdu method using the notify method after the data is responded to in the sendResponse method? Will this cause any thread safety or deadlock issues? The modified code is as follows:

    @Override
    public synchronized byte[] processCommandApdu(byte[] apdu, Bundle extras) {
        Log.d(TAG, "APDU-IN: " + Utils.bytesToHex(apdu));

        // Package the ADPU into a NfcComm object
        NfcComm nfcData = new NfcComm(false, apdu);

        // Send the object to the handler
        mNfcManager.handleData(false, nfcData);

        try {
            this.wait(5000);
        } catch (InterruptedException e) {
            return DONT_RESPOND;
        }

        // Tell the HCE implementation to wait
        return DONT_RESPOND;
    }

    public synchronized void sendResponse(byte[] apdu) {
        Log.d(TAG, "APDU-OUT: " + Utils.bytesToHex(apdu));
        sendResponseApdu(apdu);

        this.notifyAll();
    }

kleest, 2025-03-23:
Since `processCommandApdu` is executed on the main thread of the application you should not block the execution of this method.

However, this method was returning a wrong value which caused issues with your reader. I pushed [a commit](a7dcb5a4e3dd85327a7fa9d0823cae222fc38084) that fixes the return value. Can you please try the [CI build artifact](https://github.com/nfcgate/nfcgate/actions/runs/14019785313) and report back if your issue still exists?

Valmon011001, 2025-03-23:
Greetings dear companions, faced with a problem, I want to emulate cards without getting ROOT rights on android, but through HCE emulation of cards, as a wallet NFC - who has already realized such an idea? Or still no one could not and faced a lot of problems like me? If someone who could, please share please

roussosalex, 2025-03-25:
@Valmon011001 Please open a new discussion for your topic.

closed 2025-04-28 by roussosalex