nfcgate/nfcgate #209: Native hook issues and NFC is disabled or unsupported

exyxwd, 2025-04-09:
**App Information***

After I enable the module in LSPosed and reboot:
[image]

**Device Information***

Device: Samsung Galaxy A13 (SM-A137F/DSN)

OS: Android 13 (TP1A.220624.014.A137FXXS3CWL1)

LSPosed: 1.9.2 Framework version

Magisk: 28.1

**Log***

log on `killall com.android.nfc`:
[log.txt](https://github.com/user-attachments/files/19662123/log.txt)

**Additional context**

NFC is on before reboot and according to the quick settings panel it is on after reboot too. However in the settings "NFC and contactless payments" is toggled off after reboot and does not remain on even after I enable it. I don't have any NFC issues when the module is not enabled.

exyxwd, 2025-04-29:
Thanks for getting back to me!

I tried several versions and builds including commit nfcgate/nfcgate@3ae79ec, all produced this problem. Here are the logs for the CI debug version you mentioned: [log_debug_build.txt](https://github.com/exyxwd/nfcgate/blob/elf-patch/log_debug_build.txt)

I managed to patch the "NFC is disabled or unsupported" issue so that warning does not appear anymore, but the Native Hook is still not enabled: exyxwd/nfcgate@12d36a9

Here are the logs for my patch: [log_own_patch.txt](https://github.com/exyxwd/nfcgate/blob/elf-patch/log_own_patch.txt)

roussosalex, 2025-04-29:
Looking at your logs, it seems this may be a bug with the `DT_GNU_HASH` parsing on 32-bit ARM devices. Could you try changing `ELFMemoryParser.h:127`, just as a quick test:
```diff
-                 (uint8_t *)header + sizeof(DTGnuHeader) + (sizeof(uint64_t) * header->bloom_size);
+                 (uint8_t *)header + sizeof(DTGnuHeader) + (sizeof(uint32_t) * header->bloom_size);
```

exyxwd, 2025-04-30:
This solved the problem! Both NFC and Native Hook is enabled now and the relay worked perfectly in tag emulator mode. Thank you so much, I couldn't have figured out the problem without your help, since I don't have much experience in this area.

kleest, 2025-05-01:
Hi @exyxwd, the next release will include the fix. You can of course already use the debug APK generated by the CI. Thank you for reporting this issue!

closed 2025-05-01 by kleest