nfcgate/nfcgate #211: OnePlus Ace 3 - XPosed Native Hook cannot be enabled

mickey584, 2025-05-05:
Device: [OnePlus Ace 3]

OS: [ Android 15, PJE110_15.0.0.701(CN01)]

[image]

**Log***

[logcat.txt](https://github.com/user-attachments/files/20032506/logcat.txt)

**Additional context**

I use KernelSu 1.0.5 + Zygisk-Lsposed 1.10.1 + Zygisk Next 1.2.8. Everything was working fine on [ColorOS 14/android 14]. After upgrading to [ColorOS 15/android 15], Native Hook keeps showing NO. No matter what I do, I can't seem to find the hook. I've tried various switches and restarting the phone, but nothing worked. I've also tried different NFCgate versions. Could you please help me figure out where the problem is? Thank you!

mickey584, 2025-05-05:
Bug Reproduction: I tried to relay tag mode with a contactless payment card, but the POS device did not recognize it as a payment card. All my settings were correct. It worked fine on other devices, but not on this [OnePlus ace3] phone. The problem should be with the native hook.

roussosalex, 2025-05-05:
Unfortunately, the log does not show the attempted hook install. Please post another log either from boot or after terminating "com.android.nfc" service (with `killall -9 com.android.nfc` in a root shell). 

We also added a new commit to prevent the crash when trying to relay NFC after the native hook did _not_ install properly, you could try that out as well.

mickey584, 2025-05-05:
Thanks for your answers，I just tried the new commit, but there is still no native hook. When I use the NFC relay (tag), there are many error messages such as: [nfcd] Failed to set config because native hook is not fully installed | Native hook failed (fatal) and many others. Here are my detailed logs

[nfc.log](https://github.com/user-attachments/files/20051186/nfc.log)

> Unfortunately, the log does not show the attempted hook install. Please post another log either from boot or after terminating "com.android.nfc" service (with `killall -9 com.android.nfc` in a root [...]

roussosalex, 2025-05-05:
Hello, unfortunately the log still does not show the hook install. Please post another log either **from boot** or after **terminating "com.android.nfc" service** (with killall -9 com.android.nfc in a root shell).

mickey584, 2025-05-06:
> Hello, unfortunately the log still does not show the hook install. Please post another log either **from boot** or after **terminating "com.android.nfc" service** (with `killall -9 com.android.nfc` in a root shell).

Thank you again for your answer. I used root identity to killall -9 com.android.nfc, and then executed the APP again, and got the following log

[nfcgate_log.txt](https://github.com/user-attachments/files/20051392/nfcgate_log.txt)

mickey584, 2025-05-06:
> Thank you again for your answer. I used root identity to killall -9 com.android.nfc, and then executed the APP again, and got the following log [...]

Hello, in the log, line 10888 has: 05-06 08:22:01.140 24663 24714 I NATIVENFC: Command: INSTALL_HOOKS
05-06 08:22:01.140 24663 24714 E HOOKNFC: Native hook failed (fatal)

roussosalex, 2025-05-06:
> Thank you again for your answer. I used root identity to killall -9 com.android.nfc, and then executed the APP again, and got the following log

Thanks, that log now contains the hook install.

I have added a test commit to [this branch](/nfcgate/nfcgate/tree/211-oneplus-ace-3). Please try the test commit and report back if it changes anything. 
In case it does not work, please provide the log after terminating "com.android.nfc" service (with `killall -9 com.android.nfc` in a root shell)

mickey584, 2025-05-06:
[nfcgate_log.txt](https://github.com/user-attachments/files/20051944/nfcgate_log.txt)

HOOKNFC : Native hook failed (fatal)

Hello, the new branch still does not work for native hooks. After killall -9 com.android.nfc, the following is the new log: 

> > Thank you again for your answer. I used root identity to killall -9 com.android.nfc, and then executed the APP again, and got the following log
> 
> Thanks, that log now contains the hook install.
> 
> I have added a test commit to [this branch](/nfcgate/nfcgate/tree/211-oneplus-ace-3). Please try the test commit and report back if it changes anything. In case it does not work, please provide the log after terminating "com.android.nfc" service (with in a root shell)`killall -9 com.android.nfc`

roussosalex, 2025-05-06:
Thanks, I think the error is here in the log:

> 05-06 09:53:16.066 12038 12038 E NATIVENFC: Error at nfcd/src/main/jni/src/nfcd.cpp:201
> 05-06 09:53:16.066 12038 12038 E NATIVENFC: Could not obtain library handle, dl error: (null)
> 05-06 09:53:16.066 12038 12038 W NATIVENFC: Hooking setup failed with 2

I just pushed another commit to the same branch for better logging, please give it a try and post another log (after killing com.android.nfc).

mickey584, 2025-05-06:
> Thanks, I think the error is here in the log:
>
>
> I just pushed another commit to the same branch for better logging, please give it a try and post another log (after killing com.android.nfc).

Hello, this is a new log.

[logcat.log](https://github.com/user-attachments/files/20053052/logcat.log)

kleest, 2025-05-06:
Thank you, the log is very useful! The error is caused by the NFC NCI library having a different Android library namespace than the injected libnfcgate.so. Therefore, the hook cannot obtain a library handle and fails.

We try to find a fix for this behavior, however, it might take some time.

mickey584, 2025-05-06:
> Thank you, the log is very useful! The error is caused by the NFC NCI library having a different Android library namespace than the injected libnfcgate.so. Therefore, the hook cannot obtain a library handle and fails.
> 
> We try to find a fix for this behavior, however, it might take some time.

Thanks for your analysis. I have downgraded the device to ColorOs14/Android14. Although it still prompts that the native hook cannot be found and some errors about the nfcd/src/main/jni/src/nfcd.cpp file are reported, the NFC forwarding in Tag mode has been successful. At present, I suspect that all devices on ColorOs15[Android15] are unavailable, which may be a problem with the ColorOs system.

[image]

[image]

roussosalex, 2025-05-12:
Hello, we pushed a new possible fix (af2a17b84fbb70788c437e25ea817041622c6c1f) to the branch. Please try it out on your Android 15 build and report back if it works!

lazyhope, 2025-05-15:
> Hello, we pushed a new possible fix ([af2a17b](https://github.com/nfcgate/nfcgate/commit/af2a17b84fbb70788c437e25ea817041622c6c1f)) to the branch. Please try it out on your Android 15 build and report back if it works!

Hi could you please setup a build workflow for ace branch on GitHub action so we could easily test the new commit? I am also facing this issue with oxygen os 15 so would like to find out. Thank you!

kleest, 2025-05-15:
Oh right, we will add workflows for the other branches soon, in the meantime I created a PR so that the workflow runs for that branch. You can find the latest run + artifact here: https://github.com/nfcgate/nfcgate/actions/runs/15050088592?pr=213

@lazyhope Can you please get a log with `adb logcat | grep -i nfc` so that I can look at the full context. You can also send it to me via mail (see [readme](https://github.com/nfcgate/nfcgate?tab=readme-ov-file#contact)). Thanks!

kleest, 2025-05-15:
Thank you @lazyhope! I pushed a new commit 5edb0891b86d927e9aebb74f7837e111de91d2c5. Can you test again, please?

lazyhope, 2025-05-16:
> Thank you @lazyhope! I pushed a new commit 5edb0891b86d927e9aebb74f7837e111de91d2c5. Can you test again, please?

The native hook tick is green now✅, awesome!👍

lazyhope, 2025-05-17:
Hi, sorry to interrupt again, despite all checks greened and sessions being
able to be captured now, I found that the replay mode and clone mode unable
to emit NFC signals properly (no signal detected on my other phone in
reader mode), I have make sure it's not the issue with the phone or OS
itself but something wrong with the NFCGate app (I can emit properly in
other APP). I have attached you another complete logcat log for your
reference, could you please check if it's clear where the problem is? Much
appreciated!

lazyhope, 2025-05-17:
@kleest Hi, sorry to interrupt again, despite all checks greened and sessions being able to be captured now, I found that the replay mode and clone mode unable to emit NFC signal properly (no signal detected on my other phone in reader mode), I have make sure it's not the issue with the phone or OS itself but something wrong with the NFCGate app. I have emailed you another complete logcat log for you reference, could you please check if it's clear where the problem is? Much appreciated!

I asked gpt to analyse the logcat file and received the following response which seems reasonable, in case that's helpful:

From the logcat you provided, the core problem is that as soon as NFCGate tries to “take over” the NFC service’s configuration, the system NFC daemon (`com.android.nfc`) crashes with a null-pointer dereference (SIGSEGV at address 0x0). Once the NFC service crashes, there is nothing left to actually emit an NFC field or respond to cards—hence your “no-signal” symptom.

---

### 1. Evidence of the crash in the NFC service

Shortly after NFCGate injects its hook into the vendor NFC library, you see in the logs:

```log
Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0 in tid 9440 (ht), pid 8071 (com.android.nfc)
```

This is the system’s NFC service process (UID 1027) crashing outright .

Immediately afterwards:

```log
E/NfcService: Reader mode Binder was never registered.
```

…confirming that the NFC system service failed to complete its own initialization and registration of the HCE (“Host Card Emulation”) binder interface .

---

### 2. Why this happens

1. **Native-level hook mismatch**
   NFCGate uses a native library (`libnfcgate.so`) to hook into the vendor’s `NFC_SetConfig` function. On your OnePlus (Android 15) the vendor library’s function signatures and in-memory offsets have changed compared to the versions NFCGate was built against. When the hook calls through to the original function pointer, it ends up jumping to `0x0` (hence the null deref).

2. **Null-pointer dereference in `libnfc_jni`**
   If you scroll up a few frames in the stack trace, you’ll see entries like:

   ```
   #01 pc … in libnfc_jni… (nci_jni_initialize)
   #02 pc … in libnfc_jni… (JNI_OnLoad)
   ```

   That tells us the crash happens in the JNI layer of the NFC stack—i.e. where Android bridges Java’s `NfcAdapter` into the native vendor code .

3. **Service restarts and repeated failures**
   Because the NFC service is a critical system daemon, Android will try to restart it on failure, but each time NFCGate’s hook is in place it crashes again. No running service = no RF field = “emission” never happens.

---

### 3. What you can do

1. **Use the matching NFCGate build for your vendor library**

   * Check in NFCGate’s GitHub releases if there is a variant or a patch for OnePlus CPH2649 / Android 15.
   * If not, you will need to rebuild the native hook with the correct symbol names and offsets. Usually this involves dumping the vendor NFC library (`/system_ext/lib64/libnfc_jni.so`) and updating the Xposed hook definitions in `libnfcgate`.

2. **Confirm you have root and the proper hooking framework**

   * NFCGate’s native hook requires root to inject into `com.android.nfc`.
   * On Android 15 you may need Magisk + EdXposed/Riru to bypass SELinux enforcement on that process.

3. **Test a plain HCE sample**

   * Before using NFCGate, install a trivial HCE-only app (e.g. Google’s HCE sample) and verify that Host Card Emulation itself works on your device.
   * If the sample app also fails to register with the system service, the problem may be deeper (SELinux, broken NFC HAL, etc.).

4. **Watch for “Reader mode Binder was never registered”**

   * That error means the NFC service never got to the point of exposing the HCE binder interface to apps. Fixing the native-hook crash will automatically clear this.

---

#### In summary

> **The root cause** is a null-pointer dereference in the system NFC service—triggered by NFCGate’s native hook calling the wrong address in the vendor library.
> **The fix** is to use (or rebuild) the NFCGate native library exactly matched to your device’s NFC HAL symbols, and ensure your hooking framework (root + Xposed/Riru) is fully operational on Android 15.

Once the hook stops crashing the NFC daemon, “emit” mode in NFCGate will start working again.

kleest, 2025-05-17:
@lazyhope Unfortunately, I have not received your second log via mail yet.

lazyhope, 2025-05-17:
> @lazyhope Unfortunately, I have not received your second log via mail yet.

Oh that's weird, let me send it again

kleest, 2025-05-17:
@lazyhope The latest log starts with the crash and does not show the previous behavior of the NFC service, this makes it hard to debug the issue. Can you please send a new log that shows the full context and interaction of NFCGate with the NFC service? I suggest to do the following:
1. Start logcat
2. Stop NFC service by running `killall -9 com.android.nfc` in a root shell
3. Start clone/replay mode
4. Stop logcat

This ensures that the log is complete. Thank you!

closed 2025-05-19 by roussosalex