nfcgate/nfcgate #220: Cant replay cloned tags

Anajrim01, 2025-06-11:
Going into the Clone Mode, and selecting/or cloning and then trying to get the reader to read it, shows random android nfc ids and not the cloned tag ids.

**App Information**

[image]

**Device Information*** Which device are you using and what OS are you running on it?

Device: Samsung Galaxy S21 (Exynos)

OS: Android 14 and Android 15
Roms: OneUI (with stock kernel) and EliteUI with Proton Kernel

Logs for: Android 15, EliteUI with Proton Kernel

**Log**

[nfcgate_logs.txt](https://github.com/user-attachments/files/20695300/nfcgate_logs.txt)

**Steps to reproduce**

I have cloned a tag and then tried to scan it without any luck.

**Expected behavior**

The reader should think its the original tag being scanned.

**Additional context**

This has worked till one day it just stopped working.

kleest, 2025-06-14:
This should be fixed with the latest commits on [v2-dev branch](https://github.com/nfcgate/nfcgate/tree/v2-dev). Can you please test it ([link to APK](https://github.com/nfcgate/nfcgate/actions/runs/15651492794))?

Anajrim01, 2025-06-14:
The issue has indeed been fixed :)

closed 2025-06-14 by Anajrim01