nfcgate/nfcgate #221: Question regarding ATS

hurricanefrog, 2025-06-23:
I'm playing around with the replay functionality.

One thing I noticed is that libnfc/libfreefare don't detect a relayed DESfire transponder as DESfire. The reason seems to be the ATS, which is read as `78 80 40 00 80`. With an actual DESfire transponder having an ATS of `75 77 81 02 80`.

I *assume* this is the smartcard readers ATS.

Can/Should NFCGate be able to relay the ATS? (Probably not, the log view of NFCGate doesn't show anything during that identification phase).

The "Status" page of the NFCGate app shows all green on all tests.
(Samsung Galaxy M32 with an Pixel Experience GSI; Google Pixel 4a with Lineageos 22.2),

oticasimpless, 2025-06-26:
Which cell phones can you emulate the relay completely?

roussosalex, 2025-08-04:
The ATS not only contains theoretically "configurable" elements like FWI and SFGI, but also physical limitations, e.g. supported bit rates. If the hardware on the device does not support it, it cannot be emulated.