nfcgate/nfcgate #222: MITM

cabelito, 2025-06-25:
There used to be a module MITM , but no documentation how to use it. If you want to rewrite some APDU or RESPONSE on the fly , how can you accomplish it ? Is it on the server side ?
I assume it is used in relay mode, however . A little help will be appreciated.

kleest, 2025-06-25:
Which mitm module are you referring to? The `mitm` repository is archived because it does not work with the current NFCGate version. If you want to modify data, you have to use NFCGate server and implement a plugin in Python.

cabelito, 2025-06-25:
I see, I was trying to use the mitm module ... so I have to write my own plugin ? for mitm ? is this correct ?

kleest, 2025-06-25:
Yes. In [my previous comment](https://github.com/nfcgate/nfcgate/issues/222#issuecomment-3005905276), I gave a overview what to do.

harryradburn5-design, 2025-08-04:
tag mode doesn not work properly on classic NFC gate app. it shows that it was emulated but no emulation at all happens

kleest, 2025-08-04:
@harryradburn5-design please stop bumping issues with comments unrelated to the thread. If you have issues with the app on your device, please create a new issue and follow the template closely. Thank you.

closed 2025-08-04 by kleest