nfcgate/nfcgate #225: nfcgate crashes in relay mode when connecting to server with TLS enabled

Lassie111, 2025-07-27:
**Describe the bug**

The crashes happen even before involving a tag.

**App Information***

All green.

**Device Information*** Which device are you using and what OS are you running on it?

Device: Galaxy S21 FE

OS: Android 15

**Log***

> 07-28 02:36:01.563 18360 18360 E AndroidRuntime: Process: de.tu_darmstadt.seemoo.nfcgate, PID: 18360
07-28 02:36:01.563 18360 18360 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.BaseNetworkFragment.handleStatus(BaseNetworkFragment.java:102)
07-28 02:36:01.563 18360 18360 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode.lambda$onNetworkStatus$1(RelayFragment.java:89)
07-28 02:36:01.563 18360 18360 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode.$r8$lambda$TpROqs5uznKlUD9PcsQfh6aaHnA(Unknown Source:0)
07-28 02:36:01.563 18360 18360 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode$$ExternalSyntheticLambda0.run(D8$$SyntheticClass:0)
07-28 02:42:01.158 20209 20209 E AndroidRuntime: Process: de.tu_darmstadt.seemoo.nfcgate, PID: 20209
07-28 02:42:01.158 20209 20209 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.BaseNetworkFragment.handleStatus(BaseNetworkFragment.java:102)
07-28 02:42:01.158 20209 20209 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode.lambda$onNetworkStatus$1(RelayFragment.java:89)
07-28 02:42:01.158 20209 20209 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode.$r8$lambda$TpROqs5uznKlUD9PcsQfh6aaHnA(Unknown Source:0)
07-28 02:42:01.158 20209 20209 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode$$ExternalSyntheticLambda0.run(D8$$SyntheticClass:0)
07-28 02:42:06.582 20775 20775 E AndroidRuntime: Process: de.tu_darmstadt.seemoo.nfcgate, PID: 20775
07-28 02:42:06.582 20775 20775 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.BaseNetworkFragment.handleStatus(BaseNetworkFragment.java:102)
07-28 02:42:06.582 20775 20775 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode.lambda$onNetworkStatus$1(RelayFragment.java:89)
07-28 02:42:06.582 20775 20775 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode.$r8$lambda$TpROqs5uznKlUD9PcsQfh6aaHnA(Unknown Source:0)
07-28 02:42:06.582 20775 20775 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode$$ExternalSyntheticLambda0.run(D8$$SyntheticClass:0)
07-28 02:42:14.170 20888 20888 E AndroidRuntime: Process: de.tu_darmstadt.seemoo.nfcgate, PID: 20888
07-28 02:42:14.170 20888 20888 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.BaseNetworkFragment.handleStatus(BaseNetworkFragment.java:102)
07-28 02:42:14.170 20888 20888 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode.lambda$onNetworkStatus$1(RelayFragment.java:89)
07-28 02:42:14.170 20888 20888 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode.$r8$lambda$TpROqs5uznKlUD9PcsQfh6aaHnA(Unknown Source:0)
07-28 02:42:14.170 20888 20888 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode$$ExternalSyntheticLambda0.run(D8$$SyntheticClass:0)
07-28 02:42:36.865 21188 21188 E AndroidRuntime: Process: de.tu_darmstadt.seemoo.nfcgate, PID: 21188
07-28 02:42:36.865 21188 21188 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.BaseNetworkFragment.handleStatus(BaseNetworkFragment.java:102)
07-28 02:42:36.865 21188 21188 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode.lambda$onNetworkStatus$1(RelayFragment.java:89)
07-28 02:42:36.865 21188 21188 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode.$r8$lambda$TpROqs5uznKlUD9PcsQfh6aaHnA(Unknown Source:0)
07-28 02:42:36.865 21188 21188 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode$$ExternalSyntheticLambda0.run(D8$$SyntheticClass:0)
07-28 02:43:40.904 22245 22245 E AndroidRuntime: Process: de.tu_darmstadt.seemoo.nfcgate, PID: 22245
07-28 02:43:40.904 22245 22245 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.BaseNetworkFragment.handleStatus(BaseNetworkFragment.java:102)
07-28 02:43:40.904 22245 22245 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode.lambda$onNetworkStatus$1(RelayFragment.java:89)
07-28 02:43:40.904 22245 22245 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode.$r8$lambda$TpROqs5uznKlUD9PcsQfh6aaHnA(Unknown Source:0)
07-28 02:43:40.904 22245 22245 E AndroidRuntime:    at de.tu_darmstadt.seemoo.nfcgate.gui.fragment.RelayFragment$UIRelayMode$$ExternalSyntheticLambda0.run(D8$$SyntheticClass:0)
07-28 02:43:41.004  3974  4063 V WindowManager:         {WCT{RemoteToken{867704f Task{c9d2f84 #42 type=standard A=10337:de.tu_darmstadt.seemoo.nfcgate}}} m=CLOSE f=NONE leash=Surface(name=Task=42)/@0x591c729 sb=Rect(0, 0 - 1080, 2340) eb=Rect(0, 0 - 1080, 2340) d=0 taskParent=-1 inset=Rect(0, 0 - 0, 0) dim=true},
07-28 02:43:41.362  3974  4036 V WindowManager: Setting visibility of Window{953daca u0 de.tu_darmstadt.seemoo.nfcgate/de.tu_darmstadt.seemoo.nfcgate.gui.MainActivity EXITING}: false, caller=com.android.server.wm.WindowContainer.sendAppVisibilityToClients:19 com.android.server.wm.WindowToken.setClientVisible:49 com.android.server.wm.ActivityRecord.setClientVisible:8 com.android.server.wm.ActivityRecord.commitVisibility:273 com.android.server.wm.TransitionController.finishTransition:766
07-28 02:43:41.363  3974  4036 W WindowManager: Exception thrown during dispatchAppVisibility Window{953daca u0 de.tu_darmstadt.seemoo.nfcgate/de.tu_darmstadt.seemoo.nfcgate.gui.MainActivity EXITING}
07-28 02:43:41.364  3974  4036 V WindowManager: Remove Window{953daca u0 de.tu_darmstadt.seemoo.nfcgate/de.tu_darmstadt.seemoo.nfcgate.gui.MainActivity EXITING}: mSurfaceController=Surface(name=de.tu_darmstadt.seemoo.nfcgate/de.tu_darmstadt.seemoo.nfcgate.gui.MainActivity$_22245)/@0xbfcb2f0 mAnimatingExit=true mRemoveOnExit=true mHasSurface=true surfaceShowing=true animating=false app-animation=false mDisplayFrozen=false callers=com.android.server.wm.WindowToken.removeAllWindowsIfPossible:46 com.android.server.wm.ActivityRecord.removeIfPossible:4 com.android.server.wm.ActivityRecord.handleCompleteDeferredRemoval:5 com.android.server.wm.WindowContainer.handleCompleteDeferredRemoval:21 com.android.server.wm.TaskFragment.handleCompleteDeferredRemoval:17 com.android.server.wm.WindowContainer.handleCompleteDeferredRemoval:21
07-28 02:43:41.367  3974  4036 V WindowManager: Remove Window{953daca u0 de.tu_darmstadt.seemoo.nfcgate/de.tu_darmstadt.seemoo.nfcgate.gui.MainActivity EXITING}: mSurfaceController=Surface(name=de.tu_darmstadt.seemoo.nfcgate/de.tu_darmstadt.seemoo.nfcgate.gui.MainActivity$_22245)/@0xbfcb2f0 mAnimatingExit=true mRemoveOnExit=true mHasSurface=true surfaceShowing=true animating=false app-animation=false mDisplayFrozen=false callers=com.android.server.wm.WindowToken.removeAllWindowsIfPossible:46 com.android.server.wm.ActivityRecord.removeIfPossible:4 com.android.server.wm.ActivityRecord.onRemovedFromDisplay:499 com.android.server.wm.ActivityRecord.removeImmediately:46 com.android.server.wm.ActivityRecord.removeIfPossible:7 com.android.server.wm.ActivityRecord.handleCompleteDeferredRemoval:5
07-28 02:43:41.367  3974  4036 I WindowManager: Destroying surface Surface(name=de.tu_darmstadt.seemoo.nfcgate/de.tu_darmstadt.seemoo.nfcgate.gui.MainActivity$_22245)/@0xbfcb2f0 called by com.android.server.wm.WindowStateAnimator.destroySurface:11 com.android.server.wm.WindowStateAnimator.destroySurfaceLocked:54 com.android.server.wm.WindowState.removeImmediately:43 com.android.server.wm.WindowContainer.removeImmediately:37 com.android.server.wm.WindowToken.removeImmediately:11 com.android.server.wm.ActivityRecord.removeImmediately:61 com.android.server.wm.ActivityRecord.removeIfPossible:7 com.android.server.wm.ActivityRecord.onRemovedFromDisplay:499
07-28 02:43:41.410  3974  4066 W ActivityTaskManager: Activity top resumed state loss timeout for ActivityRecord{5c6b416 u0 de.tu_darmstadt.seemoo.nfcgate/.gui.MainActivity t-1 f}}
07-28 02:44:19.913  4620  4811 D TileLifecycleManager: Unbinding service Intent { act=android.service.quicksettings.action.QS_TILE cmp=com.android.nfc/com.samsung.android.nfc.quicktile.NfcTile (has extras) } UserHandle{0}
07-28 02:44:19.918  4834  4834 D [NFC_SE]NfcTile: onUnBind()
07-28 02:44:23.268  5299  5318 W id.app.launcher: ApkAssets: Deleting an ApkAssets object '<empty> and /data/app/~~0lVsyHMz-LxL7GEwd4T_hw==/de.tu_darmstadt.seemoo.nfcgate-W6pzVPQ9g13_3Tv0112h2A==/base.apk' with 1 weak references

kleest, 2025-07-29:
Can you please post the full exception log including the exception name?  Your stacktrace points to `mStatusBanner.setError(getString(R.string.network_error));`, where I don't see any obvious failure. Unfortunately, I cannot reproduce the crash the way you described it.

> Side question: nfcgate app can't connect to the server at all when TLS is enabled. The app shows "Error: Network: Error" and the server shows:

It looks like the TLS handshake between NFCGate and the server does not work. The log may contain more information about the failure. If the TLS handshake works, NFCGate shows you a dialog with the server certificate information and you can block or trust this certificate.

Lassie111, 2025-07-29:
I will attach a new log. This is everything from the logcat, filtered by `grep -E- i "xhook|nfc"`.

[log_crash.txt](https://github.com/user-attachments/files/21498763/log_crash.txt)

> If the TLS handshake works, NFCGate shows you a dialog with the server certificate information and you can block or trust this certificate.

The dialog was shown, the error appears after trusting the certificate.

Lassie111, 2025-07-29:
https://github.com/user-attachments/assets/22e0d452-1757-4d2f-8d6d-274f728cc633

kleest, 2025-08-02:
Thanks for the additional information and the video.

Two things:

1. It looks like there are some fragment lifecycle issues when switching fragments. Until we fix this, you can reset the connection by pressing the "refresh" button in the top right.
2. There is an issue with TLS certificate verification when connecting to a server using an IP address. If the certificate's CN does not match the reverse lookup of the server's DNS name, the certificate verification will fail. This is also logged in logcat. Is this the TLS issue you are facing?

kleest, 2025-08-02:
Unfortunately, there are no logs from NFCGate in the output. Can you please check if you can get all logs from the NFCGate app [via this method](https://stackoverflow.com/a/75722004)?

Lassie111, 2025-08-02:
[nfcgate.txt](https://github.com/user-attachments/files/21559914/nfcgate.txt)

It says that hostname does not match but it does? The certificate was created with command:
`openssl req -new -x509 -days 365 -nodes -out cert.pem -keyout key.pem -subj "/CN=<mydomain>"`, where `<mydomain>` is the same as the hostname entered in nfcgate.

kleest, 2025-08-02:
Does your certificate have SAN fields? Using only CN in the certificate is not supported by the default hostname verifier on Android. You have to add your hostname via SAN.

Lassie111, 2025-08-02:
It didn't have SAN fields and that was the problem. Thank you!

kleest, 2025-08-04:
Crash fixed by c2fc5ad699d5807d8576cf035ae3d9be6242e7ba